JeffersonLab/halld_recon #25: my eventnumbers not being assigned to the same event based on number of threads

wmcginle, 2018-09-12:
I see different event numbers initialized by evnt() being assigned to the same event based on the number of threads I use to process the data. I'm using the master branch of halld_recon and JANA version jana_0.7.9p1. If anyone else would check the same behavior that would be great, hopefully I have some environment or versioning set up incorrectly - thanks.

[eventnum_thread_dep.pdf](https://github.com/JeffersonLab/halld_recon/files/2376843/eventnum_thread_dep.pdf)

sdobbs, 2018-09-12:
Will, would you mind doing the comparison with this additional parameter:
-PEVIO:NTHREADS=1

wmcginle, 2018-09-12:
Hi Sean,

I added added 2 slides to show the effect of adding this additional parameter of -PEVIO:NTHREADS=1. The agreement between -PNTHREADS=1 and -PNTHREADS=4 is closer now but if I re-run with -PNTHREADS=4 I see inconsistencies between 3 trials. When I add -PEVIO:NTHREADS=1 to the -PNTHREADS=1 I still have consistency between 3 trials.

[eventnum_thread_dep.pdf](https://github.com/JeffersonLab/halld_recon/files/2376940/eventnum_thread_dep.pdf)

zihlmann, 2018-09-14:
read 10000 events and save eventnumber number of tof hits , bcal showers and fcal showers into a string.
print these strings at the end to a file and sort by event number. do this for 1 thread and 4 threads.
I do not find any difference in the output.

faustus123, 2018-09-14:
I’ve done a similar test but haven’t reproduced the issue yet. My testing has not been
exhaustive though. Will, can you make you code available so I can make sure I’m
doing what you are?

Regards,
-David

-------------------------------------------------------------
David Lawrence Ph.D.
Staff Scientist, Thomas Jefferson National Accelerator Facility
Newport News, VA
davidl@jlab.org
(757) 269-5567 W
(757) 746-6697 C

wmcginle, 2018-09-14:
Thanks for checking this, Beni and David. I made a clean plugin on the ifarm and I can't reproduce what I was seeing at CMU. This seems to be a personal issue that I'm having and will look into. Sorry for this.

closed 2018-09-14 by wmcginle

mashephe, 2018-09-14:
Will,

Please do report on the the origin though once you figure it out, assuming it isn't a trivial error.  It is possible that others are making the same mistake or having the same issue.

Matt