JeffersonLab/halld_recon #47: bad f250 pulse data

markito3, 2018-11-09:
Hi,

I have run into a problem with the evio files in run 50924 - hd_root finds a bad f250 pulse and then throws an JException and crashes.  Files 000 and 001 crash instantly, file 002 proceeds for 2.1k events and then crashes.  I'm using halld_recon version 3.2.0. 

Does anyone know which runs do and don't have these errors in them?   

Was this already found by the automatic monitoring software?   (if not, why not?)

Naomi.

nsjarvis, 2018-11-09:
Note that PlotBrowser incoming data is completely blank (not even an empty histogram) for 50924 and a few other runs.  That is probably a BIG clue - we should pay more attention to these.

faustus123, 2018-11-09:
This was seen back at the time the run was taken. See logbook:

https://logbooks.jlab.org/entry/3608646

The bottom line is, the issue was fixed by a reboot of the crate and the data for the affected runs is unrecoverable.

faustus123, 2018-11-09:
I just closed this issue with this comment:

This was seen back at the time the run was taken. See logbook:

https://logbooks.jlab.org/entry/3608646 <https://logbooks.jlab.org/entry/3608646>
The bottom line is, the issue was fixed by a reboot of the crate and the data for the affected runs is unrecoverable.

Regards,
-David

-------------------------------------------------------------
David Lawrence Ph.D.
Staff Scientist, Thomas Jefferson National Accelerator Facility
Newport News, VA
davidl@jlab.org
(757) 269-5567 W
(757) 746-6697 C

closed 2018-11-09 by faustus123

nsjarvis, 2018-11-09:
It should be marked as JUNK in RCDB then, please.

sdobbs, 2018-11-09:
We usually keep track of this info with the "status" field.  It's on my list...

nsjarvis, 2018-11-09:
Last time I looked there were no is_production runs for 2018 fall yet, let
alone status=approved.  Sean's to-do list is too long (and this run period
is a real challenge as we have not finished calibrating the previous one
yet).  Sean, if you want to start up a spreadsheet, I can try to help fill
the approved status with the obvious yes/no.  At first this is what I
thought the spring-2018 monitoring volunteer group was for.  :-(

Could we implement a really basic monitoring process to run over a few
complete evio files from each run just to check that the data is readable?
This should produce an easy yes/no with the minimum of thought required.

Naomi.