JeffersonLab/halld_recon #95: Crashes with analysis library when adding multiple similar reactions

sdobbs, 2019-02-05:
From Ashley:

I was trying to test the reaction filter with the reactions below. From what I found, the analysis library crashes when adding a pi0 or gamma  to a reaction containing particle index 23 (xim).  I've looked at similar reactions with xi0 and don't see anything crashes.

I'm using software version_3.9_jlab.xml and using the command: 

 hd_root /cache/halld/RunPeriod-2017-01/recon/ver03/REST/030730/dana_rest_030730_000.hddm --config=hd_root_test.config -PEVENTS_TO_KEEP=1000

Reaction1 1_14__11_11_23 #kpkpxim
Reaction2 1_14__7_11_11_23 #piokpkpxim
Reaction3 1_14__8_11_16_23 #pipkpksxim
Reaction4 1_14__1_11_11_23 #gkpkpxim
Reaction5 1_14__8_9_11_11_23 #pippimkpkpxim
Reaction6 1_14__7_7_11_11_23 #pi0pi0kpkpxim
Reaction7 1_14__7_8_11_16_23 #pi0pipkpksxim

From the reactions, the following combinations crash usually with 272 events processed and 282 events read: 

1,2 

1,4 

1,6 

2,6 

3,7 

4,6 

and these don't crash:

1,3

1,5

1,7

2,3

2,4

2,5

2,7

3,4

3,5

3,6

4,5

4,7

5,6

5,7

6,7

From analysis launches, Alex had issues with reaction 2 in 2017 version 20 (with reaction 1) and with reactions 2 and 4 in 2017 version 09 (with reaction 1), 

I don't see any crashes with these channels:

Reaction8 1_14__11_16_22 #kpksxi0
Reaction9 1_14__9_11_11_22 #pimkpkpxi0
Reaction10 1_14__7_11_16_22 #pi0kpksxi0
Reaction11 1_14__1_11_16_22 #gkpksxi0
Reaction12 1_14__7_9_11_11_22 #pi0pimkpkpxim
Reaction13 1_14__8_9_11_16_22#pippimkpksxi0
Reaction24 1_14__7_7_11_16_22 #pi0pi0kpksxi0

aaust, 2019-02-05:
Thanks, having an explicit example helped a lot. I know how to avoid the crash, but it would be interesting to test whether we lose any signal events with my patch. On 100k events, I did not see any differences between running 1 and 2 individually or combined.

sdobbs, 2019-02-05:
OK, what is the next step, then?  Do you want to generate 2 sets of ROOT trees with your patch (individual and combined), and have @acernst1 analyze them to see if any signal events are missing?

aaust, 2019-02-05:
Yes, unless you have skimmed REST files for the signal events. At  the moment, I am just comparing the number of entries in the produced trees.

sdobbs, 2019-02-05:
She has an event list?

closed 2019-04-10 by aaust