JeffersonLab/halld_recon #111: crashes when running analysis library over multiple runs

sdobbs, 2019-02-20:
I generated simulations for several runs, and ran ReactionFilter over them, and got a crash in the following location:

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffcb7fe700 (LWP 23426)]
DAnalysis::DParticleComboCreator::Build_ThrownCombo (this=0x60, locEventLoop=locEventLoop@entry=0x7fffc00008c0) at libraries/ANALYSIS/DParticleComboCreator.cc:800
800		if(dAnalysisUtilities == nullptr)
Missing separate debuginfos, use: debuginfo-install bzip2-libs-1.0.6-13.el7.x86_64 expat-2.1.0-8.el7.x86_64 freetype-2.4.11-11.el7.x86_64 glibc-2.17-105.el7.x86_64 keyutils-libs-1.5.8-3.el7.x86_64 krb5-libs-1.13.2-10.el7.x86_64 libcom_err-1.42.9-7.el7.x86_64 libgcc-4.8.5-4.el7.x86_64 libgfortran-4.8.5-4.el7.x86_64 libquadmath-4.8.5-4.el7.x86_64 libselinux-2.2.2-6.el7.x86_64 libstdc++-4.8.5-4.el7.x86_64 nss-softokn-freebl-3.16.2.3-13.el7_1.x86_64 openssl-libs-1.0.1e-42.el7.9.x86_64 pcre-8.32-15.el7.x86_64 xz-libs-5.1.2-12alpha.el7.x86_64 zlib-1.2.7-15.el7.x86_64
(gdb)

aaust, 2019-02-20:
What did you simulate? Which reactions did you select in the ReactionFilter? Do you have a REST file to reproduce the crash?

sdobbs, 2019-06-17:
Sorry for not updating this.  Elton sees a similar problem, email copied from the gluex-software list below:

I have now gotten back to running these files again and have reproduced the problem.

If I run over run 
30796 (last file) by itself, ok.
30797 (first file) by itself, ok.
30796 (last file) and 30797 (first file) together, I get a crash at the start of processing.

The files are located on 
/cache/halld/halld-scratch/REQUESTED_MC/2017/gen_amp__3piHighMass_t2.5__G4_20190501054440pm/hddm/dana_rest_gen_amp_030796_010.hddm
/cache/halld/halld-scratch/REQUESTED_MC/2017/gen_amp__3piHighMass_t2.5__G4_20190501054440pm/hddm/dana_rest_gen_amp_030797_000.hddm

My configuration file is
---------------------------------
PLUGINS monitoring_hists,ReactionFilter

#Ncores if you want to use all of the hyperthreads available (only useif you have the whole node!)
NTHREADS 1

JANA_CALIB_CONTEXT variation=mc
#calibtime=YYYY-MM-DD-hh-mm-ss (hr is 24-hr format)

Reaction1  1_14__7_8_9_14                      # pi0 pi+ pi- p

#Reaction1:Flags F4 B1 T3 # Default values F4 (vertex-p4) B1 (RF bunches +/-1) T3 (extra charged tracks)  
Reaction1:Flags B4
---------------------------------------

sdobbs, 2019-06-17:
I looked into this a bit, and I think the problem is similar to what was affecting the monitoring launch, except in this case there are pointers to objects being stored that are being deleted between different runs.  

The problem seems limited to the analysis library, as I'm able to run monitoring_hists over both files without a problem

Untangling this will take some work, given the interconnected nature of some of these classes...

eltonssmith, 2019-09-04:
Hi Sean,

I checked that the previous crashes have been resolved with the latest version of the master:

At first I did not update my hd_root program and reproduced the problem that you detailed two entries ago. I ran hd_root with the same jana.config file over the same runs, namely:
If I run over run
30796 (last file) by itself, ok.
30797 (first file) by itself, ok.
30796 (last file) and 30797 (first file) together, I get a crash at the start of processing.
I got the same behavior as before.

I then updated my version of the master with the latest updates and generated new executables. I reran the exact same command with the two files. It ran to completion (34.4 k events). I checked the hd_root output and it all looked normal. 

Thanks for following up on this!

sdobbs, 2019-09-05:
Closing this issue unless someone else reports a problem.

closed 2019-09-05 by sdobbs