JeffersonLab/halld_recon #116: crash in monitoring for PrimeX

aaust, 2019-02-25:
The latest monitoring runs all crashed with the same error:

```
JANA >>Reading CCAL profile data from /group/halld/www/halldweb/html/resources//CCAL/profile_data/ccal_profile_data_v1.dat ...
At line 27 of file libraries/CCAL/island.F (unit = 22, file = '�')
Fortran runtime error: File '/group/halld/www/halldweb/html/resources//CCAL/profile_data/ccal_profile_data_v1.dat^' does not exist
```
I reproduced the error with the file
/cache/halld/RunPeriod-2019-01/rawdata/Run061260/hd_rawdata_061260_000.evio

sdobbs, 2019-02-25:
Could you please point to a log file?  This should be loaded as a resource...

aaust, 2019-02-25:
/work/halld2/data_monitoring/RunPeriod-2019-01/mon_ver01/log/061260/stdout.061260_000.out
/work/halld2/data_monitoring/RunPeriod-2019-01/mon_ver01/log/061260/stderr.061260_000.err

sdobbs, 2019-02-25:
yeah this is a memory error - a fix exists, I'll get it

sdobbs, 2019-02-25:
should be fixed by #118

aaust, 2019-02-25:
sorry, it isn't

sdobbs, 2019-02-25:
#119 should fix it.  there are a bunch of warning messages which need to be limited, but we can run at least

aaust, 2019-02-25:
There is a massive amount of warning messages, and then it still crashes

sdobbs, 2019-02-25:
OK, I commented out the worst warning and cleared things up.  This works for some other files, so I would go ahead and use this for now.

Keep the issue open, we are working to get rid of global variables and other nasty memory management issues...

aaust, 2019-02-26:
Ok thanks, I don't see a crash with the file mentioned above. Just a few messages like:

> WRN bad cluster log. coord, center id = 85 322.500000

closed 2019-03-12 by aaust