JeffersonLab/halld_recon #117: Is the CompCal in the default geometry?

sdobbs, 2019-02-25:
I am seeing this error when running the latest master on run 

libraries/HDGEOMETRY/DGeometry.cc:1706 Unable to retrieve ComptonEMcal position.

rjones30, 2019-02-25:
Sean,

No, at the moment the ComptonEMcal is commented out in main_HDDS.xml, both
in the default hdds geometry and in any of the ccdb geometries that I have
looked at so far. To fix this, we need some offsets and placement angles,
together with a list of runs for which they should be applied. Do you want
to take care of this or do you want me to do it?

-Richard Jones

sdobbs, 2019-02-25:
Richard,

We are waiting on the official survey results for the final position, so
I'll coordinate adding this with Simon.

---Sena

sdobbs, 2019-04-05:
This is in the CCDB geometry now and propagated through appropriately.  Closing...

closed 2019-04-05 by sdobbs