JeffersonLab/halld_recon #125: tons of unnecessary output in cout stream from CCAL shower code And then CRASH

zihlmann, 2019-02-28:
Below is the output and the crash: (the analysis was done on run 61343 asking for CCALShower)
.......
 case 0 ch
 case 0 ch
 case 0 ch
 case 0 ch
 case 0 ch
 case 0 ch
 case 0 ch
 case 0 ch
 case 0 ch
 case 0 ch
hd_root: libraries/CCAL/DCCALGeometry.cc:83: DVector2 DCCALGeometry::positionOnFace(int) const: Assertion `channel >= 0 && channel < m_numActiveBlocks' failed.
Abort

sdobbs, 2019-02-28:
Were you running multithreaded?

zihlmann, 2019-02-28:
Yes I was! I just now run in single thread mode and things seem to look better.

sdobbs, 2019-02-28:
Yes, multithreading doesn't work with CCAL reconstruction [yet]...
Someone does need to clean up the print outs, though.

sdobbs, 2019-03-10:
Are these printouts still showing up for you?

sdobbs, 2019-03-21:
unnecessary printouts seem to be reduced, closing

closed 2019-03-21 by sdobbs