JeffersonLab/halld_recon #132: Crash for PrimEx data with CDC included

aaust, 2019-03-10:
with for example run 61462 and any high-level plugin (danarest, monitoring_hists, highlevel_online, ...), I get an immediate crash:

```
JANA >>============================
JANA >> DDetectorMatches: (evnt)     --   line:299  /u/group/halld/Software/builds/Linux_CentOS7-x86_64-gcc4.8.5/jana/jana_0.7.9p1^ccdb166/Linux_CentOS7-x86_64-gcc4.8.5/include/JANA/JFactory.h
JANA >> DDetectorMatches             
JANA >> DNeutralShower: (evnt)       --   line:299  /u/group/halld/Software/builds/Linux_CentOS7-x86_64-gcc4.8.5/jana/jana_0.7.9p1^ccdb166/Linux_CentOS7-x86_64-gcc4.8.5/include/JANA/JFactory.h
JANA >> DNeutralShower               
JANA >> JEventLoop:OneEvent  (evnt)  --   line:695  src/JANA/JEventLoop.cc
JANA >>----------------------------
```

sdobbs, 2019-03-10:
The problem seems to be with DDetectorMatches_factory::MatchToDIRC() - after commenting that out, I can run without this crash.

sdobbs, 2019-03-10:
OK, the problem seems to be that the DIRC extrapolations are not set up for the straight line tracks, since this line throws an exception:

        vector<DTrackFitter::Extrapolation_t> extrapolations=locTrackTimeBased->extrapolations.at(SYS_DIRC);                                                                     

Maybe @staylorjlab can take a look at this.

Meta issue:  why isn't the location or content of the exception given in more detail?  Do we just need to be better at checking exceptions?  Or should we make sure that a JException is being thrown rather than a standard C++ exception?

aaust, 2019-03-15:
solved by #133

closed 2019-03-15 by aaust