JeffersonLab/halld_recon #157: crash in hd_root with MC data

T-Britton, 2019-04-23:
The smeared file causing the crash was regenerated around half a dozen times so it does not appear to be an issue with a specific edge case event.  It runs with out error with the default variation. It runs without error in variation=mc when random triggers are not included. /work/halld2/home/tbritton/hd_root_debug/ contains the smeared file and a jana configuration file to reproduce the crash.

Run 51384 variation=mc causes:

======STACK TRACE BELOW=====

JANA ERROR>> didn't sleep full 0.5 seconds! 0.0Hz  (avg.: 0.2Hz)    
  2.0 events processed  (10.0 events read)  0.0Hz  (avg.: 0.2Hz)    

===========================================================
There was a crash.
This is the entire stack trace of all threads:
===========================================================

Thread 2 (Thread 0x7f1393819700 (LWP 170075)):
#0  0x00007f13b2f00159 in waitpid () from /lib64/libc.so.6
#1  0x00007f13b2e7dde2 in do_system () from /lib64/libc.so.6
#2  0x00007f13b2e7e191 in system () from /lib64/libc.so.6
#3  0x00007f13b97e7b19 in TUnixSystem::StackTrace (this=0x22d38e0) at /u/group/halld/Software/builds/Linux_RHEL7-x86_64-gcc4.8.5/root/root-6.08.06/core/unix/src/TUnixSystem.cxx:2405
#4  0x00007f13b97ea3ec in TUnixSystem::DispatchSignals (this=0x22d38e0, sig=kSigSegmentationViolation) at /u/group/halld/Software/builds/Linux_RHEL7-x86_64-gcc4.8.5/root/root-6.08.06/core/unix/src/TUnixSystem.cxx:3625
#5  <signal handler called>
#6  Calc_RFBunchShift (this=<optimized out>, locTimeToStepTo=<error reading variable: Cannot access memory at address 0x60>, locTimeToStep=17637937284.106651) at libraries/ANALYSIS/DSourceComboTimeHandler.h:302
#7  Calc_RFBunchShift (locTimeToStepTo=<error reading variable: Cannot access memory at address 0x60>, this=0x7f135c9376a0) at libraries/ANALYSIS/DSourceComboTimeHandler.h:68
#8  DAnalysis::DSourceComboVertexer::Calc_VertexTimeOffsets_WithBeam (this=0x7f135c933210, locReactionVertexInfo=0x7f136ce3c8e0, locReactionFullComboUse=std::tuple containing = {...}, locReactionFullCombo=locReactionFullCombo
entry=0x7f135c94eb10, locBeamParticle=0x7f138520a260) at libraries/ANALYSIS/DSourceComboVertexer.cc:257
#9  0x000000000085a40b in DAnalysis::DSourceComboer::Combo_WithBeam (this=this
entry=0x7f136fb6a360, locReactions=std::vector of length 1, capacity 1 = {...}, locReactionVertexInfo=locReactionVertexInfo
entry=0x7f136ce3c8e0, locReactionFullComboUse=std::tuple containing = {...}, locReactionFullCombo=0x7f135c94eb10, locRFBunch=locRFBunch
entry=0, locOutputComboMap=std::unordered_map with 1 elements = {...}) at libraries/ANALYSIS/DSourceComboer.cc:1708
#10 0x0000000000869daa in DAnalysis::DSourceComboer::Build_ParticleCombos (this=0x7f136fb6a360, locReactionVertexInfo=0x7f136ce3c8e0) at libraries/ANALYSIS/DSourceComboer.cc:1521
#11 0x000000000069d43e in DAnalysisResults_factory::evnt (this=0x7f138c12e580, locEventLoop=0x7f138c0008c0, eventnumber=<optimized out>) at libraries/ANALYSIS/DAnalysisResults_factory.cc:282
#12 0x0000000000690943 in jana::JFactory<DAnalysisResults>::Get (this=this
entry=0x7f138c12e580, d=std::vector of length 0, capacity 0) at /u/group/halld/Software/builds/Linux_RHEL7-x86_64-gcc4.8.5/jana/jana_0.7.9p1^ccdb166/Linux_RHEL7-x86_64-gcc4.8.5/include/JANA/JFactory.h:295
#13 0x00000000006914a4 in jana::JEventLoop::GetFromFactory<DAnalysisResults> (this=this
entry=0x7f138c0008c0, t=std::vector of length 0, capacity 0, tag=tag
entry=0x12cf8c7 "", data_source=
0x7f1393812190: 1553465728, allow_deftag=allow_deftag
entry=true) at /u/group/halld/Software/builds/Linux_RHEL7-x86_64-gcc4.8.5/jana/jana_0.7.9p1^ccdb166/Linux_RHEL7-x86_64-gcc4.8.5/include/JANA/JEventLoop.h:452
#14 0x0000000000691774 in jana::JEventLoop::Get<DAnalysisResults> (this=this
entry=0x7f138c0008c0, t=std::vector of length 0, capacity 0, tag=tag
entry=0x12cf8c7 "", allow_deftag=allow_deftag
entry=true) at /u/group/halld/Software/builds/Linux_RHEL7-x86_64-gcc4.8.5/jana/jana_0.7.9p1^ccdb166/Linux_RHEL7-x86_64-gcc4.8.5/include/JANA/JEventLoop.h:309
#15 0x0000000000709af0 in DEventWriterROOT::Fill_DataTrees (this=this
entry=0x7f136cd52ce0, locEventLoop=locEventLoop
entry=0x7f138c0008c0, locDReactionTag="ReactionFilter") at libraries/ANALYSIS/DEventWriterROOT.cc:960
#16 0x00007f13ab224702 in DEventProcessor_ReactionFilter::evnt (this=<optimized out>, locEventLoop=0x7f138c0008c0, locEventNumber=<optimized out>) at plugins/Analysis/ReactionFilter/DEventProcessor_ReactionFilter.cc:74
#17 0x00000000012744ba in jana::JEventLoop::OneEvent (this=this
entry=0x7f138c0008c0) at src/JANA/JEventLoop.cc:693
#18 0x0000000001275454 in jana::JEventLoop::Loop (this=this
entry=0x7f138c0008c0) at src/JANA/JEventLoop.cc:496
#19 0x000000000124c70a in LaunchThread (arg=0x7ffc6cb6b750) at src/JANA/JApplication.cc:1382
#20 0x00007f13b3c32dd5 in start_thread () from /lib64/libpthread.so.0
#21 0x00007f13b2f39b3d in clone () from /lib64/libc.so.6

Thread 1 (Thread 0x7f13ba5d7200 (LWP 160853)):
#0  0x00007f13b3c39eed in nanosleep () from /lib64/libpthread.so.0
#1  0x0000000001255aad in jana::JApplication::Run (this=this
entry=0x7ffc6cb6b750, proc=proc
entry=0x238d720, Nthreads=<optimized out>, Nthreads
entry=0) at src/JANA/JApplication.cc:1613
#2  0x000000000063f1cc in main (narg=5, argv=0x7ffc6cb6bdb8) at programs/Analysis/hd_root/hd_root.cc:45
===========================================================

The lines below might hint at the cause of the crash.
You may get help by asking at the ROOT forum http://root.cern.ch/forum.
Only if you are really convinced it is a bug in ROOT then please submit a
report at http://root.cern.ch/bugs. Please post the ENTIRE stack trace
from above as an attachment in addition to anything else
that might help us fixing this issue.
===========================================================
[... 11 lines cut ...]
entry=0x7f138c12e580, d=std::vector of length 0, capacity 0) at /u/group/halld/Software/builds/Linux_RHEL7-x86_64-gcc4.8.5/jana/jana_0.7.9p1^ccdb166/Linux_RHEL7-x86_64-gcc4.8.5/include/JANA/JFactory.h:295
#13 0x00000000006914a4 in jana::JEventLoop::GetFromFactory<DAnalysisResults> (this=this
entry=0x7f138c0008c0, t=std::vector of length 0, capacity 0, tag=tag
entry=0x12cf8c7 "", data_source=
0x7f1393812190: 1553465728, allow_deftag=allow_deftag
entry=true) at /u/group/halld/Software/builds/Linux_RHEL7-x86_64-gcc4.8.5/jana/jana_0.7.9p1^ccdb166/Linux_RHEL7-x86_64-gcc4.8.5/include/JANA/JEventLoop.h:452
#14 0x0000000000691774 in jana::JEventLoop::Get<DAnalysisResults> (this=this
entry=0x7f138c0008c0, t=std::vector of length 0, capacity 0, tag=tag
entry=0x12cf8c7 "", allow_deftag=allow_deftag
entry=true) at /u/group/halld/Software/builds/Linux_RHEL7-x86_64-gcc4.8.5/jana/jana_0.7.9p1^ccdb166/Linux_RHEL7-x86_64-gcc4.8.5/include/JANA/JEventLoop.h:309
#15 0x0000000000709af0 in DEventWriterROOT::Fill_DataTrees (this=this
entry=0x7f136cd52ce0, locEventLoop=locEventLoop
entry=0x7f138c0008c0, locDReactionTag="ReactionFilter") at libraries/ANALYSIS/DEventWriterROOT.cc:960
#16 0x00007f13ab224702 in DEventProcessor_ReactionFilter::evnt (this=<optimized out>, locEventLoop=0x7f138c0008c0, locEventNumber=<optimized out>) at plugins/Analysis/ReactionFilter/DEventProcessor_ReactionFilter.cc:74
#17 0x00000000012744ba in jana::JEventLoop::OneEvent (this=this
entry=0x7f138c0008c0) at src/JANA/JEventLoop.cc:693
#18 0x0000000001275454 in jana::JEventLoop::Loop (this=this
entry=0x7f138c0008c0) at src/JANA/JEventLoop.cc:496
#19 0x000000000124c70a in LaunchThread (arg=0x7ffc6cb6b750) at src/JANA/JApplication.cc:1382
#20 0x00007f13b3c32dd5 in start_thread () from /lib64/libpthread.so.0
#21 0x00007f13b2f39b3d in clone () from /lib64/libc.so.6

===========================================================

JANA ERROR>> didn't sleep full 0.5 seconds!

sdobbs, 2019-04-23:
Thanks for submitting this issue.  I think I have an idea of what is causing it, am tracking it down...

#5<https://github.com/JeffersonLab/halld_recon/pull/5>
#6<https://github.com/JeffersonLab/halld_recon/pull/6> Calc_RFBunchShift (this=, locTimeToStepTo=<error reading variable: Cannot access memory at address 0x60>, locTimeToStep=17637937284.106651) at libraries/ANALYSIS/DSourceComboTimeHandler.h:302
#11<https://github.com/JeffersonLab/halld_recon/pull/11> 0x000000000069d43e in DAnalysisResults_factory::evnt (this=0x7f138c12e580, locEventLoop=0x7f138c0008c0, eventnumber=) at libraries/ANALYSIS/DAnalysisResults_factory.cc:282
#12<https://github.com/JeffersonLab/halld_recon/pull/12> 0x0000000000690943 in jana::JFactory::Get (this=this
#13<https://github.com/JeffersonLab/halld_recon/pull/13> 0x00000000006914a4 in jana::JEventLoop::GetFromFactory (this=this
#14<https://github.com/JeffersonLab/halld_recon/pull/14> 0x0000000000691774 in jana::JEventLoop::Get (this=this
#16<https://github.com/JeffersonLab/halld_recon/pull/16> 0x00007f13ab224702 in DEventProcessor_ReactionFilter::evnt (this=, locEventLoop=0x7f138c0008c0, locEventNumber=) at plugins/Analysis/ReactionFilter/DEventProcessor_ReactionFilter.cc:74

entry=0x238d720, Nthreads=, Nthreads

#6<https://github.com/JeffersonLab/halld_recon/pull/6> Calc_RFBunchShift (this=, locTimeToStepTo=<error reading variable: Cannot access memory at address 0x60>, locTimeToStep=17637937284.106651) at libraries/ANALYSIS/DSourceComboTimeHandler.h:302
#11<https://github.com/JeffersonLab/halld_recon/pull/11> 0x000000000069d43e in DAnalysisResults_factory::evnt (this=0x7f138c12e580, locEventLoop=0x7f138c0008c0, eventnumber=) at libraries/ANALYSIS/DAnalysisResults_factory.cc:282
#12<https://github.com/JeffersonLab/halld_recon/pull/12> 0x0000000000690943 in jana::JFactory::Get (this=this
#13<https://github.com/JeffersonLab/halld_recon/pull/13> 0x00000000006914a4 in jana::JEventLoop::GetFromFactory (this=this
#14<https://github.com/JeffersonLab/halld_recon/pull/14> 0x0000000000691774 in jana::JEventLoop::Get (this=this
#16<https://github.com/JeffersonLab/halld_recon/pull/16> 0x00007f13ab224702 in DEventProcessor_ReactionFilter::evnt (this=, locEventLoop=0x7f138c0008c0, locEventNumber=) at plugins/Analysis/ReactionFilter/DEventProcessor_ReactionFilter.cc:74

sdobbs, 2019-04-29:
The cause of this crash is that DSourceComboer only resets some of its internal memory management when the event number in the JEvent object actually changes, and the input files had two back-to-back "Event 1's".  

For reference, the important if-statement is in:
https://github.com/JeffersonLab/halld_recon/blob/de076695f16956fe649366b53959c7582285001f/src/libraries/ANALYSIS/DSourceComboer.cc#L1100

Since this is not well-formed input, and it's not clear how to easily handle this situation in the current framework, I am marking this as wontfix.

closed 2019-04-29 by sdobbs