JeffersonLab/halld_recon #166: Vertex z dependent reconstruction efficiency

aaust, 2019-05-23:
I generated pi+ pi- p events for t in [0,0.1] GeV2. For the events that make it through the analysis, the protons have up to 300MeV and are emitted at 73 degrees:
 (black thrown, red thrown for reconstructed, green reconstructed)

[image]
[image]

For these events, the z vertex distribution shows a hump between 75 and 80cm:
[image]

As this happens for both Geant3 and Geant4, I did not believe this is specific to the simulation part. As a matter of fact, I also observe a strange vertex z distribution for real data at low t (between 0.05 and 0.10):
[image]

rjones30, 2019-05-23:
Hello Alex,

Why do the red and green curves not have the same integral, if both are for
the same set of reconstructed protons?

-Richard

aaust, 2019-05-23:
@rjones30: the plugin monitoring_hists requires a minimal PID FOM of 5.73303E-7 to assign particle types to the tracks. This cut is applied to the green histograms for monitoring purposes.

rjones30, 2019-05-23:
Alex,

Is there any reason to suppose that the red histogram is representative of
what the green histogram should show, given that the same cut was not
applied to them? Or is the green not to be compared to the red? -Richard

aaust, 2019-05-23:
@rjones30: Yes, you can ignore the green histograms for now. The ratio between red and black gives you a crude reconstruction efficiency, that does not take resolution effects into account.