JeffersonLab/halld_recon #256: ReactionFilter plugin is not working with an (atomic) electron target

igjaegle, 2020-01-06:
For the following Reaction1 1_3__1_3 ie to analyze single Compton simulation, the ReactionFilter is not working

sdobbs, 2020-01-07:
Could you please point us to some samples of simulated events?

igjaegle, 2020-01-07:
Thank you! For some simu. here:

/work/halld/home/ijaegle/primex-d-pre-production-test/Geant4_bkg_BeamPhotons_runnb_61333_gen_compton_simple/hddm/

aaust, 2020-01-07:
I just ran through all your simulated REST files and I did not see a problem, except that no events were selected. Must be a problem with your simulation.

With a REST file for real data (e.g. /cache/halld/RunPeriod-2018-08/recon/ver02/REST/051040/dana_rest_051040_130.hddm), several events made it successfully into the tree.

sdobbs, 2020-01-07:
Yeah, looking through the REST files, I'm seeing some showers and an
occasional track.  So it's probably worth looking at the properties of
these objects and understanding why they are being rejected by the standard
analysis library.
I'd also try not using a kinematic fit to start out with.

aaust, 2020-01-07:
There is no magnetic field in the simulation, so the charges of the electrons seem to be defaulted to "+".

Also, you have to bypass the trigger emulation, since it was not implemented yet:
TRIG:BYPASS 1

igjaegle, 2020-01-07:
What is happening if the track fitter does not find the sign?

igjaegle, 2020-01-07:
in principal there should be no default sign

sdobbs, 2020-01-07:
If the solenoid off, we aren't able to determine the sign and momentum of
charged particles, so I think these are generally set to default values?
Simon would know better.

It might make sense to write a specialized track factory for this case.

---Sean

closed 2020-01-07 by igjaegle