JeffersonLab/halld_recon #374: Standardizing the approach to PID cuts

jrstevenjlab, 2020-05-13:
During a discussion at the [Collaboration Meeting](https://halldweb.jlab.org/wiki/index.php/GlueX-Collaboration-May-2020#Afternoon), we agreed that standardizing the Particle ID cuts in the current analysis framework was needed, which I'll try to summarize here.

For each [DChargedTrackHypothesis](https://github.com/JeffersonLab/halld_recon/blob/master/src/libraries/PID/DChargedTrackHypothesis.h) object the following information should be accessible:
- PID information from each detector comparing the measured quantity to the expected value for each mass hypothesis (e/pi/K/p).  In the current PID Figure-of-Merit (PIDFOM) scheme (Ref. [1](https://halldweb.jlab.org/doc-private/DocDB/ShowDocument?docid=4185), [2](https://halldweb.jlab.org/doc-private/DocDB/ShowDocument?docid=4457) are given by chi^2 variables with measured resolutions coming from the CCDB, but only for the intrinsic mass hypothesis of the object (not the alternate mass hypotheses).
- A combined probability or FOM (using all the available PID information) for each mass hypothesis (e/pi/K/p)

The goal is to provide the necessary PID variables in the standard Analysis TTree format for users to make cuts in the DSelector or equivalent flattened tree analysis.  Some options discussed include storing for each track in the TTree:
1.  Only the combined probability for each of the mass hypothesis.  This would allow users to make simple selections directly from the TTree in the DSelector or when creating a flattened tree.
2.  All the PID information from each detector (and each mass hypothesis), where a simple function would be provided to combine these inputs into the combined probability value in option 1).  This would allow users to further use the detector level PID information for more sophisticated analysis such as Machine Learning classifiers, etc.
3.  Both 1) and 2) to allow simple 

The first step would be to determine if this is the correct strategy for standardizing the PID selections.  Once we've done, that we should discuss the implementation and usage (which will require coordination with updates to the gluex_root_analysis package).  

Comments/corrections/discussion/questions are more than welcome.

sdobbs, 2020-05-13:
For the halld_recon modifications, we do have the [DChargedTrack](https://github.com/JeffersonLab/halld_recon/blob/master/src/libraries/PID/DChargedTrack.h) object which collects the various DChargedTrackHypothesis objects.  Maybe this is the location to add additional interfaces to get the PID probabilities for different mass hypotheses?

The DChargedTrackHypothesis format that's written to the Analysis TTree would still need to be extended to write out some of the PID information for each hypothesis.  I would be in favorite of having a default and verbose option for these variables.

mashephe, 2020-05-14:
Just a note from experience with past experiments:  I don't recall ever seeing PID the chi^2 (or probability) in an absolute sense being used in an analysis.  It is most useful when one compares it to alternates.  One typically wants to know how much more likely kaon is than pion or proton.

The reason is evident if one thinks about any of the classic PID plots, e.g. dE/dx at fixed momentum or Cherenkov angle.  If one does pi/K separation, in a simple case, there are two overlapping Gaussians:  one for true pions and one for true kaons.  If you are trying to select a kaon then the absolute chi^2 or probability tells you how close you are to the average kaon, but it doesn't tell you whether you are on the "pion side" of the peak or not.  You need to compare the difference between the pion and kaon chi^2 to get this info.  It would be good to keep kaons that are in the tails, but not on the pion side of the peak.

I would recommend avoiding at "combo time", any algorithm that has some default where pion (or kaon or proton) tracks are only limited those tracks where the relevant hypothesis is most likely.  There will be ambiguities in PID (dE/dx bands cross) and we'll simply be randomly throwing away candidates there.  For such ambiguous cases at the track level, the kinematic fit is helpful in resolving ambiguities (but they have to make it to the fitter).

(I may be partially restating what you all know or are advocating for.)

Matt