JeffersonLab/halld_recon #378: Issues with reconstructing backward going K+ tracks around 120 deg

niwgit, 2020-05-19:
Do we have a volunteer to peel off the backwards track reconstruction/hit matching part of the discussion to a halld_recon issue?

_Originally posted by @sdobbs in https://github.com/JeffersonLab/HDGeant4/issues/137#issuecomment-627657604_

sdobbs, 2020-05-19:
Here is what I see when I look at slow backward K+ tracks in G4, G3(HADR=1) and G3(HADR=4). The output is just what monitoring_hists saves for K+ under Hist_DetectorParticleKinematics. I can remake these histograms with higher resolution if desired, but I think these are adequate.

- K+ particle gun, 0.25 < p < 0.75 GeV, 80 < theta < 140 degrees, run 71500, 1M events each
[image]
[image]
[image]

Here are the x and y projections of these plots, shown for comparison in the same plot.
[image]
[image]

It is important to have both G3 results in front of us when we discuss this, otherwise we are apt to suppose that any differences we are due to things like tracking and/or hits collection that are being done differently in G3 and G4. The two G3 simulations have exactly the same algorithms in play for tracking and hits collection, so those differences arise entirely from within the hadronic models.

_Originally posted by @rjones30 in https://github.com/JeffersonLab/HDGeant4/issues/137#issuecomment-624082997_

sdobbs, 2020-05-19:
Note that the studies from the previous post should be redone with PRESELECT:HAS_DETECTOR_MATCH_FLAG=0 in order to remove possible effects with matching to other detectors.

sdobbs, 2020-05-19:
It seems that this cutoff around 120degrees might be related to tracking at backward angles. Here is another plot from monitoring_hists that suggests this is happening.
[image]
I looked in hdview2 at several events where I generated 0.6 GeV/c k+ tracks at 125o. These tracks looked fine visually at the DTrackWireBased and DTrackTimeBased level in hdview with >20 straw hits assigned to the track, but they had very low FOM values for all of the mass hypotheses, including k+, and no DTrackTimeBased:Combos were created.

I will report this to the tracking experts and ask them to look into this. If there is a critical angle where the tracking is breaking down then it is possible that small differences between G3 and G4 are being amplified by the sharpness of the cutoff of the acceptance in the CDC. If so then there might also be a big difference between G3(HADR=1) and G3(HADR=4).

_Originally posted by @rjones30 in https://github.com/JeffersonLab/HDGeant4/issues/137#issuecomment-624596170_

sdobbs, 2020-05-19:
I took the k+ generated kinematics shared by Nilanga and generated 3 samples of 1M events each, one with hdgeant4, one with hdgeant(HADR=1) and one with hdgeant(HADR=4). Here is the reconstructed p vs theta 2d plot from the hdgeant4 sample.
[image]
All three of these plots look the same. The x and y projections are more useful, since this is really a 1d distribution.
[image]
[image]
The dip structure is seen in all three samples, with slight differences between them, but nothing that would make G4 stand out from G3. In fact, G4 seems to be intermediate between the two G3 models. I see nothing here to suggest a problem with the simulation. There are probably issues with the PID at backward angles that are not well understood because u-channel processes have not been looked at much so far in GlueX.

_Originally posted by @rjones30 in https://github.com/JeffersonLab/HDGeant4/issues/137#issuecomment-625148613_

sdobbs, 2020-05-19:
To rule out the possibility that the blob structure might be coming from a bug in the geometry that affects all 3 simulations, I produced REST output from the earlier particle gun samples and plot all of the time-based tracks that are reconstructed as kaons. There is no FOM cut here. The first plot is generated, followed by reconstructed G4, G3(HADR=1), and G3(HADR=4). These plots suggest that the tracking is working just fine. I conclude from this that the blobs are coming from some combination of timing chisquare cuts and other PID algorithm issues, eg. competition between k+ and proton hypotheses leaning the wrong way in certain regions. 
[image]
[image]
[image]
[image]

_Originally posted by @rjones30 in https://github.com/JeffersonLab/HDGeant4/issues/137#issuecomment-625154222_

sdobbs, 2020-05-19:
Here are the monitoring_plots histograms for the 8 samples that I have run so far, for easy reference. The studies are:
0: k- particle gun, 0.45<p<0.55 GeV/c, 80o<𝜃<100o, 𝜙=180o
1: k- particle gun, 0.10<p<0.90 GeV/c, 80o<𝜃<100o, 𝜙=180o
2: k- particle gun, 0.25<p<0.85 GeV/c, 5o<𝜃<75o, all 𝜙
3: k- spectrum from recoil Y* decays, generated by Peter Pauli
4: k- particle gun, 0.25<p<0.85 GeV/c, 5o<𝜃<75o, all 𝜙, with CUTS enabled in G4
5: k- particle gun, 0.25<p<0.85 GeV/c, 5o<𝜃<75o, all 𝜙, with CUTS, with G4 fix in place
6:  k+ particle gun, 0.25 < p < 0.75 GeV/c and 80o < 𝜃 < 140o, no CUTS
7: k+ spectrum from u-channel Y* reaction, generated from Nilanga's genr8 spec, no CUTS

Each of these was studied in 3 different runs, 1M events each, tagged by its own suffix.
* hdgeant4: suffix=""
* hdgeant (HADR=1): suffix="3"
* hdgeant (HADR=4): suffix="34"

The monitoring_hists output files are available over https at:
https://zeus.phys.uconn.edu/halld/kstudy-4-2020/kstudyrec${suffix}_${sample}xx.root

_Originally posted by @rjones30 in https://github.com/JeffersonLab/HDGeant4/issues/137#issuecomment-625222017_

niwgit, 2020-05-29:
Sean suggested to look at the monitoring_hists plots of K+ Hist_DetectorParticleKinematics with the use of TRIGGER:FORCE_PHYSICS_TRIGGER=1 and with and without PRESELECT:HAS_DETECTOR_MATCH_FLAG=0.

I made these plots from Richard's MC sample of 1M events for k+ particle gun, 0.25 < p < 0.75 GeV/c and 80o < 𝜃 < 140o, no CUTS for G4.
I don't see much difference when comparing the plots.
-Nilanga

[image]

[image]

[image]

[image]

[image]

[image]

sdobbs, 2020-05-29:
For what it's worth, the TRIGGER tag in Nilanga's analysis is something I put in to a private branch of mine because I forgot TRIG:BYPASS existed.

In any case, it looks like this dip is probably due to tracking not detector matching.

rjones30, 2020-05-29:
Sean, good to know. In my earlier study, I rebuilt my own private copy of
the monitoring_hists plugin with the trigger cut disabled, most likely
equivalent to the methods you listed. This TRIG:BYPASS is something to
remember.
-Richard

niwgit, 2020-06-01:
Richard's previous plots of p vs theta without FOM cut are interesting. I see two FOM defined in DChargedTrack.h - Get_BestFOM and Get_BestTrackingFOM. The cut that hasn't been applied, is for these variables or something else?
(https://www.jlab.org/Hall-D/software/HDSoftware_Documentation/_d_charged_track_8h_source.html#l00086)

Any suggestion for moving forward with this study is appreciated.
-Nilanga

sdobbs, 2020-06-05:
There are a couple cuts to check out in DHistogramAction_DetectedParticleKinematics::Perform_Action():

1. For a particular charged track, only the best hypothesis (out of e,pi,K,p) is considered:
https://github.com/JeffersonLab/halld_recon/blob/23e7c2d0cf2a73bc23c6ea75c93e5d8b50ad9adf/src/libraries/ANALYSIS/DHistogramActions_Independent.cc#L3023

But each track has all the hypotheses fitted.  Maybe in this corner, sometimes the "right" hypothesis is not always the "best"

It seems worth trying all the combinations.

2. A very loose requirement that the track is positively identified [PID FOM > 5.73303E-7]:  https://github.com/JeffersonLab/halld_recon/blob/23e7c2d0cf2a73bc23c6ea75c93e5d8b50ad9adf/src/libraries/ANALYSIS/DHistogramActions_Independent.cc#L3032

It seems worth trying to remove this requirement.

Unfortunately, it seems like the way to do these studies is either to hack your own version of halld_recon or write your own plugin to make these histograms.

niwgit, 2020-06-13:
Thanks Sean for the suggestions.
I modified the code in DHistogramAction_DetectedParticleKinematics::Perform_Action(): in a local version of halld_recon to consider all positive charged hypotheses for a charged track. PID FOM cut was kept as default.

Here are the histograms now. 
TRIG:BYPASS = 1 and PRESELECT:HAS_DETECTOR_MATCH_FLAG=0 flags were also used.
-Nilanga

[image]

[image]

[image]

niwgit, 2020-06-20:
Here's the p vs theta plot for only selecting the K+ hypothesis and removing the PID FOM cut.
This seems similar to Richard's plot, but here the PRESELECT:HAS_DETECTOR_MATCH_FLAG=0 has also been used.
-Nilanga

[image]

niwgit, 2020-06-25:
These histograms are with the best hypothesis selected and the PID FOM cut applied.
It looks like the top right hole in K+ get filled as the pi+ hypothesis.
-Nilanga

[image]
[image]
[image]

zihlmann, 2020-06-25:
> KPlus_p_vs_theta
> PiPlus_p_vs_theta
> Proton_p_vs_theta

This does seem to say even more:
a) the "pion"-choice takes away k+'es at higher momenta around 0.7 GeV 
and peak at these momenta at high angles
b) the "proton"-choice takes away k+'es at low momenta around 0.35 GeV 
peaking around 112 degrees.

curtisameyer, 2020-06-25:
As Beni notes, this is pretty telling.

Curtis

—

Curtis A. Meyer | MCS Associate Dean for Research
The Otto Stern Professor of Physics
cmeyer@cmu.edu<mailto:cmeyer@cmu.edu> | 412 268 2745
[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]
Carnegie Mellon University
Department of Physics, Wean Hall 8414
5000 Forbes Ave., Pittsburgh PA, 15213
www.curtismeyer.com<http://www.curtismeyer.com>
[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]
[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]

niwgit, 2020-07-02:
When I use Get_BestTrackingFOM() instead of Get_BestFOM() to select the hypothesis the histograms for the three 
hypotheses look very different from the previous plots.
-Nilanga

[image]
[image]
[image]

niwgit, 2020-07-23:
I looked at the FOM values of the K+, pi+ and proton hypotheses that get selected as the best hypothesis with the
Get_BestFOM() and the default minimum PID FOM cut. Also I only selected events in the region 120 < theta < 140 deg and 
0.4 < momentum < 0.8 GeV/c.
It looks like large fraction of K+ and proton hypotheses has very small FOM values.
-Nilanga

[image]
[image]
[image]