JeffersonLab/halld_recon #387: Data corruption in 2019-11 data

sdobbs, 2020-06-01:
In a recent calibration run, I found the following errors:  (first one is most concerning)

run 71757, file 001: 
JANA ERROR>>Slot from SSP/DIRC event header does not match slot from last block header (18 != 3)

run 71912, file 001:
corrupted on cache, submitted 

run 72188, file 001:
segfault as listed below (currently trying to confirm that it's reproducable):

===========================================================
#6  0x00002b6f764b52ad in ?? () from /lib64/libstdc++.so.6
#7  0x00000000006e34dc in ~basic_string (this=0x2b6faef42fa0, __in_chrg=<optimized out>) at /usr/include/c++/4.8.2/bits/basic_string.h:539
#8  jana::JEventLoop::GetFromFactory<DParticleID> (this=0x2b6fd00008c0, t=std::vector of length -647603552683825760, capacity 550456894365317400 = {...}, tag=0x1138e0d "", data_source=
0x2b6faef430b0: jana::JEventLoop::DATA_NOT_AVAILABLE, allow_deftag=<optimized out>) at /u/group/halld/Software/builds/Linux_CentOS7.7-x86_64-gcc4.8.5/jana/jana_0.7.9p1^ccdb166/Linux_CentOS7.7-x86_64-gcc4.8.5/include/JANA/JEventLoop.h:456
#9  0x0000000000000001 in ?? ()
#10 0x00000000006e3734 in jana::JEventLoop::Get<DParticleID> (this=0x2b6faef43040, this
entry=0x2b6fd00008c0, t=std::vector of length 0, capacity 0, tag=0x2b6faef43080 "8003vvo+", tag
entry=0x1138e0d "", allow_deftag=allow_deftag
entry=true) at /u/group/halld/Software/builds/Linux_CentOS7.7-x86_64-gcc4.8.5/jana/jana_0.7.9p1^ccdb166/Linux_CentOS7.7-x86_64-gcc4.8.5/include/JANA/JEventLoop.h:309
#11 0x00000000006fa1e5 in jana::JEventLoop::GetSingle<DParticleID> (this=this
entry=0x2b6fd00008c0, t=
0x2b6faef431b0: 0x0, tag=tag
entry=0x1138e0d "", exception_if_not_one=exception_if_not_one
entry=true) at /u/group/halld/Software/builds/Linux_CentOS7.7-x86_64-gcc4.8.5/jana/jana_0.7.9p1^ccdb166/Linux_CentOS7.7-x86_64-gcc4.8.5/include/JANA/JEventLoop.h:239
#12 0x0000000000772717 in DHistogramAction_EventVertex::Perform_Action (this=0x3b8e6f8, locEventLoop=0x2b6fd00008c0, locParticleCombo=<optimized out>) at libraries/ANALYSIS/DHistogramActions_Independent.cc:2770
#13 0x00002b6f90fa07ce in operator() (locEventLoop=0x2b6fd00008c0, this=0x3b8e6f8) at libraries/ANALYSIS/DAnalysisAction.h:125
#14 DEventProcessor_monitoring_hists::evnt (this=0x3b8cb30, locEventLoop=0x2b6fd00008c0, eventnumber=<optimized out>) at plugins/Analysis/monitoring_hists/DEventProcessor_monitoring_hists.cc:130
#15 0x00000000010d8efa in jana::JEventLoop::OneEvent (this=this
entry=0x2b6fd00008c0) at src/JANA/JEventLoop.cc:693
#16 0x00000000010d9e94 in jana::JEventLoop::Loop (this=this
entry=0x2b6fd00008c0) at src/JANA/JEventLoop.cc:496
#17 0x00000000010b114a in LaunchThread (arg=0x7ffe98bb8b70) at src/JANA/JApplication.cc:1382
#18 0x00002b6f76041e65 in start_thread () from /lib64/libpthread.so.0
#19 0x00002b6f76d7788d in clone () from /lib64/libc.so.6

===========================================================

zihlmann, 2020-06-01:
> jana::JEventLoop::GetFromFactory (this=0x2b6fd00008c0, t=std::vector
> 0x00000000006e3734 in jana::JEventLoop::Get (this=0x2b6faef43040, this
> 0x00000000006fa1e5 in jana::JEventLoop::GetSingle (this=this
> (this=0x3b8e6f8, locEventLoop=0x2b6fd00008c0, locParticleCombo=) at
> locEventLoop=0x2b6fd00008c0, eventnumber=) at

which event is that?
with hd_dump I see not issues in the first 30 events.

zihlmann, 2020-06-01:
hdview2 crashes because it looks for TOF tables in the wrong directory 
(TOF instead of TOF2)!

JANA >>In DTOFHit_factory, loading constants...
TOF: USE WALK CORRECTION TYPE 4
Error [1270]: in [MySQLDataProvider::GetAssignmentShort(int, const 
string&, time_t, const string&)] No data was selected. Table 
'/TOF/timewalk_parms_5PAR' for run='71757', timestampt='0' and 
variation='default'
JANA >>Error loading TOF/timewalk_parms_5PAR !
Error [1270]: in [MySQLDataProvider::GetAssignmentShort(int, const 
string&, time_t, const string&)] No data was selected. Table 
'/TOF/timing_offsets_5PAR' for run='71757', timestampt='0' and 
variation='default'
JANA >>Error loading TOF/timing_offsets_5PAR !

sdobbs, 2020-06-01:
It’s pretty far in. Still need to extract the actual number

zihlmann, 2020-06-01:
something changed with ccdb:

on Friday I run run 70969 and got this output at the start:
TOF: USE WALK CORRECTION TYPE 4

and just now doing it again I get:
TOF: USE WALK CORRECTION TYPE 3

this is just wrong! I always use:
CCDB_CONNECTION=mysql://ccdb_user@hallddb.jlab.org/ccdb

sdobbs, 2020-06-01:
> which event is that? with hd_dump I see not issues in the first 30 events.
> […](#)

It's within the first 250 events, though it doesn't seem to actually cause the crash.

sdobbs, 2020-06-01:
Beni, I don't see a problem with the walk correction type, but do you see the following lines when running hdview2:

JANA >>Generated via: JCalibration using CCDB for MySQL and SQLite databases
JANA >>Run:9999
JANA >>URL: mysql://ccdb_user@hallddb.jlab.org/ccdb
JANA >>context: default
JANA >>comment: Default constants for analyzing data
JANA >>Creating DGeometry:
JANA >>  Run requested:9999  found:9999
JANA >>  Run validity range: 9999-9999
JANA >>  URL="ccdb:///GEOMETRY/main_HDDS.xml"  context="default"
JANA >>  Type="JGeometryXML"
JANA >>Found 25 material maps in calib. DB
JANA >>Read in 25 material maps for run 9999 containing 76153 grid points total

zihlmann, 2020-06-01:
yes, see the the output below, I get still the wrong TOF directory!

JANA >>Opening source 
"/cache/halld/RunPeriod-2019-11/rawdata/Run070969/hd_rawdata_070969_002.evio" 
of type: EVIOpp  - Reads EVIO formatted data from file or ET system
JANA >>Created JCalibration object of type: JCalibrationCCDB
JANA >>Generated via: JCalibration using CCDB for MySQL and SQLite databases
JANA >>Run:9999
JANA >>URL: mysql://ccdb_user@hallddb.jlab.org/ccdb
JANA >>context: default
JANA >>comment: Default constants for analyzing data
loading VERSION 3
JANA >>Creating DGeometry:
JANA >>  Run requested:9999  found:9999
JANA >>  Run validity range: 9999-9999
JANA >>  URL="ccdb:///GEOMETRY/main_HDDS.xml" context="default"
JANA >>  Type="JGeometryXML"
JANA >>Found 25 material maps in calib. DB
JANA >>Read in 25 material maps for run 9999 containing 76153 grid 
points total
src/JANA/JGeometryXML.cc:348 Node or attribute not found for xpath 
"//composition[@name='forwardTOF_bottom3']/mposY[@volume='FTOL']/@ncopy".
src/JANA/JGeometryXML.cc:348 Node or attribute not found for xpath 
"//composition[@name='forwardTOF_top3']/mposY[@volume='FTOL']/@ncopy".
src/JANA/JGeometryXML.cc:348 Node or attribute not found for xpath 
"//composition[@name='forwardTOF_bottom3']/mposY[@volume='FTOL']/@ncopy".
src/JANA/JGeometryXML.cc:348 Node or attribute not found for xpath 
"//composition[@name='forwardTOF_top3']/mposY[@volume='FTOL']/@ncopy".
Reading preferences from "/home/zihlmann/.hdview2" ...
JANA >>Created JCalibration object of type: JCalibrationCCDB
JANA >>Generated via: JCalibration using CCDB for MySQL and SQLite databases
JANA >>Run:70969
JANA >>URL: mysql://ccdb_user@hallddb.jlab.org/ccdb
JANA >>context: default
JANA >>comment: Default constants for analyzing data
JANA >>Reading Magnetic field map from 
Magnets/Solenoid/solenoid_1350A_poisson_20160222 ...
  Nx=221 Ny=1 Nz=701 )  at 0x3c9a990
Reading fine-mesh B-field data from 
/group/halld/www/halldweb/html/resources/Magnets/Solenoid/finemeshes/solenoid_1350A_poisson_20160222
  rmin: 0 rmax: 88.5 dr: 0.1 zmin: 0 zmax: 600 dz: 0.1
  Number of points in z = 6000
  Number of points in r = 885
JANA >>154921 entries found (Created Magnetic field map of type 
DMagneticFieldMapFineMesh
JANA >>Creating DTranslationTable for run 70969
JANA >>Reading translation table from calib DB: Translation/DAQ2detector ...
JANA >>39752 channels defined in translation table
----------- New Event 0  (run 70969) -------------
JANA >>In DSCHit_factory, loading constants...
src/JANA/JGeometryXML.cc:348 Node or attribute not found for xpath 
"//composition[@name='forwardTOF_bottom3']/mposY[@volume='FTOL']/@ncopy".
src/JANA/JGeometryXML.cc:348 Node or attribute not found for xpath 
"//composition[@name='forwardTOF_top3']/mposY[@volume='FTOL']/@ncopy".
src/JANA/JGeometryXML.cc:348 Node or attribute not found for xpath 
"//composition[@name='forwardTOF_bottom3']/mposY[@volume='FTOL']/@ncopy".
src/JANA/JGeometryXML.cc:348 Node or attribute not found for xpath 
"//composition[@name='forwardTOF_top3']/mposY[@volume='FTOL']/@ncopy".
Created TGeoManager :0xb434a80
src/JANA/JGeometryXML.cc:348 Node or attribute not found for xpath 
"//section/composition/posXYZ[@volume='DIRC']/@X_Y_Z".
libraries/HDGEOMETRY/DGeometry.cc:1843 Unable to retrieve DIRC position.
src/JANA/JGeometryXML.cc:348 Node or attribute not found for xpath 
"//section/composition/posXYZ[@volume='TRDGEM']/@X_Y_Z".
libraries/HDGEOMETRY/DGeometry.cc:1869 Unable to retrieve TRD position.
JANA >> Beam spot: x=0.200005 y=-0.017351 z=65 dx/dz=-0.000684869 
dy/dz=0.000356484
JANA >>vertex constraint: <none>
JANA >>In DTOFHit_factory, loading constants...
TOF: USE WALK CORRECTION TYPE 3
JANA >>Reading DIRC LUT TTree from 
/group/halld/www/halldweb/html/resources//DIRC/LUT/lut_production_2020_v01.root 
...
JANA >>In DCDCHit_factory, loading constants...
JANA >>In DFDCHit_factory, loading constants...
src/JANA/JGeometryXML.cc:348 Node or attribute not found for xpath 
"//section/composition/posXYZ[@volume='DIRC']/@X_Y_Z".
libraries/HDGEOMETRY/DGeometry.cc:1843 Unable to retrieve DIRC position.
JANA >>In DBCALHit_factory, loading constants...
JANA >>In DFCALHit_factory, loading constants...
libraries/HDGEOMETRY/DGeometry.cc:1812 FCAL position: 
(x,y,z)=(0.529,-0.002,624.906)
Error [1270]: in [MySQLDataProvider::GetAssignmentShort(int, const 
string&, time_t, const string&)] No data was selected. Table 
'/FCAL/energy_dependence_correction_vs_ring' for run='70969', 
timestampt='0' and variation='default'

aaust, 2020-06-02:
Look at the event number. It's 9999!

sdobbs, 2020-06-02:
The run number, but yeah, this is a known issue with how certain parts of the application are initialized.  I'm working on a workaround...

sdobbs, 2020-06-02:
Note that you can specify a particular run number, like (in this case) -PRUNNUMBER=70969

zihlmann, 2020-06-02:
this 9999 stuff was not present when running on Firday.
today I updated to  the latest code and now get this 9999 crap.

sdobbs, 2020-06-02:
For run 71757, file 001, after ~1.5k events I get the following stack trace (clearly not 100% reproducible):

===========================================================
#6  DEVIOWorkerThread::ParseSSPBank (this=this
entry=0x7f3734004820, rocid=rocid
entry=92, iptr=
0x7f37317f32f8: 0x7f36bc00e324, iend=iend
entry=0x7f36bc016684) at /usr/include/c++/4.8.2/bits/stl_vector.h:734
#7  0x0000000000f7b67c in DEVIOWorkerThread::ParseDataBank (this=this
entry=0x7f3734004820, iptr=
0x7f37317f32f8: 0x7f36bc00e324, iend=iend
entry=0x7f36bc016684) at libraries/DAQ/DEVIOWorkerThread.cc:1009
#8  0x0000000000f7b917 in DEVIOWorkerThread::ParsePhysicsBank (this=this
entry=0x7f3734004820, iptr=
0x7f37317f32f8: 0x7f36bc00e324, iend=iend
entry=0x7f36bc089af8) at libraries/DAQ/DEVIOWorkerThread.cc:757
#9  0x0000000000f7d1e0 in DEVIOWorkerThread::ParseBank (this=this
entry=0x7f3734004820) at libraries/DAQ/DEVIOWorkerThread.cc:352
#10 0x0000000000f7e098 in DEVIOWorkerThread::MakeEvents (this=this
entry=0x7f3734004820) at libraries/DAQ/DEVIOWorkerThread.cc:271
#11 0x0000000000f7f74a in DEVIOWorkerThread::Run (this=0x7f3734004820) at libraries/DAQ/DEVIOWorkerThread.cc:111
#12 0x00007f37547f8070 in ?? () from /lib64/libstdc++.so.6
#13 0x00007f3754c55e65 in start_thread () from /lib64/libpthread.so.0
#14 0x00007f3753f5b88d in clone () from /lib64/libc.so.6

===========================================================

zihlmann, 2020-06-02:
ignore my comments above, they are not related to the issue at hand:
I do see the same error message when using hd_root and the plugin monitoring_hists
however hd_root does not crash:

here the error message as shown already at the start of the issue:

JANA ERROR>>Slot from SSP/DIRC event header does not match slot from last block header (18 != 3)
ERROR: request for channel number of inactive block!  row 28 column 28     

however hd_root does not crash after the message and continues past event 1.5k events
using the plugin monitoring_hists.
this is with file /cache/halld/RunPeriod-2019-11/rawdata/Run071757/hd_rawdata_071757_001.evio

hd_root was compiled using the latest branch content on github

sdobbs, 2020-06-02:
Yeah I tried 4 times interactively and got the crash shown above once out
of these four times.

jrstevenjlab, 2020-06-02:
Looking back at the monitoring launches: this SSP message appeared for the same file during the ver01, ver08 and ver12 monitoring launches.  In all three cases the monitoring histograms were produced and from a first look I don't see problems with the DIRC data in that file or the file directly after it (i.e. hits are correlated in time with other detectors/tracks and the photon yield is roughly as expected).

zihlmann, 2020-06-02:
running 5 times single threaded and 5 times multi threaded(6) and I do NOT see a crash.
always running way past 1.5k events

the two errors
JANA ERROR>>Slot from SSP/DIRC event header does not match slot from last block header (18 != 3)
ERROR: request for channel number of inactive block! row 28 column 28
Are NOT from the same event. The first error is related to event 177 or around that number
while the second error is related to event 1136 or a little later.
the event numbers are most likely a littler higher because the reader is always a little ahead
of the analyzer.

sdobbs, 2020-06-02:
It's good to hear that it doesn't look like there's data corruption - probably it's just some bug in the parser?

I don't keep logs for each calibration launch, but it looks like this run fails more often than not.  A typical set of plugins is:  HLDetectorTiming,monitoring_hists,PSPair_online,RF_online,TAGM_TW,CDC_amp,CDC_TimeToDistance,CDC_dedx

In any case, I'll mark this as low priority.

zihlmann, 2020-06-02:
I finally see the crash too. However I see two different ways it crashes
1) just dies with "segmentation fault" no outer output
2) it crashes with lots of info: (note: 2 times it was event 336, this time it is 358)
JANA ERROR>> didn't sleep full 0.5 seconds!d)  16.0Hz  (avg.: 20.2Hz)     
  358.0 events processed  (368.0 events read)  30.0Hz  (avg.: 21.0Hz)     

===========================================================
There was a crash.
This is the entire stack trace of all threads:
===========================================================

Thread 12 (Thread 0x7f0e9770a700 (LWP 137181)):
#0  0x00007f0ecfaba9f5 in pthread_cond_wait

GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x00000000010be680 in jana::JApplication::EventBufferThread (this=this
entry=0x7ffda16a9440) at src/JANA/JApplication.cc:742
#2  0x00000000010be949 in LaunchEventBufferThread (arg=0x7ffda16a9440) at src/JANA/JApplication.cc:666
#3  0x00007f0ecfab6e65 in start_thread () from /lib64/libpthread.so.0
#4  ......

sdobbs, 2020-06-30:
From monitoring version 14:  Run 71561, file 004

libraries/DAQ/DEVIOWorkerThread.cc:357 Unknown outer EVIO bank tag: c4a7                                                                                                                          
JANA ERROR>>                                                                                                                                                                                      
JANA ERROR>>?JException:    code = 0    text = WARNING: unknown bank type (0x98)                                                                                                                  
JANA ERROR>>                                                                                                                                                                                      
JANA ERROR>>File: libraries/DAQ/swap_bank.cc,   Line: 91                                                                                                                                          
JANA ERROR>>                                                                                                                                                                                      
JANA ERROR>>?JException:    code = 0    text = WARNING: unknown bank type (0x91)                                                                                                                  
JANA ERROR>>                                                                                                                                                                                      
JANA ERROR>>File: libraries/DAQ/swap_bank.cc,   Line: 91                                                                                                                                          
Exit code: -1