JeffersonLab/halld_recon #391: Runs to update accidental scaling factors

sdobbs, 2020-06-02:
These (short) runs need to have values entered for the following runs - 

2017-01:
30980
30981
30982
31049
31050

2018-01:
41831
41832
41833
41834
41835
41836
42187

2018-08:
50941
51072
51384
51386
51387
51388
51389
51390
51391
51392
51393
51394
51395
51396
51397
51398
51399
51400
51401
51402
51408
51409
51410
51427
51431
51432
51433
51434
51435
51436
51437
51438
51440
51441
51442
51443
51444
51445
51446
51447
51448
51449
51450
51451
51453
51454
51455
51456
51457
51637
51683
51685
51686
51687
51725
51726
51727
51728
51729
51730
51731
51732
51733
51734
51735
51748
51749
51764
51765
51766
51767

sdobbs, 2020-06-15:
Hi @zihlmann - just checking in, do you need any data or anything else to update values for these runs?  Or is it just finding a reliable method for these low-statistics runs?

zihlmann, 2020-06-15:
Neither of those. I am in discussions with Richard about the proper 
approach and the results I see. Because
some of the differences I see between microscope and hodoscope in 
particular for the spring 2019 period
are very large and seem to contradict our understanding of the source of 
the accidental scaling. I think we first have
to resolve these inconsistencies before we update the ccdb with final 
results.

rjones30, 2020-06-15:
Sean,

Yes, Beni and I met last week and discussed a path forward on this very
issue. We are working on it, and will report back to you with our progress.

-Richard

sdobbs, 2020-07-28:
The CCDB has been updated with preliminary values for 2018-01 and 2018-08.  It still needs to be updated for 2017-01.