JeffersonLab/halld_recon #439: Problem with analysis library on MC

aaust, 2020-08-20:
Nilanga alerted me about a crash when running the analysis library version analysis-2017_01-ver40.xml over his MC sample. The crash does not happen with analysis-2017_01-ver39.xml. 
To my knowledge, the only major change was the fix to the IsGenerator flag in PR #417. I successfully tested this fix in version set analysis-2017_01-ver40.xml to generate rho MC. Nilanga used the exact same reconstruction version (recon-2017_01-ver03_18.xml), but the analysis step fails. I reproduced the problem with his dana_rest files and the current master with this command:
```
hd_root --config=/group/halld/data_monitoring/run_conditions/RunPeriod-2017-01/jana_analysis_2017-01_ver41.config /cache/halld/gluex_simulations/REQUESTED_MC/Lambda_1520_Sigma_pi0_new_20200810062243pm/hddm/dana_rest_genr8_030808_000.hddm
```
It stops after processing 16 events and the error message reads:
```
terminate called after throwing an instance of 'unsigned long'
Aborted
```

aaust, 2020-08-20:
Using analysis-2017_01-ver40.xml instead of the current master changes the error message, but not the outcome. The stack trace may have more information about the problem:
```
===========================================================
#6  mass (this=0x0) at libraries/PID/DKinematicData.h:44
#7  energy (this=0x0) at libraries/PID/DKinematicData.h:60
#8  DEventWriterROOT::Fill_BeamData (this=this
entry=0x7f7fd1c73260, locTreeFillData=locTreeFillData
entry=0x7f7fd1def9a0, locArrayIndex=locArrayIndex
entry=0, locBeamPhoton=0x7f7fd1bdc260, locVertex=locVertex
entry=0x7f7fd1c51eb0, locMCThrownMatching=locMCThrownMatching
entry=0x7f7fc1df4420) at libraries/ANALYSIS/DEventWriterROOT.cc:1529
#9  0x0000000000734019 in DEventWriterROOT::Fill_DataTree (this=this
entry=0x7f7fd1c73260, locEventLoop=locEventLoop
entry=0x7f803c0008c0, locReaction=0x7f7fd1d91a50, locParticleCombos=std::deque with 1 elements = {...}) at libraries/ANALYSIS/DEventWriterROOT.cc:1201
#10 0x000000000073577e in DEventWriterROOT::Fill_DataTrees (this=this
entry=0x7f7fd1c73260, locEventLoop=locEventLoop
entry=0x7f803c0008c0, locDReactionTag="ReactionFilter") at libraries/ANALYSIS/DEventWriterROOT.cc:1026
#11 0x00007f80661d9e12 in DEventProcessor_ReactionFilter::evnt (this=<optimized out>, locEventLoop=0x7f803c0008c0, locEventNumber=<optimized out>) at plugins/Analysis/ReactionFilter/DEventProcessor_ReactionFilter.cc:75
#12 0x000000000110054a in jana::JEventLoop::OneEvent (this=this
entry=0x7f803c0008c0) at src/JANA/JEventLoop.cc:693
#13 0x00000000011014e4 in jana::JEventLoop::Loop (this=this
entry=0x7f803c0008c0) at src/JANA/JEventLoop.cc:496
#14 0x00000000010d85ca in LaunchThread (arg=0x7ffc6239d120) at src/JANA/JApplication.cc:1382
#15 0x00007f806e79fe65 in start_thread () from /lib64/libpthread.so.0
#16 0x00007f806daa588d in clone () from /lib64/libc.so.6

===========================================================
```

sdobbs, 2020-08-20:
Looks like the crash comes from

  Bool_t locIsGeneratorFlag = (locBeamPhotonTaggedMCGEN->energy() ==
locBeamPhoton->energy() && locBeamPhotonTaggedMCGEN->dCounter ==
locBeamPhoton->dCounter && locBeamPhotonTaggedMCGEN->time() ==
locBeamPhoton->time()) ? kTRUE : kFALSE;

in
//MATCHING
if(locMCThrownMatching != NULL)
{
// Tag the thrown beam photon by comparing energy, time and counter
const DBeamPhoton* locBeamPhotonTaggedMCGEN = locMCThrownMatching->
Get_TaggedMCGENBeamPhoton();
Bool_t locIsGeneratorFlag = (locBeamPhotonTaggedMCGEN->energy() ==
locBeamPhoton->energy() && locBeamPhotonTaggedMCGEN->dCounter ==
locBeamPhoton->dCounter && locBeamPhotonTaggedMCGEN->time() ==
locBeamPhoton->time()) ? kTRUE : kFALSE;
locTreeFillData->Fill_Array<Bool_t>(Build_BranchName(locParticleBranchName,
"IsGenerator"), locIsGeneratorFlag, locArrayIndex);
} It certainly seems possible that this beam photon might not exist, if we
are requiring that it be tagged.

aaust, 2020-09-28:
PR #440 fixed the issue

closed 2020-09-28 by aaust