JeffersonLab/halld_recon #447: Crash when analysing MC simu file where the background is overlaid

igjaegle, 2020-08-26:
On the master branch, hd_root is crashing when analyzing MC simu file where the background is overlaid.

Simulation for run 30730 w/ Random:recon-2017_01-ver03

A example file can be found here:
/work/halld/home/ijaegle/amp-inter-gluex_vertex-studies-with-etapr-a2pr/Geant4_bkg_Random_runnb_30730_gen_amp/hddm/gen_amp_030730_000_geant4_smeared.hddm

the command used is:
hd_root --config=/work/halld/home/ijaegle/gluex_MCwrapper/reaction_a2.config gen_amp_030730_000_geant4_smeared.hddm -o test.root

the stack is enclosed.

This problem is not there with older software version.

tks ig.

[log.txt](https://github.com/JeffersonLab/halld_recon/files/5132634/log.txt)

aaust, 2020-08-26:
OK, I can confirm the crash. Probably related to PR #417

aaust, 2020-08-27:
I no longer believe this is related to recent changes in halld_recon, but rather somewhere earlier in the chain. When I try to run the reconstruction on your smeared files, I do not see any DBeamPhoton objects.