JeffersonLab/halld_recon #462: TOF dE/dx written as all zeros in monitoring hists and ROOT trees since 2020-07-10

acschick, 2020-11-09:
Using the most recent software build, I ran the reaction filter plugin to look at the CPP test run (72958), and I noticed that the TOF dE/dx was zero for every event in the standard ROOT tree that is produced. This was also the case when I checked the histograms produced by the monitoring_hists plugin, under the folder Independent/Hist_DetectorPID/TOF--all histograms involving TOF dE/dx had it's value as zero. 

Using the same monitoring hists however you can confirm that there are hits being recorded in the TOF, and tracks getting matched, so it's not like all info regarding the TOF is missing. 

dE/dx for all other detectors, (FDC, CDC....) all have nonzero values still and I think are having their values written correctly.

I checked the monitoring hists that get made for the nightly builds (/w/halld-scifs17exp/data_monitoring/RunPeriod-2019-11/recon_tests) to see how long the problem has been present. I skipped one month at a time until I found a plot with non-zero TOF dE/dx and was able to find out that **2020-07-10** was the first file that had TOF dE/dx uniformly zero. 

Alex A. noted that there was something checked in on July 8 (PR 410).

[image]

[image]

zihlmann, 2020-11-10:
> 2020-7-07 TOF dEdx
> 2020-7-10 TOF dEdx

It looks like the assignment dEdx was deleted when the assigments dEdx1 
and dEdx2 were introduced into the code
in DParicleID.cc at line 1002.
the line:

locTOFHitMatchParams->dEdx = locHitEnergy/dx;

	
	

has to be put back in in line 1000 and 1300.

see below:

https://github.com/JeffersonLab/halld_recon/pull/410/commits/1898ee8af48cdb16f8c73d6c3a1e1dd274396155#diff-83c8df81f35c044e2730e237fe5aedf138cb43f73c2ab48ce8ad97413f945086L1001-L1005

zihlmann, 2020-11-10:
I am testing a fix right now.....

zihlmann, 2020-11-10:
test seems to work, added pull request #463

zihlmann, 2020-11-11:
[image]
as the pictures shows it looks like the dEdx is fixed with pull request #463

closed 2020-11-24 by aaust