JeffersonLab/halld_recon #484: itrigger in Df125TriggerTime.h is (almost) always 0

nsjarvis, 2020-12-11:
itrigger when retrieved from src/libraries/DAQ/Df125TriggerTime.h is almost always set to 0, with rare apparently random exceptions.   I think this has been the case forever, I found a very old plugin of mine that also reports itrigger=0. 

I suspect that itrigger from Df125TriggerTime is only used in [PR 483](https://github.com/JeffersonLab/halld_recon/pull/483) for my plugin cdc_scan. There is some redundancy as it is present in so many other places, Df125CDCPulse and Df125FDCPulse and also I think Df125WindowRawData and probably also the event header.  

hd_dump displays itrigger from Df125TriggerTime correctly.

nsjarvis, 2020-12-11:
@faustus123

faustus123, 2020-12-11:
The itrigger member of the Df125TriggerTime class should not have been there. The value was moved into the generic base class DDAQAddress long ago since it was present in all (or nearly all) hit objects. The code happened to be set up in a way that the actual value was being copied to the DDAQAddress member which is also why it was printing correctly from hd_dump. When trying to access it from the Df125TriggerTime pointer, it was seeing the wrong member that should have had zero.

A pull request was submitted that removes the itrigger member from the Df125TriggerTime class which should fix the issue. If you were wanting to fix your plug to work with older code, you should be able to access the DDAQAddress member by just casting the pointer. Something like this:

Df125TriggerTime *trigtime =...
auto itrigger = static_cast<DDAQAddress*>(trigtime)->itrigger;

closed 2020-12-11 by nsjarvis