JeffersonLab/halld_recon #495: Dramatic change in multi-threaded processing rate

aaust, 2021-01-06:
The reconstruction test, which periodically runs the nightly-build master on the same files, showed a dramatic drop in the processing rate from December 13 to December 14, which caused all subsequent reconstruction tests to time out.

Repeating the exercise interactively with 24 threads and only the monitoring_hists plugin, I see an average rate of 60Hz for the 2020-12-13 code, while I measure only 3Hz for 2020-12-14.
 
I could not find any change in the halld_recon code itself, however the hd_root executable from December 14 produces an additional file `gmon.out` which is not created with the December 13 software stack.

sdobbs, 2021-01-06:
Was there a change in the recon test script?  This sounds like hd_root is being run with a profiler (I think gmon.out is usually created by gprof)

aaust, 2021-01-06:
This additional file is also create when I am running hd_root from the command line, without changing any thing except the environment. Maybe there was a change in the settings of nightly build scripts?

maltealbrecht, 2021-01-06:
Just a thought: When the executables were created, it seems that the compiler flags "-pg" or similar were added to enable gprof (at least thats how I once used it in the past). Were other compiler flags maybe altered as well, like for example the optimization level?

aaust, 2021-01-06:
Ok, thanks. This directly points to a series of pull request in build-scripts: https://github.com/JeffersonLab/build_scripts/pull/182
@markito3 ?

markito3, 2021-01-06:
I did enable profiling in the nightly builds. I did not anticipate that 
that would defeat mullti-threaded running.

I have since found out that gprof is no good for multi-threaded programs 
in any case. And I think that jana runs with multiple threads even when 
running "single threaded" since there are administrative threads running 
other than the main event processing ones.

I've turned profiling off for now and we can see if that makes a 
difference for the next run. Let me know if we want to do a test sooner 
than that.

Finally, the pofiling switch (-p) is not turned on by default, or so I 
allege, so I am surprised that Malte had it on.

maltealbrecht, 2021-01-06:
Hi Mark, 
sorry for not specifying it in my comment: I did not mean this package - I used gprof in the past for a completely different project and used the "-pg" flags there, so that just came to my mind when Alex was wondering about the gmon.out file.

aaust, 2021-01-08:
All recon-test jobs completed successfully now.

closed 2021-01-08 by aaust