JeffersonLab/halld_recon #498: BCAL Showers Peaking at 212 cms when using RandomBKG

karthik18495, 2021-02-03:
As per the discussions form the CalWG meeting on [3rd Feb 2021](https://halldweb.jlab.org/wiki/index.php/Feb_3,_2021_Calorimeter). This issue is being opened

I did look into sample (Run 50715) with and without Random Bkg. There seems to be a spike at 212 cms (middle of the BCAL) when using a random Bkg.

Output.pdf contains the following.

The first page has plots of a Photon gun shot at theta = 40 and momentum = 1GeV. 
* The top left plot shows the reconstructed z position [cms] for a Photon Gun sample without any BKG. 
* The top right plot shows the reconstructed z position [cms] for a Photon Gun sample with BKG=Random:recon-2018_08-ver02
* The bottom left plot shows the reconstructed z position [cms] for the Photon Gun sample with BKG=Random:recon-2018_08-ver02 and cuts applied (NumOfPointsInShower >=2 and ShowerE>=0.1GeV)
* The bottom right plot shows the energy [GeV] of the showers peaking at 212cms with the same cuts applied as in previous step

The second page of the plots has the same set of plots but for a bggen sample with and without the Random BKG.
[Output.pdf](https://github.com/JeffersonLab/halld_recon/files/5920802/Output.pdf)

zihlmann, 2021-02-03:
are the noise hits at 212cm associated with the same readout channels?

markdalton, 2021-02-03:
I asked for this to be made as an issue because I suspect that it is somehow related to the timing of the random background.  In order for points in the BCAL to have the position 212 cm, the time of the hits on each end must be exactly the same.  I don't know the details but I understand that the timing of the random events must be adjusted when they are merged with simulation events.  It seems to me that this timing adjustment is not being done quite right for the BCAL hits.

Karthik, it might be useful to point to hddm files with and without the random background in order for people to look more closely.

zihlmann, 2021-02-03:
maybe a 2d plot with phi (azimuth angle) vs z might give some insight.

karthik18495, 2021-02-03:
No, I believe it is not. Here you can see in the 2D plot (Phi Vs Z) for the bggen with noise. It doesnt happen at a single  channel.

Also the folders that has the hddm files are at

/work/halld/ksuresh/zPi0
Each folder is named as (particle)(with/without)Noise

[PhiVsZ.pdf](https://github.com/JeffersonLab/halld_recon/files/5920961/PhiVsZ.pdf)

zisis0, 2021-02-03:
This can be investigated by Karthik.  But it is probably not just a few channels.

> 
> 
> are the noise hits at 212cm associated with the same readout channels?

Dr. Zisis Papandreou
Head, Department of Physics
Professor of Physics, Ph.D., P.Phys.
---------------------------------------------------------
Department of Physics   
University of Regina	    
3737 Wascana Parkway
Regina, SK  S4S 0A2	 CANADA

Phone: (306) 585-5379
Fax: (306) 585-5659
Email:  zisis@uregina.ca <mailto:zisis@uregina.ca>
Website: http://www.uregina.ca/science/physics/people/faculty-research/zisis-papandreou/index.html <http://www.uregina.ca/science/physics/people/faculty-research/zisis-papandreou/index.html>

zihlmann, 2021-02-03:
ah, now I notice, 212cm is exaclty the center of the BCAL.  and the other two smaller spikes are at the start and end of it.

sdobbs, 2021-02-03:
Looking at one of these events, the hits that cause the shower in the middle don't seem to have exactly the same time, though it's close - I extracted a dump below.

We should check to see if these showers show up in the random trigger data themselves - maybe there really is some background in there?  I pulled some files from tape, Karthik could you please run your code over them, when they show up in this location:  `/cache/halld/RunPeriod-2018-08/recon/ver02/converted_random/merged`

`================================================================
Event: 27
DBCALCluster:
     r:   phi:      z: theta:      t:     E: E_preshower:  E_L2:  E_L3:  E_L4: N_cell: charge: t_rms:
-----------------------------------------------------------------------------------------------------
 68.95   1.58  144.84   0.71    3.31   1.06         0.05   0.64   0.31   0.05      17       0   0.39 
 68.30   4.60  232.39   0.39  -32.59   0.13         0.00   0.13   0.00   0.00       1       0   0.00 
 65.30   3.55  218.18   0.40  -44.48   0.12         0.12   0.00   0.00   0.00       1       0   0.00 

DBCALDigiHit:
 module: layer: sector:        end: pulse_integral: pulse_peak: pulse_time: pedestal: QF: nsamples_integral: nsamples_pedestal:
-------------------------------------------------------------------------------------------------------------------------------
      9      1       3  downstream            1035           0        5678         1   1                  1                  1 
     13      1       3    upstream            1404         109        1784         1   1                  1                  1 
     13      1       3  downstream            1080          84        1910         1   1                  1                  1 
     13      2       1    upstream             428          33        1775         1   1                  1                  1 
     13      2       1  downstream             460          35        1906         1   1                  1                  1 
     13      2       2    upstream            3332         260        1788         1   1                  1                  1 
     13      2       2  downstream            1920         150        1905         1   1                  1                  1 
     13      2       3    upstream           12328         963        1777         1   1                  1                  1 
     13      2       3  downstream            9096         710        1907         1   1                  1                  1 
     13      2       4    upstream             732          57        1782         1   1                  1                  1 
     13      2       4  downstream             720          56        1912         1   1                  1                  1 
     13      3       1    upstream             480          37        1781         1   1                  1                  1 
     13      3       1  downstream             316          24        1923         1   1                  1                  1 
     13      3       2    upstream            1980         154        1782         1   1                  1                  1 
     13      3       2  downstream            1968         153        1912         1   1                  1                  1 
     13      3       3    upstream            4464         348        1787         1   1                  1                  1 
     13      3       3  downstream            3360         262        1908         1   1                  1                  1 
     13      3       4    upstream             332          25        1787         1   1                  1                  1 
     13      3       4  downstream             496          38        1907         1   1                  1                  1 
     13      4       1    upstream             200          15        1810         1   1                  1                  1 
     13      4       1  downstream             148          11        1903         1   1                  1                  1 
     13      4       2    upstream             280          21        1806         1   1                  1                  1 
     13      4       2  downstream             576          45        1910         1   1                  1                  1 
     13      4       3    upstream             188          14        1792         1   1                  1                  1 
     13      4       3  downstream             252          19        1884         1   1                  1                  1 
     13      4       4    upstream             188          14        1781         1   1                  1                  1 
     13      4       4  downstream             232          18        1901         1   1                  1                  1 
     14      3       3    upstream             204          15        1801         1   1                  1                  1 
     14      3       3  downstream             284          22        1907         1   1                  1                  1 
     26      1       2    upstream            1352           0        5422         1   1                  1                  1 
     28      1       3    upstream            2836           0        1082         1   1                  1                  1 
     28      1       3  downstream            2807           0        1070         1   1                  1                  1 
[... cut ...]
 0.133   -7.81  -67.85  232.4  -32.2   68.3  -1.685        0.000  0.129  0.000  0.000       1  0  0.019  0.633  1.237   4.28  0.180  -0.045  -0.196   0.150   0.079   0.168  -0.148  -0.071  -0.716  -0.380   0.354    2.873     0.009     0.009    0.000 
 0.129  -59.91  -25.97  218.2  -44.4   65.3  -2.733        0.125  0.000  0.000  0.000       1  0  0.019  1.196  0.770   4.25  0.182  -0.188  -0.135   0.141   0.081   0.504  -0.690  -0.363  -0.480  -0.256   0.351    2.874     0.009     0.008    0.000 

`

karthik18495, 2021-02-03:
Here I had run over 50k events over the file at /cache/halld/RunPeriod-2018-08/recon/ver02/converted_random/merged/converted_random_051302.hddm 

The plots now show peaking at 2 places one at 212cm and another at around 175 cms
[OutputData.pdf](https://github.com/JeffersonLab/halld_recon/files/5921343/OutputData.pdf)

sdobbs, 2021-02-05:
So, it looks like these features exist in the random triggered data that we are using to simulate backgrounds.  We could look at data taken with other triggers, to see if it's there, maybe look at some other run periods.  Does anyone have a hypothesis for why we might have these correlated hits?  @markdalton ?

zihlmann, 2021-02-05:
maybe?: random hit means equally possible hits for any time within the 
readout window. However
hits close to either end of this window on both ends of the module will 
result in a time difference
of either zero or +- full length of the module.
Not sure how to go about this.