JeffersonLab/halld_recon #512: hd_root memory leak?

yunjie-yang, 2021-04-07:
I have ~500 MC generator output files that I want to run over using hd_root to produce some histogram (e.g. for denominator in some efficiency calculation). I use a script to generate a long command like:
“hd_root -PPLUGINS=some_plugin /path/to/file1.hddm /path/to/file2.hddm … /path/to/file500.hddm” This runs, but I notice that the memory consumption grows as it runs. For instance, if I watch using top command as it runs, about ~250 files in, the VIRT and RES are both ~20g and are both still growing.

The hd_root version I am using is from $gxenv:
/group/halld/Software/builds/Linux_CentOS7.7-x86_64-gcc4.8.5/halld_recon/halld_recon-4.25.0/Linux_CentOS7.7-x86_64-gcc4.8.5/bin/hd_root

aaust, 2022-04-26:
Richard saw the same issue when streaming input files on the OSG. 

@rjones30 : Was this fixed in the master?

rjones30, 2022-04-26:
This may or may not be the same issue as I found when running over multiple input files. In my logbook, I called it a "thread leak" because one is left with these dangling threads that never get reaped each time the input file changes. Here is a plot of the thread count over one of these runs.
[image]

This probably also results in a memory leak, but it is the number of open file descriptors that was eventually killing me in this particular instance. Is this the same problem? If so, I can tell you the work-around that David Lawrence and I eventually came up with. It is not pretty, but it works.

sdobbs, 2022-04-27:
I think this is the same problem, but my suggestion is to wait until we move to JANA2 and see if the problem still exists.  It seems like it might be better to put effort in that direction and see if it resolves the problem before trying to implement a not-pretty-but-works solution IMO.

rjones30, 2022-04-27:
Agreed, -Richard