JeffersonLab/halld_recon #522: Charged AntiSigma Problem

aaust, 2021-05-11:
The analysis library crashes when asked for a charged AntiSigma particle. It was first discovered in the reaction gamma proton to Sigma+ (19) AntiSigma- (27) proton (14), where the AntiSigma- decays into pi0 and antiproton:

Reaction1 1_14__27_14_19
Reaction1:Decay1 27__7_15

It works without the Sigma+, which decays into pi0 proton by default. I suspect there to be interference between the pi0s from two separate displaced vertices.

sdobbs, 2021-05-11:
Yeah, I would also guess there's an issue with reusing the intermediate pi0
fits.  Would you mind adding a stack trace and/or a way to reproduce this
bug?

aaust, 2021-05-11:
The reaction line above should be enough to reproduce the crash with every REST file. Here is the stack trace:

===========================================================
#6  0x00000000008aea33 in DAnalysis::DSourceComboVertexer::Calc_VertexTimeOffsets_WithBeam (this=0x7fb111b98030, locReactionVertexInfo=<optimized out>, locReactionFullComboUse=std::tuple containing = {...}, locReactionFullCombo=locReactionFullCombo
entry=0x7fb111bcb370, locBeamParticle=0x7fb109cc4fb0) at libraries/ANALYSIS/DSourceComboVertexer.cc:331
#7  0x00000000008bc8fb in DAnalysis::DSourceComboer::Combo_WithBeam (this=this
entry=0x7fb11c99c2e0, locReactions=std::vector of length 1, capacity 1 = {...}, locReactionVertexInfo=locReactionVertexInfo
entry=0x7fb109f3dd20, locReactionFullComboUse=std::tuple containing = {...}, locReactionFullCombo=0x7fb111bcb370, locRFBunch=0, locOutputComboMap=std::unordered_map with 1 elements = {...}) at libraries/ANALYSIS/DSourceComboer.cc:1879
#8  0x00000000008cb3b2 in DAnalysis::DSourceComboer::Combo_WithNeutralsAndBeam (this=this
entry=0x7fb11c99c2e0, locReactions=std::vector of length 1, capacity 1 = {...}, locReactionVertexInfo=locReactionVertexInfo
entry=0x7fb109f3dd20, locPrimaryComboUse=std::tuple containing = {...}, locReactionChargedCombo=0x7fb111c0e580, locBeamBunches_Charged=std::vector of length 1, capacity 1 = {...}, locOutputComboMap=std::unordered_map with 1 elements = {...}) at libraries/ANALYSIS/DSourceComboer.cc:1830
#9  0x00000000008cc349 in DAnalysis::DSourceComboer::Build_ParticleCombos (this=0x7fb11c99c2e0, locReactionVertexInfo=0x7fb109f3dd20) at libraries/ANALYSIS/DSourceComboer.cc:1697
#10 0x00000000006d9951 in DAnalysisResults_factory::evnt (this=0x7fb13012fa40, locEventLoop=0x7fb1300008c0, eventnumber=<optimized out>) at libraries/ANALYSIS/DAnalysisResults_factory.cc:372
#11 0x00000000006cc083 in jana::JFactory<DAnalysisResults>::Get (this=this
entry=0x7fb13012fa40, d=std::vector of length 0, capacity 0) at /u/group/halld/Software/builds/Linux_CentOS7.7-x86_64-gcc4.8.5/jana/jana_0.8.2/Linux_CentOS7.7-x86_64-gcc4.8.5/include/JANA/JFactory.h:295
#12 0x00000000006ccbe4 in jana::JEventLoop::GetFromFactory<DAnalysisResults> (this=this
entry=0x7fb1300008c0, t=std::vector of length 0, capacity 0, tag=tag
entry=0x11e5a4d "", data_source=
0x7fb134d87150: 1545425720, allow_deftag=allow_deftag
entry=true) at /u/group/halld/Software/builds/Linux_CentOS7.7-x86_64-gcc4.8.5/jana/jana_0.8.2/Linux_CentOS7.7-x86_64-gcc4.8.5/include/JANA/JEventLoop.h:452
#13 0x00000000006cceb4 in jana::JEventLoop::Get<DAnalysisResults> (this=this
entry=0x7fb1300008c0, t=std::vector of length 0, capacity 0, tag=tag
entry=0x11e5a4d "", allow_deftag=allow_deftag
entry=true) at /u/group/halld/Software/builds/Linux_CentOS7.7-x86_64-gcc4.8.5/jana/jana_0.8.2/Linux_CentOS7.7-x86_64-gcc4.8.5/include/JANA/JEventLoop.h:309
#14 0x0000000000754130 in DEventWriterROOT::Fill_DataTrees (this=this
entry=0x7fb109e28fc0, locEventLoop=locEventLoop
entry=0x7fb1300008c0, locDReactionTag="ReactionFilter") at libraries/ANALYSIS/DEventWriterROOT.cc:1126
#15 0x00007fb14126a372 in DEventProcessor_ReactionFilter::evnt (this=<optimized out>, locEventLoop=0x7fb1300008c0, locEventNumber=<optimized out>) at plugins/Analysis/ReactionFilter/DEventProcessor_ReactionFilter.cc:75
#16 0x0000000001184eca in jana::JEventLoop::OneEvent (this=this
entry=0x7fb1300008c0) at src/JANA/JEventLoop.cc:693
#17 0x0000000001185e64 in jana::JEventLoop::Loop (this=this
entry=0x7fb1300008c0) at src/JANA/JEventLoop.cc:496
#18 0x000000000115cf4a in LaunchThread (arg=0x7ffdb7a6f850) at src/JANA/JApplication.cc:1382
#19 0x00007fb15c3e1e65 in start_thread () from /lib64/libpthread.so.0
#20 0x00007fb15b6e788d in clone () from /lib64/libc.so.6

===========================================================

aaust, 2021-05-12:
Looks like, the same pi0 (0x7f22a0a95260) is used for the Sigma+ and the AntiSigma- decay:
```
CREATED COMBO USE:
Use (decay pid, z-bin, combo info, has-missing-decay-product, mass-cut-pid-to-exclude): Unknown, -2, 0x7f22a0ac2300, 0, 0:
        Particles:
        1 of Use (decay pid, z-bin, combo info, has-missing-decay-product, mass-cut-pid-to-exclude): AntiSigma-, -2, 0x7f22a0abfb70, 0, 0:
                Particles:
                1 of Use (decay pid, z-bin, combo info, has-missing-decay-product, mass-cut-pid-to-exclude): Pi0, -2, 0x7f22a0a95260, 0, 0:
                        Particles: 2 Photon,
                1 of Use (decay pid, z-bin, combo info, has-missing-decay-product, mass-cut-pid-to-exclude): Unknown, -2, 0x7f22a0ab7b00, 0, 0:
                        Particles: 1 AntiProton,
        1 of Use (decay pid, z-bin, combo info, has-missing-decay-product, mass-cut-pid-to-exclude): Sigma+, -2, 0x7f22a0aae610, 0, 0:
                Particles:
                1 of Use (decay pid, z-bin, combo info, has-missing-decay-product, mass-cut-pid-to-exclude): Pi0, -2, 0x7f22a0a95260, 0, 0:
                        Particles: 2 Photon,
                1 of Use (decay pid, z-bin, combo info, has-missing-decay-product, mass-cut-pid-to-exclude): Unknown, -2, 0x7f22a0aad300, 0, 0:
                        Particles: 1 Proton,
        1 of Use (decay pid, z-bin, combo info, has-missing-decay-product, mass-cut-pid-to-exclude): Unknown, -2, 0x7f22a0aad300, 0, 0:
                Particles: 1 Proton,
DSourceComboInfo OBJECTS CREATED
```

sdobbs, 2021-05-12:
Yeah, we probably want to prohibit pi0 reuse in particle decays from
displaced vertices...