JeffersonLab/halld_recon #531: Apparent angle in photon beam direction

aaust, 2021-06-09:
I used rho signal events to reconstruct the beam momentum using the 4-vectors of the final state particle pi+, pi- and proton. In the energy range of the coherent peak, there seems to be a considerable momentum component in x (the peak is at -4MeV) :
[image]

Furthermore, a linear dependence of this momentum component on the photon energy can be observed:
[image]

This effect can either be a real angle in the incoming photon direction or a global rotation of the detector coordinate system. Since the kinematic fit assumes an incoming photon beam exactly along the z axis without uncertainty, this could potentially cause the phi-dependent effects we see in the pulls.

rjones30, 2021-06-09:
Alexander and all,

This looks like a tilt in x by 0.5 mrad. Is that within the error of the
FDC alignment calibration?

-Richard J.

curtisameyer, 2021-06-09:
Simon was having some luck with tilts in the FDC, but cannot remember how big those have been.

Curtis

—

Curtis A. Meyer | MCS Associate Dean for Research
The Otto Stern Professor of Physics
[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]
Carnegie Mellon University
Department of Physics, Wean Hall 8414
5000 Forbes Ave., Pittsburgh PA, 15213
www.curtismeyer.com<http://www.curtismeyer.com>
[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]
[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]

staylorjlab, 2021-06-09:
The scale of the tilt  for one angle I tried  was 0.04 degrees= ~ 0.7 mrad.

Simon

[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]
[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]
[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]

rjones30, 2021-06-09:
Simon,

Was that a relative tilt of one package relative to the others, or a global
rotation of the entire tracker?

-Richard J.

staylorjlab, 2021-06-09:
Hi, Richard.
It was a global rotation.

Simon

curtisameyer, 2021-06-09:
*) Just to keep nomenclature consistent, Simon’s rotation is what many of us refer to as a tilt.

*) Could one check the impact of this by giving the initial beam photons a slight x-component and then looking at the rho events? This might mess up the
    protons in the CDC, but that is probably quite small as pions carry the majority of the momentum.

Curtis

—

Curtis A. Meyer | MCS Associate Dean for Research
The Otto Stern Professor of Physics
[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]
Carnegie Mellon University
Department of Physics, Wean Hall 8414
5000 Forbes Ave., Pittsburgh PA, 15213
www.curtismeyer.com<http://www.curtismeyer.com>
[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]
[https://s3.amazonaws.com/htmlsig-assets/spacer.gif]

aaust, 2021-06-09:
I also see a tilt in y, but it the magnitude is only half. 
I am trying to modify the beam photon to have non-zero momentum components in x and y, but so far I was not successful. Just hard-coding them in the DBeamPhoton factory was not enough.

rjones30, 2021-06-09:
Alex and all,

The correct way to do this would be to rotate the detector elements, not
the beam. The definition of the coordinate system in the simulation starts
with the z axis being defined by the direction of the beam, and the y
direction being away from the center of the earth. If we start tilting
everything, it will be difficult to know what changes in the ccdb geometry
are physical and which ones are just a change of coordinates.

-Richard Jones

rjones30, 2021-06-09:
In case you would like to see if the effect you see can be reproduced in
Monte Carlo just by rotating the beam around the target center, here is a
small post-processing script that you can place between your rho event
generator and the hdgeant4 simulation that will rotate the beam and
everything else by a general pair of tilts around the target center.

-Richard

#!/usr/bin/env python3
#
# tilter.py - post-processor to run on MonteCarlo generator output
#             before it is fed into the simulation. The purpose is
#             to rotate the incident beam direction away from the
#             direction (0,0,1) by a rotation around the y axis by
#             tilt_x, followed by a rotation around the x axis by
#             tilt_y, both angles being expressed in radians.
#
# author: richard.t.jones at uconn.edu
# version: june 9, 2021
#
# Notes:
# 1. The sign convention is such that tilt_x > 0 induces a
#    positive x-component to the incident photon momentum,
#    and likewise tilt_y > 0 induces a positive y-component.
# 2. The center of rotation is assumed to be the middle of
#    the target, and the magnitude of the tilt is assumed
#    to be small enough that the coordinates of the primary
#    vertex are unaffected.

tilt_x = -0.5e-3   # radians
tilt_y = -0.25e-3  # radians

import sys
import math
import hddm_s

if len(sys.argv) < 2:
   print("Usage: tilter.py file.hddm [ file2.hddm ... ]")
   sys.exit(1)

for infile in sys.argv[1:]:
   if infile[-5:] != ".hddm":
      print("Error: argument {} ignored, <file>.hddm
expected!".format(infile))
      continue
   outfile = hddm_s.ostream(infile[:-5] + "_tilted.hddm")
   for rec in hddm_s.istream(infile):
      for mom in rec.getMomenta():
         mom_1 = (mom.px * math.cos(tilt_x) + mom.pz * math.sin(tilt_x),
                  mom.py,
                  mom.pz * math.cos(tilt_x) - mom.px * math.sin(tilt_x))
         mom.px = mom_1[0]
         mom.py = mom_1[1] * math.cos(tilt_y) + mom_1[2] * math.sin(tilt_y)
         mom.pz = mom_1[2] * math.cos(tilt_y) - mom_1[1] * math.sin(tilt_y)
      outfile.write(rec)

aaust, 2021-06-10:
Thank you Richard, the script was very useful! I looked at 1M rho events with the tilts you suggested and the shift and even the energy dependence are really similar:
[image]

Furthermore, the pull distributions qualitatively reproduce what we see in data, especially in x and z:
[image]
Compared to data, the magnitude seems to be smaller though
https://halldweb.jlab.org/talks/2021/rho_pulls/ReactionFilter_PiMinus_Pulls_REST.png

rjones30, 2021-06-10:
Alex,

Wow, there is an unmistakable resemblance to what we were seeing in real
data. I just guessed on the sign of the x tilt. If you reversed that, would
the effects remain or grow?

-Richard

aaust, 2021-06-10:
Good guess! Inverting the signs reverses the effect on the reconstructed momentum
[image]
and on the pulls as well
[image]

I will try to find a pair of angles to match the data even better.

jrstevenjlab, 2021-06-11:
Hi Alex, All,

Would Simon’s measurement of the vertex position “slopes” reported in the logbook https://logbooks.jlab.org/entry/3817771 provide some useful input here?  A “tilt” of the whole detector relative to the beam would also be evident there.  If I remember right, those vertex positions also show a slope in x which is larger than the slope in y, but I don’t know if the direction or magnitude are anywhere near what you’ve found with these recent x, y momentum measurements.

The vertex positions and slopes from Simon’s fits are in the CCDB for each run period https://halldweb.jlab.org/cgi-bin/ccdb/versions?table=/PHOTON_BEAM/beam_spot as dxdx and dydz.

-Justin

[image]<https://urldefense.proofpoint.com/v2/url?u=https-3A__user-2Dimages.githubusercontent.com_8807112_121604970-2D7de3b400-2Dca19-2D11eb-2D94f5-2Dfaa77063aab8.png&d=DwMCaQ&c=CJqEzB1piLOyyvZjb8YUQw&r=nTWoLox3qePKkU3jkN4zKWpZiD5wwsYi6X23ArE2v3s&m=KXel3rVaSUmnoxhbno54IAB6H7R00vocuI8AHc_m4q8&s=rjp5G64A3Ap8xSNhztDqIGMSNtphNZvrA-HFcHCOoAY&e=>
[image]<https://urldefense.proofpoint.com/v2/url?u=https-3A__user-2Dimages.githubusercontent.com_8807112_121605352-2D4295b500-2Dca1a-2D11eb-2D903e-2Dd652de8dd25c.png&d=DwMCaQ&c=CJqEzB1piLOyyvZjb8YUQw&r=nTWoLox3qePKkU3jkN4zKWpZiD5wwsYi6X23ArE2v3s&m=KXel3rVaSUmnoxhbno54IAB6H7R00vocuI8AHc_m4q8&s=g0mXqWpMK8C-oSNCTFsxUL9KwsSb16xO_OFZ5030M04&e=>

rjones30, 2021-06-11:
Justin and all,

If I understand correctly, a rotation of the detector around the target
center [equivalent to a change in the beam direction without a shift in its
position at the target] would not produce any noticeable effects in the
vertex position. The target is too short for that to be visible. The
observables have to be momentum, not position, and you have to sum over all
of the tracks (+clusters? we have not tried this yet)  in an exclusive
process to see it.

-Richard J.

jrstevenjlab, 2021-06-11:
Hi Richard,

Ok, then maybe I’m misunderstanding the suggested rotation of the detector.  As an example, if the beam direction was at an angle relative to the detector (or vice versa) with a tilt in x of 0.5 mrad, then the change in the x position of the vertices in the target over a 30 cm length would be 30 * tan (0.0005) = 0.015 cm.  When fitting the vertex distribution x vs z like Simon does, this would predict a slope of 0.015/30 = 0.0005 which is pretty close to Simon’s observation.  Did I miss something in this back of the envelope calculation?

-Justin

rjones30, 2021-06-11:
Justin,

A rotation of the beam around the center of the target by 0.5mrad would
push/pull the beam center by a RMS value of 30cm * 0.5e-3 / sqrt(12) = 43
microns transverse to the target axis. Given that the intrinsic size of the
transverse beam spot is more like 5mm, one is looking for a shift in the
beam centroid by 1% of the beam spot sigma. There are many effects that can
mock up such a tiny shift in the beam spot centroid by 0.01 sigma. I would
say that the agreement between the vertex slope in Simon's logbook entry
and the slope of the momentum vs Ebeam plot shown by Alex is probably
accidental. Said another way, if the values were very different, we would
not need to explain why.

-Richard J.

>

jrstevenjlab, 2021-06-11:
Point taken, the effect on the vertex position over the length of the target is small and it could be coincidence that the slope is on the same order as the apparent shift.  I'd still think it'd be worth checking if the magnitude and direction of the vertex slopes are in agreement with the momentum-based measurements Alex is reporting.

Also if we need a longer lever arm for the vertex slope, we do reconstruct track pairs coming from the air and other material downstream of the target.  So using those downstream vertices together with those from the target might provide some new insight if these two effects are correlated?

aaust, 2021-06-11:
From data, I measured approximately -0.5mrad in x and -0.1mrad in y. A rotated and sufficiently large (10M) MC sample reproduces these values very well.

The x and z pulls of the pi- and proton qualitatively shows the same features we see in data. y looks a little different, as does the pi+.

aaust, 2021-06-21:
I analyzed a run with an incoming beam rotated by the values measured above. The histogram of the missing momentum in x looks definitely improved (blue: original, red: rotated beam):
[image]

The pull distributions for the momenta in x and z look a little better, but this effect does not seem to be the main contribution to the underlying problem. As expected, the vertex pulls are not affected by the rotation of the beam momentum vector.
[image]

aaust, 2021-06-21:
PS: This rotation improves the proton momentum pulls tremendously:
[image]

aaust, 2021-06-29:
Rotating the CDC in the xml geometry by the same amount does not have the desired effect. Black is the default, red is rotated by 0.0286 degrees in x and 0.0057 degrees in y, blue is (-0.0286, -0.0057) to check the sign.
[image]

The pull in z on the right side is not changed at all, probably because most of the z momentum is carried by the pions. Unfortunately, reconstructing the real data with a rotated FDC is not possible at present.

aaust, 2021-09-09:
I generated a rho MC sample with a rotated FDC geometry (θx=-0.0275) and reconstructed the events with the nominal geometry. This produces almost exactly the same slope in the apparent photon beam direction:
[image]

After finding the appropriate rotation in y, I will have a look at the pulls again.

aaust, 2026-02-16:
Reevaluating this issue with the recently completed 2017-01 REST ver05, we find a huge improvement.

[image]

[image]

A small offset of about 1-2MeV remains, but no visible energy dependence can be discerned.