JeffersonLab/halld_recon #600: compilation error for libraries/FMWPC/DCTOFPoint_factory.cc

markito3, 2021-11-12:
The following error is seen in the nightly build on jlabl5 only. That node is running RHEL8. The error shown is from an interactive re-try of the nightly build.

[nightly_build_error.txt](https://github.com/JeffersonLab/halld_recon/files/7530953/nightly_build_error.txt)

staylorjlab, 2021-11-13:
Looks like #include <cmath> is needed now?  I would check this but on jlabl5 I can't find scons...??

Simon

markito3, 2021-11-13:
Yes. You can't find scons. On jlabl5 the command "scons" does not exist. If you look at [night_build_error.txt ](https://github.com/JeffersonLab/halld_recon/files/7530953/nightly_build_error.txt), you see that the correct command is:

`scons-3 install PYTHONCOMMAND=python3 SHOWBUILD=1`

and please, don't blame the messenger.

aaust, 2022-04-26:
Looks like this has been solved by PR #609

closed 2022-04-26 by aaust