JeffersonLab/halld_recon #607: hd_root hangs reading certain random skim files

sdobbs, 2021-12-03:
Processing of some of the most random trigger EVIO skim files will hang for some reason.  Running this in a debugger, it looks like the hang might be in `libraries/DAQ/async_filebuf.cc` ?

You can reproduce this problem with the following command:

`hd_root -PPLUGINS=randomtrigger_skim /cache/halld/RunPeriod-2019-11/recon/ver01/random/073070/random_073070_046.evio`

The underlying problem might be the data, but any insight would be appreciated.

rjones30, 2021-12-03:
Sean, I think this one falls on my plate. Can I have a few days to clear my
desk before I get to this? In the meantime, a prepared directory full of
everything I need to reproduce it would be most helpful.
-Richard

sdobbs, 2021-12-03:
Richard,
Sure, that's not a problem.  The input file is pinned to the cache, and I
was using a recent version of halld_recon, though I think the code that
goes into this bug has been around for a couple of years.
If you want something more formal to reproduce this, I can do that.

--Sean

sdobbs, 2022-02-15:
This problem is now avoided by PR #627 - closing

closed 2022-02-15 by sdobbs