JeffersonLab/halld_recon #613: ReactionFilter plugin crashes on rhel8 with DTreeInterface::Fill error

nsjarvis, 2022-01-07:
The ReactionFilter crashes frequently on os8 and rhel8, and usually after processing most of the events.   I first noticed this at CMU with version set 5.1.0 and so ran a test on jlabl5 (rhel8) using the nightly build labelled Jan 6, and it crashes there too.  

So far I have not seen the crashes in jobs using PNTHREADS=1.   

`gxenv /u/scratch/gluex/nightly/2022-01-06/Linux_RHEL8-x86_64-gcc8.5.0/version_2022-01-06.xml

hd_root /cache/halld/RunPeriod-2017-01/recon/ver03/REST/030283/dana_rest_030283_095.hddm -PPLUGINS=ReactionFilter -PReaction1=1_14__1_1_11_12_14 -PReaction1:Flags=B4 -PReaction2=1_14__1_1_8_9_14 -PReaction2:Flags=B4 -PNTHREADS=16`

```
#5  DTreeInterface::Fill (this=0x7f40388396b0, locTreeFillData=...) at libraries/ANALYSIS/DTreeInterface.cc:292
#6  0x00000000007fbf3d in DEventWriterROOT::Fill_DataTree (this=0x7f40385101e0, locEventLoop=0x7f4038000b60, locReaction=<optimized out>, locParticleCombos=std::deque with 19 elements = {...}) at libraries/ANALYSIS/DEventWriterROOT.cc:1408
#7  0x00000000007fc8bd in DEventWriterROOT::Fill_DataTrees (this=this
entry=0x7f40385101e0, locEventLoop=locEventLoop
entry=0x7f4038000b60, locDReactionTag="ReactionFilter") at libraries/ANALYSIS/DEventWriterROOT.cc:1173
#8  0x00007f404f57b9d5 in DEventProcessor_ReactionFilter::evnt (this=<optimized out>, locEventLoop=0x7f4038000b60, locEventNumber=<optimized out>) at /usr/include/c++/8/ext/new_allocator.h:79
#9  0x000000000118cedf in jana::JEventLoop::OneEvent (this=0x7f4038000b60) at src/JANA/JEventLoop.cc:693
#10 0x000000000118d4f4 in jana::JEventLoop::Loop (this=this
entry=0x7f4038000b60) at src/JANA/JEventLoop.cc:496
#11 0x000000000116a6a9 in LaunchThread (arg=0x7ffc4b5781a0) at src/JANA/JApplication.cc:1382
#12 0x00007f40684ce17a in start_thread () from /lib64/libpthread.so.0
#13 0x00007f40681fddc3 in clone () from /lib64/libc.so.6

===========================================================
```

nsjarvis, 2023-01-27:
This is still the case with version set 5.9.0.

aaust, 2023-12-18:
This problem is still present on Alma9

rjones30, 2024-01-30:
Hello,
I am now running a test as described below within the alma9 container. So far 1000 repeats over 10000 event input (simulation) rest files, and I cannot get it to crash. Is it only on raw data? Only on specific runs? Only on CMU cluster? I am running hd_root from the standard alma9 container build that Alex maintains for us, which I access via cvmfs.
-Richard Jones

simulation Run 50986
software version 5.14.2
- PNTHREADS=8
- PLUGINS monitoring_hists,ReactionFilter COMBO:MAX_NEUTRALS 15 Reaction1 1_14__7_8_9_14 Reaction1:Flags B0_M7 Reaction2 1_14__1_7_14 Reaction2:Flags B0_M7
-PEVENTS_TO_SKIP=0
-PEVENTS_TO_KEEP=100000000
-PTHREAD_TIMEOUT_FIRST_EVENT=3600
-PTHREAD_TIMEOUT=600
--nthreads=8

nsjarvis, 2024-01-30:
Try jlabl5  (if that is still os8) -  I verified that it crashed there
before posting it as an issue.

I found the problem using real data.   I don't recall if I tried it with
simulated.

Naomi.

rjones30, 2024-01-30:
Naomi, would be best if you could specify a container. Specifying a machine
is too variable, because everyone has a different environment. For me, the
first thing I do on any machine is pop into a container, so my experience
is platform-agnostic.
-Richard Jones

nsjarvis, 2024-01-30:
Could you try the exact same command & file that I mentioned in the original post?

nsjarvis, 2024-01-30:
I cannot specify a container, sorry, I haven't been using them.  Maybe someone else could find one matching the environment mentioned in my original post.

rjones30, 2024-01-30:
To a broader audience, has anyone else encountered this crashing
ReactionFilter on any platform besides Naomi? I am unable to reproduce it
in a controlled environment like a container.
-Richard Jones

rjones30, 2024-01-30:
Good news, after many attempts, I finally saw a crash. It might take a while to diagnose it, being as rare as it seems to be in this alma9 environment. But it is definitely a thing. Here is the full crash dump. You can see that it happens after the last event has been read from the input file, so it is a program termination bug. -Richard
JANA >> --- Configuration Parameters --
JANA >> COMBO:MAX_NEUTRALS         = 15                             
JANA >> EVENTS_TO_KEEP             = 1000000000                     
JANA >> JANA:RESOURCE_DEFAULT_PATH =                                
JANA >> NTHREADS                   = 8                              
JANA >> PLUGINS                    = monitoring_hists,ReactionFilter
JANA >> Reaction1                  = 1_14__7_8_9_14                 
JANA >> Reaction1:Flags            = B0_M7                          
JANA >> Reaction2                  = 1_14__1_7_14                   
JANA >> Reaction2:Flags            = B0_M7                          
JANA >> THREAD_TIMEOUT             = 600                            
JANA >> THREAD_TIMEOUT_FIRST_EVENT = 3600                           
JANA >> -------------------------------
JANA >>events processed  (9.7k events read)  764.0Hz  (avg.: 474.5Hz)         
JANA >>
JANA >>No more event sources
JANA >>No more event sources
JANA >>
JANA >>No more event sources
JANA >>Thread 0x7f565f7fe640 completed gracefully: Tue Jan 30 10:43:33 2024
JANA >>Thread 0x7f5671189640 completed gracefully: Tue Jan 30 10:43:33 2024
JANA >>Thread 0x7f565ffff640 completed gracefully: Tue Jan 30 10:43:33 2024
JANA >>
JANA >>No more event sources
JANA >>Thread 0x7f565dffb640 completed gracefully: Tue Jan 30 10:43:33 2024
JANA >>
JANA >>No more event sources
JANA >>Thread 0x7f565effd640 completed gracefully: Tue Jan 30 10:43:33 2024
JANA >>
JANA >>No more event sources
JANA >>Thread 0x7f5664d8e640 completed gracefully: Tue Jan 30 10:43:33 2024
JANA >>
JANA >>No more event sources
JANA >>Thread 0x7f5670988640 completed gracefully: Tue Jan 30 10:43:33 2024
JANA >>Merging thread 0 (0x7f565f7fe640) ...)  530.0Hz  (avg.: 475.8Hz)     
JANA >>Merging thread 1 (0x7f5671189640) ...
JANA >>Merging thread 2 (0x7f565ffff640) ...
JANA >>Merging thread 3 (0x7f565dffb640) ...
JANA >>Merging thread 4 (0x7f565effd640) ...

===========================================================
There was a crash.
This is the entire stack trace of all threads:
===========================================================

Thread 6 (Thread 0x7f565e7fc640 (LWP 1838177) "hd_root"):
#0  0x00007f5688cfa30f in wait4 () from /lib64/libc.so.6
#1  0x00007f5688c43953 in do_system () from /lib64/libc.so.6
#2  0x00007f568b993cbc in TUnixSystem::StackTrace() () from /group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.4.1-cntr/root/root-6.24.04/lib/libCore.so
#3  0x00007f568b9912f5 in TUnixSystem::DispatchSignals(ESignals) () from /group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.4.1-cntr/root/root-6.24.04/lib/libCore.so
#4  <signal handler called>
#5  0x0000000000a3051a in DTreeInterface::Fill(DTreeFillData&) ()
#6  0x0000000000827e0e in DEventWriterROOT::Fill_DataTree(jana::JEventLoop*, DAnalysis::DReaction const*, std::deque<DAnalysis::DParticleCombo const*, std::allocator<DAnalysis::DParticleCombo const*> >&) const ()
#7  0x0000000000828fda in DEventWriterROOT::Fill_DataTrees(jana::JEventLoop*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) const ()
#8  0x00007f5678478d93 in DEventProcessor_ReactionFilter::evnt(jana::JEventLoop*, unsigned long) () from /group/halld/Software/builds/Linux_Alma9-x86_64-gcc11.4.1-cntr/halld_recon/halld_recon-4.43.1^ccdb1610/Linux_Alma9-x86_64-gcc11.4.1-cntr/plugins/ReactionFilter.so
#9  0x0000000001486e62 in jana::JEventLoop::OneEvent (this=0x7f5640000b60) at src/JANA/JEventLoop.cc:693
#10 0x0000000001487484 in jana::JEventLoop::Loop (this=this
entry=0x7f5640000b60) at src/JANA/JEventLoop.cc:496
#11 0x000000000145c3b5 in LaunchThread (arg=0x7ffc7a5f93d0) at src/JANA/JApplication.cc:1382
#12 0x00007f5688c81802 in start_thread () from /lib64/libc.so.6
#13 0x00007f5688c21314 in clone () from /lib64/libc.so.6

Thread 5 (Thread 0x7f565effd640 (LWP 1838176) "hd_root"):
#0  0x00007f5688c7e319 in __futex_abstimed_wait_common () from /lib64/libc.so.6
#1  0x00007f5688c87d8f in pthread_rwlock_wrlock
GLIBC_2.2.5 () from /lib64/libc.so.6
#2  0x0000000000a2f647 in DTreeInterface::~DTreeInterface() ()
#3  0x0000000000800899 in DEventWriterROOT::~DEventWriterROOT() ()
#4  0x0000000000800e69 in DEventWriterROOT::~DEventWriterROOT() ()
[... 54 lines cut ...]
===========================================================

The lines below might hint at the cause of the crash.
You may get help by asking at the ROOT forum https://root.cern.ch/forum
Only if you are really convinced it is a bug in ROOT then please submit a
report at https://root.cern.ch/bugs Please post the ENTIRE stack trace
from above as an attachment in addition to anything else
that might help us fixing this issue.
===========================================================
#5  0x0000000000a3051a in DTreeInterface::Fill(DTreeFillData&) ()
#6  0x0000000000827e0e in DEventWriterROOT::Fill_DataTree(jana::JEventLoop*, DAnalysis::DReaction const*, std::deque<DAnalysis::DParticleCombo const*, std::allocator<DAnalysis::DParticleCombo const*> >&) const ()
[... cut ...]
#9  0x0000000001486e62 in jana::JEventLoop::OneEvent (this=0x7f5640000b60) at src/JANA/JEventLoop.cc:693
#10 0x0000000001487484 in jana::JEventLoop::Loop (this=this
entry=0x7f5640000b60) at src/JANA/JEventLoop.cc:496
#11 0x000000000145c3b5 in LaunchThread (arg=0x7ffc7a5f93d0) at src/JANA/JApplication.cc:1382
#12 0x00007f5688c81802 in start_thread () from /lib64/libc.so.6
#13 0x00007f5688c21314 in clone () from /lib64/libc.so.6

===========================================================

Quitting after error, code=139
Reason: hd_root crashed
total 1784

nsjarvis, 2024-01-30:
In reply to your earlier question, Alex reproduced it at JLab months ago, using the os8 node.   I believe that CMU is the only GlueX group using os8.

aaust, 2024-01-30:
I remember vaguely that the EVENTS_TO_KEEP option was suppressing the issue, but this could be anecdotal. I will also try to reproduce it again.

zihlmann, 2024-01-30:
> std::__cxx11::basic_string<char, std::char_traits, std::allocator >)
> (this=0x7f564c000b60, __in_chrg=) at
> (this=0x7f564c000b60, __in_chrg=) at src/JANA/JEventLoop.cc:152
> (this=, __in_chrg=) at /usr/include/pthread.h:578
> __pthread_cleanup_class::__pthread_cleanup_class (this=, __in_chrg=)
> (this=0x7f5650000b60, __in_chrg=) at
> (this=0x7f5650000b60, __in_chrg=) at src/JANA/JEventLoop.cc:152
> (this=0x7f5658000b60, __in_chrg=) at
> (this=0x7f5658000b60, __in_chrg=) at src/JANA/JEventLoop.cc:152
>   proc=, Nthreads=) at /usr/include/c++/11/bits/stl_vector.h:1043

I suspect that if you increase the number of threads the probability to 
fail will increase.

--------------DFZvuPsjD53Tm0ouNDR7mqUR
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <font size="2">I suspect that if you increase the number of threads
      the probability to fail will increase.</font><br>
    <br>
    <div class="moz-cite-prefix">On 1/30/24 10:49, Richard Jones wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">Good news, after many attempts, I finally saw a
        crash. It might take a while to diagnose it, being as rare as it
        seems to be in this alma9 environment. But it is definitely a
        thing. Here is the full crash dump. You can see that it happens
        after the last event has been read from the input file, so it is
        a program termination bug. -Richard<br>
        JANA &gt;&gt; --- Configuration Parameters --<br>
        JANA &gt;&gt; COMBO:MAX_NEUTRALS = 15<br>
        JANA &gt;&gt; EVENTS_TO_KEEP = 1000000000<br>
        JANA &gt;&gt; JANA:RESOURCE_DEFAULT_PATH =<br>
        JANA &gt;&gt; NTHREADS = 8<br>
        JANA &gt;&gt; PLUGINS = monitoring_hists,ReactionFilter<br>
        JANA &gt;&gt; Reaction1 = 1_14__7_8_9_14<br>
        JANA &gt;&gt; Reaction1:Flags = B0_M7<br>
        JANA &gt;&gt; Reaction2 = 1_14__1_7_14<br>
        JANA &gt;&gt; Reaction2:Flags = B0_M7<br>
        JANA &gt;&gt; THREAD_TIMEOUT = 600<br>
        JANA &gt;&gt; THREAD_TIMEOUT_FIRST_EVENT = 3600<br>
        JANA &gt;&gt; -------------------------------<br>
        JANA &gt;&gt;events processed (9.7k events read) 764.0Hz (avg.:
        474.5Hz)<br>
        JANA &gt;&gt;<br>
        JANA &gt;&gt;No more event sources<br>
        JANA &gt;&gt;No more event sources<br>
        JANA &gt;&gt;<br>
        JANA &gt;&gt;No more event sources<br>
        JANA &gt;&gt;Thread 0x7f565f7fe640 completed gracefully: Tue Jan
        30 10:43:33 2024<br>
        JANA &gt;&gt;Thread 0x7f5671189640 completed gracefully: Tue Jan
        30 10:43:33 2024<br>
        JANA &gt;&gt;Thread 0x7f565ffff640 completed gracefully: Tue Jan
        30 10:43:33 2024<br>
        JANA &gt;&gt;<br>
        JANA &gt;&gt;No more event sources<br>
        JANA &gt;&gt;Thread 0x7f565dffb640 completed gracefully: Tue Jan
        30 10:43:33 2024<br>
        JANA &gt;&gt;<br>
        JANA &gt;&gt;No more event sources<br>
        JANA &gt;&gt;Thread 0x7f565effd640 completed gracefully: Tue Jan
        30 10:43:33 2024<br>
        JANA &gt;&gt;<br>
        JANA &gt;&gt;No more event sources<br>
        JANA &gt;&gt;Thread 0x7f5664d8e640 completed gracefully: Tue Jan
        30 10:43:33 2024<br>
        JANA &gt;&gt;<br>
        JANA &gt;&gt;No more event sources<br>
        JANA &gt;&gt;Thread 0x7f5670988640 completed gracefully: Tue Jan
        30 10:43:33 2024<br>
        JANA &gt;&gt;Merging thread 0 (0x7f565f7fe640) ...) 530.0Hz
        (avg.: 475.8Hz)<br>
        JANA &gt;&gt;Merging thread 1 (0x7f5671189640) ...<br>
        JANA &gt;&gt;Merging thread 2 (0x7f565ffff640) ...<br>
[... 93 lines cut ...]
        <a class="issue-link js-issue-link" data-error-text="Failed to load title" data-id="350039859" data-permission-text="Title is private" data-url="https://github.com/JeffersonLab/halld_recon/issues/6" data-hovercard-type="pull_request" data-hovercard-url="/JeffersonLab/halld_recon/pull/6/hovercard" href="https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_JeffersonLab_halld-5Frecon_pull_6&amp;d=DwMFaQ&amp;c=CJqEzB1piLOyyvZjb8YUQw&amp;r=Hy7ijcc6pcMoP-QxZxtQH4-vodW_VGkrA9xiBc7InXk&amp;m=DexCmBqjcX1a_lAGikdL06Lb8BJ93cikZh12UdrDhswuNWVa2CKaBfFmOr__IvZq&amp;s=ZF7uihl1F4SuXtH-nvIRNTjNkaewBQO6fLGChdH8AzA&amp;e=" moz-do-not-send="true">#6</a> 0x00000000014867e9 in
        jana::JEventLoop::<del>JEventLoop (this=0x7f5650000b60,
          __in_chrg=) at /usr/include/c++/11/bits/stl_vector.h:1043<br>
          <a class="issue-link js-issue-link" data-error-text="Failed to load title" data-id="350848456" data-permission-text="Title is private" data-url="https://github.com/JeffersonLab/halld_recon/issues/7" data-hovercard-type="pull_request" data-hovercard-url="/JeffersonLab/halld_recon/pull/7/hovercard" href="https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_JeffersonLab_halld-5Frecon_pull_7&amp;d=DwMFaQ&amp;c=CJqEzB1piLOyyvZjb8YUQw&amp;r=Hy7ijcc6pcMoP-QxZxtQH4-vodW_VGkrA9xiBc7InXk&amp;m=DexCmBqjcX1a_lAGikdL06Lb8BJ93cikZh12UdrDhswuNWVa2CKaBfFmOr__IvZq&amp;s=uesfbdwVYK5UOmG2hdZqWORU-KtkEs95j0Kzl0PXCJY&amp;e=" moz-do-not-send="true">#7</a> 0x0000000001486a99 in
          jana::JEventLoop::<del>JEventLoop (this=0x7f5650000b60,
            __in_chrg=) at src/JANA/JEventLoop.cc:152<br>
            <a class="issue-link js-issue-link" data-error-text="Failed to load title" data-id="351174442" data-permission-text="Title is private" data-url="https://github.com/JeffersonLab/halld_recon/issues/8" data-hovercard-type="pull_request" data-hovercard-url="/JeffersonLab/halld_recon/pull/8/hovercard" href="https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_JeffersonLab_halld-5Frecon_pull_8&amp;d=DwMFaQ&amp;c=CJqEzB1piLOyyvZjb8YUQw&amp;r=Hy7ijcc6pcMoP-QxZxtQH4-vodW_VGkrA9xiBc7InXk&amp;m=DexCmBqjcX1a_lAGikdL06Lb8BJ93cikZh12UdrDhswuNWVa2CKaBfFmOr__IvZq&amp;s=xmZtBr8izlFNu08MILXcC1KgXLtU4gMfz3tOKAim4hI&amp;e=" moz-do-not [... cut ...] tvToEdHbdzMlK9uIbUcNR6plFyXwTDIIhfD100EpbE&amp;e=" moz-do-not-send="true">#1</a> 0x00007f5688c87d8f in
        pthread_rwlock_wrlock<br>
        GLIBC_2.2.5 () from /lib64/libc.so.6<br>
        <a class="issue-link js-issue-link" data-error-text="Failed to load title" data-id="348007916" data-permission-text="Title is private" data-url="https://github.com/JeffersonLab/halld_recon/issues/2" data-hovercard-type="pull_request" data-hovercard-url="/JeffersonLab/halld_recon/pull/2/hovercard" href="ht

rjones30, 2024-02-10:
I have found and fixed this bug. It is an inplementation error in the libraries/ANALYSIS code, not a problem with ROOT or other external libraries. There was a serious blunder in the way storage for TClonesArray branches of the reactions tree was being allocated. The original scheme assumed that once a std::map entry was created for a given key, the storage for that element never moves. That is not guaranteed by the std::map standard, and in fact is not respected by any of the g++ compiler versions in the last 5 years.  The fact that we have not seen segfaults when running under RHEL7 builds does NOT mean that the results are free of the effects of this bug. In fact, it was probably there at some level all along, and in principle it affected both single-threaded and multi-threaded running. The fact that segfaults do not occur with RHEL7 builds is not proof the the bug was not affecting the output.

I have submitted a PR that I claim fixes the problem. I have only tested it on alma9, but it should work on RHEL8 as well. Please check and confirm. I also included another bug fix to DEventWriterROOT.cc in the same PR. These fixes together are needed to run safely, including but not limited to platforms with gcc v.6 or greater

nsjarvis, 2024-02-10:
Awesome. I will check it out. Thank you.

nsjarvis, 2024-02-10:
I confirm that the code completes correctly with Richard's PR  (but does not with version set 5.11.0).
Will merge it.   I tested using 32 threads, about 1/64 of the events were missing from the buggy trees; hd_root.root was a few kB & unreadable instead of MB.

closed 2024-02-10 by nsjarvis

mashephe, 2024-02-10:
> On Feb 9, 2024, at 8:53 PM, Richard Jones ***@***.***> wrote:
> 
> I have found and fixed this bug. It is an inplementation error in the libraries/ANALYSIS code, not a problem with ROOT or other external libraries. There was a serious blunder in the way storage for TClonesArray branches of the reactions tree was being allocated. The original scheme assumed that once a std::map entry was created for a given key, the storage for that element never moves. That is not guaranteed by the std::map standard, and in fact is not respected by any of the g++ compiler versions in the last 5 years. The fact that we have not seen segfaults when running under RHEL7 builds does NOT mean that the results are free of the effects of this bug.

Richard:  that is a great catch and a very subtle bug!  Indeed this is a recipe to get garbage output without warning.

I once made the same mistake but with a stl::vector instead of stl::map.  The vector is much more prone to this problem because while the stl standard does not require the memory location be fixed, for a vector it must be continuous.  As the vector grows then this often necessitates movement in memory to a larger free block in order to maintain the continuity requirement.  The stl is great and flexible, but indeed sometimes one has to think carefully about what is happening behind the scenes.  

Thanks so much for tracking this down.

Matt

aaust, 2024-02-13:
PS: Fixed by PR #785 . The total number of events in a test tree case did not change.

aaust, 2024-02-13:
**PPS: Comparison of the above-mentioned test case**

The differences have the same order of magnitude as previously observed reproducibility issues, they are not caused or solved by this fix. 

|Reaction | 4.43.1 | master with PR #785  |
|----------|------------|---------|
| tree_ggkpkm__B4 | 16427 | 16429 |
| tree_ggpippim__B4 | 54361 | 54357 |
| tree__pippim__B4 | 46760 | 46759 |

`-PREST:JANACALIBCONTEXT=calibtime=2020-07-24-00-00-01` was added to the command to obtain the correct tagger energies

`-PReaction3=1_14__8_9_14 -PReaction3:Flags=B4` was added to check high-statistics reaction