JeffersonLab/halld_recon #618: DTOFHitMatchParams::dEdx set to zero in RunPeriod-2019-11 REST files

jrstevenjlab, 2022-01-21:
I ran across an issue when looking at the Analysis Launch version 3 ROOT trees: the ChargedHypo__dEdx_TOF branch is exactly zero for all the events I’ve looked at so far.  Attached is the TBrowser for that variable

[image]

Looking back in the history, for the tagged version of halld_recon used for the RunPeriod-2019-11 production (4.19.0) the DTOFHitMatchParams::dEdx variable is set to zero because there was a change made to add dEdx1 and dEdx2 variables for the two planes as shown in this commit and the dEdx variable was left unset https://github.com/JeffersonLab/halld_recon/commit/1898ee8af48cdb16f8c73d6c3a1e1dd274396155#diff-83c8df81f35c044e2730e237fe5aedf138cb43f73c2ab48ce8ad97413f945086L1002

This shows up in the analysis when the ReactionFilter is run to produce ROOT analysis trees (where the ChargedHypo__dEdx_TOF variable is written).  In the DSelector the TOF dEdx variable is used to determine what timing detector was used for the DeltaT measurement https://github.com/JeffersonLab/gluex_root_analysis/blob/master/libraries/DSelector/DChargedTrackHypothesis.h#L467

The symptom is that there are no entries in the Hist_ParticleID directory for the TOF since the dEdx information is used to determine if those histograms get filled.  So the TOF hits will not be used correctly in the analysis (they'll likely be assumed to be FCAL hits).

A possible solution is to change the DEventSourceREST library to compute the dEdx variable from the average of dEdx1 and dEdx2, if the dEdx variable is set to zero.  Then all the downstream code would work as expected and have the usual information for halld_recon or DSelector analysis.  This would also be backwards compatible with prior run periods (where dEdx1 and dEdx2 don't exist).  Are there other suggestions or comments on this approach?

aaust, 2022-01-21:
This must be related to issue #462 . There was a fix for the TOF entry (#463), but it does not correct existing REST files. The proposed solution seems logical.

jrstevenjlab, 2022-01-24:
Thanks @aaust, I missed the previous issue, which was fixed for incoming data but not REST.  I just submitted a PR with a fix for reading REST files https://github.com/JeffersonLab/halld_recon/pull/619 that you and @zihlmann should check

aaust, 2022-04-26:
This has been fixed by PR #619 . Here is the ChargedHypo__dEdx_TOF branch in 2019-11 analysis ver04:
[image]

closed 2022-04-26 by aaust