JeffersonLab/halld_recon #643: FDC emulation triggered by MWPC raw mode data

nsjarvis, 2022-04-25:
When Df125windowrawdata for rocid > 30 is present, without an FDC pulse, the software generates an emulated FDC pulse.  

https://github.com/JeffersonLab/halld_recon/blob/2c9a71a53e6ca9285be0d0d7e8a7afc9bd19e57c/src/libraries/DAQ/JEventSource_EVIO.cc#L2183

aaust, 2022-04-26:
@faustus123 suggested to use the translation table to avoid the hard-wired numbers:

```
vector<const DTranslationTable *> ttab;
loop->Get(ttab);
if( ! ttab.empty() ){
 auto rocid_by_system = ttab[0]->Get_ROCID_By_System();
 if( rocid_by_system[ DTranslationTable::CDC ].count( rocid ) != 0 ){
  // This rocid belongs to the CDC
 }
}
```

sdobbs, 2022-04-26:
Is there a data file I can use to test this?

aaust, 2022-04-26:
Elton was using this:
```
/work/halld/home/elton/monitoring/hd_rawdata_100077_000.evio
```

nsjarvis, 2022-04-28:
Looks good now.  Thanks.

closed 2022-04-28 by nsjarvis