JeffersonLab/halld_recon #694: Unusual distribution for Lambda z-vertex in data with vertex fit for K+ Sigma0 pi0

niwgit, 2022-11-09:
Hi All,

I was comparing z-vertex distributions of K+, Lambda, pi- and proton between the two kinematic fit options "all masses constrained" (pi0kpsig0__B4) and "all masses constrained + no vertex fit" (pi0kpsig0__B4_F1) for the K+ Sigma0 pi0 channel using Fall 2018 data and MC. All the distributions are found using measured 4-vectors. 

The K+, pi- and proton z-vertex was found directly using the measured X4 (e.g. dKPlusWrapper->Get_X4_Measured()).

For both kinematic fit options I found the Lambda decay vertex using DOCA between pi- and proton tracks. (i.e. I used measured pi- and proton tracks and found the distance of closest approach between them and use the mid point as the Lambda decay vertex.)

For the K+ z-vertex plots I removed the cut on K+ vertex z used in my analysis (51.2-78.8 cm). The cut is included for all other plots. All the plots are for the Lambda(1520) mass region in Sigma0 pi0 and after all analysis cuts.

I see significant gaps in Lambda z-vertex distribution from data when vertex fit is used ! The pi- and proton distributions also seem to show this. The K+ vertex-z distribution looks fine in data for both kinematic fit options.

[image]

All the distributions from MC look smooth for the two kinfit options. I'm not sure what's making the unusual gaps in the distribution for data with vertex fit. 

[image]

I also checked the Lambda z-vertex distribution in data using kinfit X4 for the "vertex fit" case. I used "dDecayingLambdaWrapper->Get_X4()" to get the Lambda decay vertex. There also I see a similar effect.

[image]

Here are some information about the data and MC used.

Data - Fall 2018 analysis launches ver17 for "pi0kpsig0__B4" and ver18 for "pi0kpsig0__B4_F1"
root files are in:
/cache/halld/RunPeriod-2018-08/analysis/ver17/tree_pi0kpsig0__B4/merged/
/cache/halld/RunPeriod-2018-08/analysis/ver18/tree_pi0kpsig0__B4_F1/merged/

MC - produced using MCWrapper - project id 2612 with software 
Version Set:recon-2018_08-ver02_24.xml
halld_recon Version:recon-2018_08-ver02.8
halld_sim Version:4.42.0
Analysis Launch Emulation Version:analysis-2018_08-ver17.xml
output path - /cache/halld/gluex_simulations/REQUESTED_MC/Sigma0_pi0_2018_Fall_20220715094401am/

Any comments and suggestions are greatly appreciated.

Thanks,
Nilanga

niwgit, 2022-11-16:
I looked at few more kinematic fit options in data and plot the z-vertex distribution for Lambda. I was trying to see whether constraining Sigma0 mass or both Lambda, Sigma0 masses in the kinematic fit with the vertex fit lead to an issue with Lambda z-vertex.

But from the plots I don't see such an effect.

[image]

jrstevenjlab, 2022-11-21:
For the record, in discussions with others about reactions with Cascades I was reminded of another (possibly related) issue where two neutral particles come from a displaced vertex and many events have CL=1 i.e. chi^2=0

This was seen previously by Chandra (see slide 9 of https://halldweb.jlab.org/DocDB/0047/004734/003/Fall2020CollaborationmeetingPresentation.pdf) and Brandon (see slide 24 of https://halldweb.jlab.org/DocDB/0057/005750/001/CollabMeeting922.pdf)

Since the z-vertex showed such pathological behavior in Nilanga's case it's suggested that the same plots be made for the Cascade channels with and without the vertex constraint in the KinFit for comparison.

jrstevenjlab, 2022-12-05:
@sdobbs reminded us that there is a z-vertex binning for how combos are composed with BCAL photons whose kinematics can depend significantly on the z-vertex location.  The definition is in https://github.com/JeffersonLab/halld_recon/blob/master/src/libraries/ANALYSIS/DSourceComboTimeHandler.h#L139 where the bins are 10 cm wide and the central bin is in the middle of the target.  This matches the location of structure observed by Nilanga rather well.

niwgit, 2022-12-15:
@sdobbs and @jrstevenjlab, thanks for the information. 

I made a plot for Lambda z-vertex following suggestion by Susan at the cross section meeting on 12/05/2022 to look at the events for E_gamma3 > 100 MeV (i.e for photons from Sigma0 decay). 

It seems the strange behavior of z-vertex of Lambda is not directly caused by low energy photons.

[image]

niwgit, 2023-03-08:
Hi Alex, Thanks for looking into this. You can use this DSelector to reproduce the Lambda z-vertex distribution. The relevant histogram in the code is "dHist_z_Lambda".

/work/halld2/home/nwickjlb/analysis/Lambda_1405/above_6_5GeV/analysis_note/DSelector_pi0kpsig0.C