JeffersonLab/halld_recon #717: Error in DEventSourceREST - tagger hodoscope could not look up counter for energy

aaust, 2023-02-22:
Since January 26, 2023, running over certain REST files will produce millions of error messages like the one in the title of this issue. Does it mean that the tagger counter hits are ignored?

Here is one example to reproduce the error with the latest official release (5.9.0) or the current master:
`
hd_root -PPLUGINS=monitoring_hists /cache/halld/RunPeriod-2018-08/recon/ver02/REST/051035/dana_rest_051035_000.hddm`

rjones30, 2023-02-22:
These are OLD rest files, aren't these all replaced by now? Yes, these tags are being thrown away when this error message is written. I can submit a PR that has a work-around to avoid the problem by mapping them to the nearest tagger bin. Does that make sense?
-Richard Jones

sdobbs, 2023-02-22:
For the first question - yes, for the hits with that error, they are ignored.

This issue comes from the fact that for some of the older REST files (I think all of GlueX-I?) only have the tagger hit energies are saved, not the actual counter ids.  So if we update the energy mapping in the CCDB, to apply the new mapping, we need to recalculate what the counter id was at the time the REST file was generated.

This is already done automatically IF the correct CCDB timestamp is saved in the REST file.  Somehow, this was not consistently done?  Some REST files seem to have proper metadata, others do not... this is definitely a problem we should see if we can track down.

For the analysis launches, this can be fixed by passing in the right info on the command line.  For example, for REST 2018-08ver02, you would use the flag ` -PREST:JANACALIBCONTEXT="calibtime=2019-07-21-12-00-00"`

rjones30, 2023-02-22:
I submitted a PR for this issue, that takes the nearest tagger bin in the tagm or tagh instead of throwing away the tag. This is better behavior for old REST files, and does not affect the behavior for the later rest format. -rtj

rjones30, 2023-02-24:
Cancelling my PR. -rtj

aaust, 2023-05-10:
Supplying the REST calibration time explicitly is the solution:
REST:JANACALIBCONTEXT calibtime=2019-07-21-12-00-00

closed 2023-05-10 by aaust