JeffersonLab/halld_recon #722: Missing Thrown__ParentIndex information when generating MC with EvtGen postprocessing.

bgrube, 2023-03-03:
I'm using MCWrapper to generate phase-space events for the reaction $\gamma p \to \pi^- \eta' \Delta^{++}$ with $\eta' \to \pi+ \pi^- \eta$, $\eta \to \gamma \gamma$, and $\Delta^{++} \to p \pi^+$. To make the simulation more efficient I would like to use EvtGen as postprocessor in order to have 100% branching fractions for $\eta' \to \pi+ \pi^- \eta$ and $\eta \to \gamma \gamma$. However, if I run the MC chain with EvtGen, the information about the decaying particles is not propagated to the root tree with the thrown data. The `Thrown__ParentIndex` branch is always -1 for all particles and all events. I traced this back to the fact that in line https://github.com/JeffersonLab/halld_recon/blob/ebec788adab9325f1ac123f453bdbf027a36cc26/src/libraries/ANALYSIS/DEventWriterROOT.cc#L1109 the vector `locMCThrowns_Decaying` of decaying MC particles is empty. The final-state MC particles in `locMCThrowns_FinalState` seem to have the correct parent ID's:
```
    FinalState: myid = 1, pdgtype = 2212, parentid = 0
    FinalState: myid = 2, pdgtype = -211, parentid = 0
    FinalState: myid = 4, pdgtype = 211, parentid = 0
    FinalState: myid = 5, pdgtype = 211, parentid = 3
    FinalState: myid = 6, pdgtype = -211, parentid = 3
    FinalState: myid = 8, pdgtype = 22, parentid = 7
    FinalState: myid = 9, pdgtype = 22, parentid = 7
```

If I run without EvtGen and let GEANT decay the eta' and eta, the information about the decaying particles is correctly propagated and if I print the info in `locMCThrowns_Decaying` I get
```
    Decaying: myid = 3, pdgtype = 331, parentid = 0
    Decaying: myid = 7, pdgtype = 221, parentid = 3
```
in addition to the above.

I have put together a MWE at `/w/halld-scshelf2101/bgrube/pippimetapr/testMCwrapper`. I'm happy to dig further, but I would need a pointer where, the data that should be retrieved by `locEventLoop->Get(locMCThrowns_Decaying, "Decaying")` are written.

sdobbs, 2023-03-03:
Thanks, Boris - I think the next step you are looking for is in:  https://github.com/JeffersonLab/halld_recon/blob/master/src/libraries/PID/DMCThrown_factory_Decaying.cc

If you get stuck, I can look into things as well.

bgrube, 2023-03-03:
Thanks, Sean, for the quick reply! I'll have a look.

bgrube, 2023-03-03:
Looking into `dAnalysisUtilities->Get_ThrownParticleSteps()` which is called at line https://github.com/JeffersonLab/halld_recon/blob/ebec788adab9325f1ac123f453bdbf027a36cc26/src/libraries/PID/DMCThrown_factory_Decaying.cc#L38, I found that for $\eta'$ and $\eta$ the GEANT type of the MC particle, i.e. the value returned by `DMCThrown::PID()`, is 0 instead of 35 and 17, respectively, when running with EvtGen. Hence these particles are treated as `Unknown` and are discarded. Interestingly, the PDG types are set correctly for $\eta'$ and $\eta$. So something goes wrong with the filling of the GEANT type information further upstream. Any idea where I could look?

Printing `locMCThrowns` after line https://github.com/JeffersonLab/halld_recon/blob/ebec788adab9325f1ac123f453bdbf027a36cc26/src/libraries/ANALYSIS/DAnalysisUtilities.cc#L644 I get:
```
myid = 1, pdgtype = 2212, geanttype = 14 = Proton,  parentid = 0
myid = 2, pdgtype = -211, geanttype =  9 = PiMinus, parentid = 0
myid = 3, pdgtype =  331, geanttype =  0 = Unknown, parentid = 0
myid = 4, pdgtype =  211, geanttype =  8 = PiPlus,  parentid = 0
myid = 5, pdgtype =  211, geanttype =  8 = PiPlus,  parentid = 3
myid = 6, pdgtype = -211, geanttype =  9 = PiMinus, parentid = 3
myid = 7, pdgtype =  221, geanttype =  0 = Unknown, parentid = 3
myid = 8, pdgtype =   22, geanttype =  1 = Gamma,   parentid = 7
myid = 9, pdgtype =   22, geanttype =  1 = Gamma,   parentid = 7
```
and `locThrownSteps` at line https://github.com/JeffersonLab/halld_recon/blob/ebec788adab9325f1ac123f453bdbf027a36cc26/src/libraries/ANALYSIS/DAnalysisUtilities.cc#L719 is
```
0: 1, 2, 4, 5, 6, 8, 9
```

Letting GEANT decay $\eta'$ and $\eta$ things are fine:
```
myid = 1, pdgtype = 2212, geanttype = 14 = Proton,   parentid = 0
myid = 2, pdgtype = -211, geanttype =  9 = PiMinus,  parentid = 0
myid = 3, pdgtype =  331, geanttype = 35 = EtaPrime, parentid = 0
myid = 4, pdgtype =  211, geanttype =  8 = PiPlus,   parentid = 0
myid = 5, pdgtype =  211, geanttype =  8 = PiPlus,   parentid = 3
myid = 6, pdgtype = -211, geanttype =  9 = PiMinus,  parentid = 3
myid = 7, pdgtype =  221, geanttype = 17 = Eta,      parentid = 3
myid = 8, pdgtype =   22, geanttype =  1 = Gamma,    parentid = 7
myid = 9, pdgtype =   22, geanttype =  1 = Gamma,    parentid = 7
```
and `locThrownSteps` is
```
0: 1, 2, 3, 4,
3: 5, 6, 7,
7: 8, 9,
```

sdobbs, 2023-03-03:
This is what I suspected - the geant particle type is set to 0 for these decayed particles so that hdgeant4 doesn't think of the particles as "live" and tries to redecay them.

Throwing away particles of type "0" that are otherwise okay (e.g. reasonable PDG type) seems odd.

I think a reasonable fix would be to change DEventSourceHDDM::Extract_DMCThrown() and add a few lines after the PID of the new MCThrown is set, to check if the PID == 0, and then try setting it instead from the PDG type, e.g.:

`if(mcthrown->PID() == Unknown) {
    mcthrown->setPID(PDGtoPType(mcthrown-> pdgtype));
}`

Let me know what you think...

bgrube, 2023-03-03:
Thanks, Shawn, for the helpful reply! To me this would be fine and I'd be happy to prepare a pull request. The only downside of this workaround would be that the `hddm_s::ProductList` the data are copied from still contains the wrong particle ID information. Not sure, though, whether this is of any relevance. Please let me know how you want me to proceed.

sdobbs, 2023-03-03:
I don't think it's a big problem, and in any case, the hddm_s file is really an intermediate format on the way to the REST (hddm_r) format - so once the fix is applied, the correct PID information should be saved in the REST file.

Technically, there is no PID info hddm_s, it's just the Geant or PDG particle type.  We could try revising the Geant particle ID in the same function, honestly.  There is a function in particleType.h to do this too...

bgrube, 2023-03-03:
I can check for the consistency of the GEANT and PDG type information and resolve them in case one of the two is `Unknown` while the other one is a valid type. I guess there would be no way to tell which one is correct if they both would be different from `Unknown`. Should I generate a warning/error/exception in the latter case?

sdobbs, 2023-03-03:
Right, you would have to check to see which one to prefer, so I think we should recalculate them if one but not both are Unknown.

If both are Unknown... printing a warning makes sense?  But before merging, we should check this with several different types of output, like bggen...

bgrube, 2023-03-03:
I think, both of them being `Unknown` might be even okay, but if both types differ and both are not `Unknown` then something is clearly wrong. I haven't seen any logging mechanism so I would use `jerr` to print the warning.

aaust, 2023-09-08:
I assume we can close this issue?

bgrube, 2023-09-08:
Yes, thanks for the reminder, Alex.

closed 2023-09-08 by bgrube