JeffersonLab/halld_recon #733: Kinematic fit over constrained in fitted pi0 channels?

acschick, 2023-05-15:
I'm looking at events from the gp->pi0(p) and gp->pi0 pi0 (p) reaction filters. My intuitions tell me that the kin fit momenta of the two photons summed together should match the DecayingPi0 kin fit momentum--instead I see a large discrepancy. This occurs in both the single pi0 and double pi0 channels.

[image]

[Invariant Mass Plots.pdf](https://github.com/JeffersonLab/halld_recon/files/11480772/Invariant.Mass.Plots.pdf)

Is this a bug? I'm just using the auto-generated DSelector code to pull out the momentum vectors and histogram them.
Example DSelector here: /work/halld/home/shong/channels/pi0missprot/BasicRunScripts/DSelector_pi0missprot.C

zihlmann, 2023-06-14:
Using Andrews code and running over 10 files then look at the two histogram 
DecayingPi0_W and 2Photon_W do not compute because the DecayingPi0_W
is the mass constraint Pi0, hence a delta function, while the 2Photon_W is a distribution
which should not be because it is made from the two photons from the kin fit.
[image]

[image]

jonzarling, 2023-06-14:
Definitely looks wonky. Maybe the interplay between mass constrained pi0 and missing proton is some edge case we haven't encountered before? Like maybe it forgets that a missing proton can still be mass constrained, and this somehow leaks into the calculation of gamma1 and gamma2. Maybe checking the hd_root.root folders with kinfit info or some sanity checks calculating the same quantity different ways will turn something up....

zihlmann, 2023-06-15:
The missing Mass looks like the plot below requiring events with exactly 2 reconstructed photons. the red histograms
is after cutting 2 sigma around the pi0 invariant mass.
[image]

zihlmann, 2023-06-15:
Yes, indeed, I also think that the code somehow does not contrain the pi0 mass the same way as usual when there is a missing proton involved. In the plot below the red histogram represents the measured values while the blue the kinfit values. there are only small changes best seen at the "edges".
[image]

sdobbs, 2023-06-15:
I'll see if I have some time to look into this in the near future, but let me suggest you redo the fits with the "F1" option, which only imposes the 4-momentum constraints, and removes the vertex constraint from the fit - since there's no charged particle in the final state, there's no possible measurement of the vertex position, so that could be confounding the fit.

As a note, for the gp -> pi0 (p) fit, this should then be a 2C fit - one constraint for the pi0 mass in pi0 -> gg, and 1 constraint for the missing mass in the reaction being consistent with that of the proton.  If the F1 fits still show a problem, the next step would be to look at these constraints and make sure they are not beating against each other in some strange way.