JeffersonLab/halld_recon #745: memory blow up

zihlmann, 2023-09-06:
when analyzing a file from run 81421 using the plugin HLDetectorTiming the memory usage of hd_root starts climbing and climbing
till on the batch farm the job eventually will fail due to SLURM_OUT_OF_MEMORY

If I do not include  the plugin HLDetectorTiming and only use my plugin then this does not happen. The memory usage still shows
a slight increase with running time but never as much as with the plugin HLDetectorTiming.

It turns out that run 81421 is with Solenoid field OFF. In fact the above behavior can be seen with any run from the PrimeX run time
where the magnetic field was OFF.
Note that 
ccdb dump Magnets/Solenoid/solenoid_map:81421
#Set by hand
  NoField     

is correctly set as it should.

sdobbs, 2023-09-27:
@zihlmann - just checking, did you pass any command line flags to HLDetectorTiming ?  or just the defaults?

sdobbs, 2023-09-27:
and how much memory are you requesting for your jobs?

zihlmann, 2023-09-27:
I started with the 4GB that works for all other jobs/runs. And then I 
gradually increased
to 12GB I think and it sill crashed.

--------------3m9h8Jf6aHJS5a1XZQa6W8zz
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    I started with the 4GB that works for all other jobs/runs. And then
    I gradually increased<br>
    to 12GB I think and it sill crashed.<br>
    <br>
    <div class="moz-cite-prefix">On 9/27/23 10:55, Sean Dobbs wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">and how much memory are you requesting for your
        jobs?</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          visibility: hidden; overflow: hidden; opacity: 0; width: 0;
          height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
          ID: <span>&lt;JeffersonLab/halld_recon/issues/745/1737569073</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/JeffersonLab/halld_recon/issues/745#issuecomment-1737569073",
"url": "https://github.com/JeffersonLab/halld_recon/issues/745#issuecomment-1737569073",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
    <br>
  </body>
</html>

--------------3m9h8Jf6aHJS5a1XZQa6W8zz--

sdobbs, 2023-09-27:
running on the command-line, I'm seeing the jobs peak at 5-6 GB for 8 threads, which is not out of the expectation from what I see running other standard plugins, like danarest and monitoring_hists.  That's not to say that there's nothing wrong with the memory usage - but I'm not seeing anything excessive like that.  Was there a particular file this was happening to, or was it for all of the runs?

zihlmann, 2023-09-27:
as mentioned in the start of this thread my observation was that it 
happens only for run where the solenoid field was off  so for example 
run 81421

--------------C6tMp2h70xLCNEf6PvGmpYpA
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    as mentioned in the start of this thread my observation was that it
    happens only for run where the solenoid field was off  so for
    example run 81421<br>
    <br>
    <div class="moz-cite-prefix">On 9/27/23 11:06, Sean Dobbs wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">running on the command-line, I'm seeing the jobs
        peak at 5-6 GB for 8 threads, which is not out of the
        expectation from what I see running other standard plugins, like
        danarest and monitoring_hists. That's not to say that there's
        nothing wrong with the memory usage - but I'm not seeing
        anything excessive like that. Was there a particular file this
        was happening to, or was it for all of the runs?</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          visibility: hidden; overflow: hidden; opacity: 0; width: 0;
          height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
          ID: <span>&lt;JeffersonLab/halld_recon/issues/745/1737589965</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/JeffersonLab/halld_recon/issues/745#issuecomment-1737589965",
"url": "https://github.com/JeffersonLab/halld_recon/issues/745#issuecomment-1737589965",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
    <br>
  </body>
</html>

--------------C6tMp2h70xLCNEf6PvGmpYpA--

staylorjlab, 2023-09-27:
I'll take a look.

staylorjlab, 2023-09-27:
The CCAL shower factory has a memory leak.  I found the location of the problem and am looking at finding a fix.

staylorjlab, 2023-10-25:
This memory leak was fixed by https://github.com/JeffersonLab/halld_recon/pull/752

closed 2023-10-25 by staylorjlab