JeffersonLab/halld_recon #751: Pedestal subtraction "mistakes" for all Detectors that employ fADC250

zihlmann, 2023-09-25:
There are several issues regarding the handling of the pedestals in the detectors that
using fADC250. The algorithm on the FPGA is providing a value for the pedestal, however
if the algorithm fails to determine a proper value a quality factor QF is non zero and 
in digi hits is reported as QF & 0x40. There are other QF bits set for other problems
the algorithm may have had but in case of a failed pedestal determination the 7th bit
is set.

The base approach when constructing detector hits in a DANA factory is to use the pedestal 
provided by the FPGA algorithm but in the case the pedestal determination failed an average 
pedestal from CCDB pedestal table is to be used.

So in the hit factory code an initial value for the pedestal is set using CCDB data
       // For now, only keep events with a correct pedestal
        double pedestal = a_pedestals[digihit->sector-1];

And later on the pedestal from the FPGA is used if the quality factor QF is ok
        // digihit->pedestal is the sum of "nsamples_pedestal" samples
        // Calculate the average pedestal per sample
        if( (digihit->pedestal>0) && locTTabUtilities->CheckFADC250_PedestalOK(digihit->QF) ) {
            pedestal = (double)digihit->pedestal/nsamples_pedestal;
        }

the above code snipets are from libraries/START_COUNTER/DSCHit_factory.cc but other detectors
have very similar codes in their factories with the same result. 
 
It turns out that there are ISSUES with this approach for various reasons:
a) the ccdb tables are always set to zero: Start Counter
   This means no pedestal subtraction if QF == 0x40!
b) the ccdb tables are always set to a wrong value (250 instead of 100): Tagger hodoscope and microscope
   this means the wrong pedestal of 250 is subtracted if QF == 0x40!
   for the tagger hodoscope this is even more problematic since the base line shifts significantly with 
   increasing beam current to lower values.
c) the ccdb table is NOT used at all since in the case of a failed pedestal the hit is ignored: BCAL
   what is meant by this statement is that if QF == 0x40 the factory discards the hit. 
d) the ccdb table is not updated for all run periods and just fixed to 100: TOF

one can verify these problems in the factories that create the detector hits:
DBCALHit_factory.cc
DSCHit_factory.cc
DTAGHHit_factory_Calib.cc
DTAGMHit_factory_Calib.cc
DTOFHit_factory.cc

and to check what is in CCDB regarding pedestal XXXXXXX==runnumber:

ccdb dump PHOTON_BEAM/microscope/fadc_pedestals:XXXXXX     all 250
ccdb dump PHOTON_BEAM/hodoscope/fadc_pedestals:XXXXXX      all 250
ccdb dump StartCounter/pedestals:XXXXXX                    all zero
ccdb dump BCAL/ADC_pedestals:XXXXXX                        all zero
ccdb dump TOF2/pedestals:XXXXXX                            all 100 for run > 80k
ccdb dump FCAL/pedestals:XXXXXX                            looks OK   

This is a subjective order of severity:
----------------------------------------
a) all BCAL hits where the pedestal determination failed are discarded
a) all tagger hits where the pedestal failed 250 is subtracted from the ADC values
b) all Start Counter hits where the pedestal failed zero is subtracted (no pedestal subtraction)
c) all TOF hits where the pedestal failed the pedestal subtraction is fixed to 100 for run periods 80k and onward

markdalton, 2023-09-25:
It would be very useful to know how often this happens for each detector.

zihlmann, 2023-09-25:
rate of QF occurance as a function of readout channel number for various detectors using fADC250. this is for a GlueX production run at 300nA on Diamond radiator. (run 121204) 

[image]

zihlmann, 2023-09-26:
the following is the same but for run 121153 with 900nA beam current and specialized trigger thresholds to handle
the rates. Note that some of the detectors were turned off during this run test.

[image]

nsjarvis, 2023-09-26:
Hi Beni,

is the rate of occurrence the number of mistakes/triggers or
mistakes/number of hits in that detector element ?

I would expect it to be mistakes/triggers which should mostly explain the
shape of the distributions, but for the SC I am surprised that the rate is
double on one side with respect to the other, even though it's not exactly
parallel to the beam that seems large.

Naomi.

zihlmann, 2023-09-26:
the normalizing number is the "trigger number"
the nominator is the number of hits found. This number can be bigger 
than 1 for a trigger since
there can be more than one signal in the same counter for a given 
trigger. I did not want to count
only once since I think what matters is how often a pedestal has to be 
"fixed".

regarding the SC. The rate is completely dominated by the tip. Any small 
changes in the location of
the tip will have a huge impact on the rate. The "failing" of the 
pedestal determination is heavily
rate dependent because that is what defines the probability for a "tail" 
of a signal at the start of
the readout window.

--------------SikUE2TcKDQzMro6AS09ktbn
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    the normalizing number is the &quot;trigger number&quot;<br>
    the nominator is the number of hits found. This number can be bigger
    than 1 for a trigger since <br>
    there can be more than one signal in the same counter for a given
    trigger. I did not want to count<br>
    only once since I think what matters is how often a pedestal has to
    be &quot;fixed&quot;.<br>
    <br>
    regarding the SC. The rate is completely dominated by the tip. Any
    small changes in the location of<br>
    the tip will have a huge impact on the rate. The &quot;failing&quot; of the
    pedestal determination is heavily<br>
    rate dependent because that is what defines the probability for a
    &quot;tail&quot; of a signal at the start of<br>
    the readout window.<br>
    <br>
    <div class="moz-cite-prefix">On 9/26/23 09:28, nsjarvis wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      Hi Beni,
      <br>
      <br>
      is the rate of occurrence the number of mistakes/triggers or
      <br>
      mistakes/number of hits in that detector element ?
      <br>
      <br>
      I would expect it to be mistakes/triggers which should mostly
      explain the
      <br>
      shape of the distributions, but for the SC I am surprised that the
      rate is
      <br>
      double on one side with respect to the other, even though it's not
      exactly
      <br>
      parallel to the beam that seems large.
      <br>
      <br>
      Naomi.
      <br>
      <br>
      <br>
      <br>
      &gt; the following is the same but for run 121153 with 900nA beam
      current and
      <br>
      &gt; specialized trigger thresholds to handle
      <br>
      &gt; the rates. Note that some of the detectors were turned off
      during this run
      <br>
      &gt; test.
      <br>
      &gt;
      <br>
      &gt; [image: qf_pedestal_fail_run121153]
      <br>
      &gt;
<a class="moz-txt-link-rfc2396E" href="https://user-images.githubusercontent.com/13365259/270511588-2ae8164b-a06f-4030-ba8a-6cad5b022a72.gif">&lt;https://user-images.githubusercontent.com/13365259/270511588-2ae8164b-a06f-4030-ba8a-6cad5b022a72.gif&gt;</a><br>
      &gt;
      <br>
      &gt; —
      <br>
      &gt;
      <br>
      &gt;
<a class="moz-txt-link-rfc2396E" href="https://github.com/JeffersonLab/halld_recon/issues/751#issuecomment-1734701224">&lt;https://github.com/JeffersonLab/halld_recon/issues/751#issuecomment-1734701224&gt;</a>,<br>
      <br>
      &gt;
      &gt; .
      <br>
      thread.Message
      <br>
      &gt; ID: ***@***.***&gt;
      <br>
      &gt;
      <br>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
          visibility: hidden; overflow: hidden; opacity: 0; width: 0;
          height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
          ID: <span>&lt;JeffersonLab/halld_recon/issues/751/1735544085</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/JeffersonLab/halld_recon/issues/751#issuecomment-1735544085",
"url": "https://github.com/JeffersonLab/halld_recon/issues/751#issuecomment-1735544085",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
    <br>
  </body>
</html>

--------------SikUE2TcKDQzMro6AS09ktbn--

nsjarvis, 2023-09-26:
[The occupancy in 2023's SC monitoring plots is only ~ 30% higher in the middle-number counters](https://halldweb.jlab.org/work/halld2/data_monitoring/RunPeriod-2023-01/mon_ver01/Run121204/ST_occupancy.png) (15 ish), but I suppose the algorithm failure rate is some power of this.

zihlmann, 2023-09-26:
for completeness these are the pedestals for the high intensity run 121153 where a pedestal value is reported.

[image]

zihlmann, 2023-10-23:
Please take a look and consider updating the CCDB with pedestal values that make sens to be used as alternative to when the pedestal data is not available from the raw data.

sdobbs, 2023-12-14:
We should resolve this for all detectors before the CPP data start production - I suggest we review the situation and make some decisions in the next C&P meeting.

@nsjarvis pointed out this high priority action item - https://halldweb.jlab.org/wiki-private/index.php/December_13,_2023,_Calibration_%26_Production#Old_Action_Items

rjones30, 2024-01-12:
Verified that the TAGM and TAGH default pedestals were all set to 250 in ccdb, whereas should be 100. Changed to 100 for all runs 10000-inf , var=mc and var=default, for both microscope and hodoscope. Updates verified.
-Richard Jones