JeffersonLab/halld_recon #760: Cannot analyse multiple run numbers together

igjaegle, 2023-10-23:
hd_root failed if multiple runs were analyzed together due to a memory leak/calibration table which are vector are not cleared when a new run is analysed.

igjaegle, 2023-10-23:
Here an example of cfg
PLUGINS monitoring_hists

JANA_CALIB_CONTEXT variation=mc

NTHREADS 12

# no trigger applied
TRIG:BYPASS 1

and if you do

hd_root --config=test.cfg /work/halld/home/gxproj2/gluex_simulations/primex-eta-simu-for-src-runs-09062023-batch1-with-rdm-bkg-1M/*all_and_eta_to_gg/hddm/*smeared*.hddm -o ~/test-src.root

it will crash at some points. It is true also with evio file and for all plugins.

igjaegle, 2023-10-23:
> Can you please add more information to this issue? How can the failure
> mode be reproduced?

Sure.

example of cfg file:
PLUGINS monitoring_hists
JANA_CALIB_CONTEXT variation=mc
NTHREADS 12
# no trigger applied
TRIG:BYPASS 1

and then do

hd_root --config=test.cfg
/work/halld/home/gxproj2/gluex_simulations/primex-eta-simu-for-src-runs-09062023-batch1-with-rdm-bkg-1M/*all_and_eta_to_gg/hddm/*smeared*.hddm
-o ~/test-src.root

it will crash for all run periods, hddm or evio files, and plugins

tks ig.

aaust, 2023-10-23:
Thanks @igjaegle . It looks like the TOF geometry is not properly loaded here:

https://github.com/JeffersonLab/halld_recon/blob/master/src/libraries/TOF/DTOFPaddleHit_factory.cc#L56

Every time I saw the crash, TOF_NUM_BARS is 0 even though it should be 46.

aaust, 2023-10-23:
These routines look suspicious:
https://github.com/JeffersonLab/halld_recon/blob/master/src/libraries/TOF/DTOFGeometry_factory.h#L42

aaust, 2023-10-25:
PR #761 addresses this issue by removing the dependency on the TOFGeometry altogether. The above mentioned test case runs throught 100k events without crash.

@staylorjlab Do you know why exactly the DTOFGeometry caused a problem? I see there are many other detector specific geometry factories in the code, that are constructed in the same manner, e.g. DBCALGeometry_factory.cc
 In order to prevent the usage in the future, we should remove unused factories from the code base.

sdobbs, 2023-10-25:
did someone check if this solution gives the correct results? it's not clear to me how the underlying problem is actually solved, and I worry it's being hidden

zihlmann, 2023-10-25:
the underlying problem is solved because it is removed.
there is no need to make a separate geometry object for every detector 
if the same functionality can already
be obtained by using "app->" from JANA.
the problem is that this additional factory based geometry produces 
racing conditions when it is deleted and
created again every time the run number changes. This is not a problem 
with single are small number of threads
but will cause uncontrollable/random issues with many threads.

--------------5Fu5HRHG5JRJ3s0aXHk44OdM
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    the underlying problem is solved because it is removed.<br>
    there is no need to make a separate geometry object for every
    detector if the same functionality can already<br>
    be obtained by using &quot;app-&gt;&quot; from JANA.<br>
    the problem is that this additional factory based geometry produces
    racing conditions when it is deleted and<br>
    created again every time the run number changes. This is not a
    problem with single are small number of threads<br>
    but will cause uncontrollable/random issues with many threads.<br>
    <br>
    <div class="moz-cite-prefix">On 10/25/23 10:29, Sean Dobbs wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">did someone check if this solution gives the correct
        results? it's not clear to me how the underlying problem is
        actually solved, and I worry it's being hidden</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
        thread.[image]<span style="color: transparent; font-size: 0; display: none; visibility: hidden; overflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
          ID: <span>&lt;JeffersonLab/halld_recon/issues/760/1779411670</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/JeffersonLab/halld_recon/issues/760#issuecomment-1779411670",
"url": "https://github.com/JeffersonLab/halld_recon/issues/760#issuecomment-1779411670",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
    <br>
  </body>
</html>

--------------5Fu5HRHG5JRJ3s0aXHk44OdM--

staylorjlab, 2023-10-25:
At least for the TOF geometry, the code that parsed the XML for the TOF geometry parameters was in the DGeometry.cc file in the HDGEOMETRY library and reconstituted into the DTOFGeometry factory.   I am not sure what the benefit of this extra layer was.   Furthermore, there is no need for the full factory implementation with an evnt method -- we only need to do things when the run number changes.

aaust, 2023-10-25:
I can confirm that TOF_NUM_BARS is now always 46 as it is supposed to be. It sometimes had random values (e.g. 0, 32518, 1080126229, -1130256858, ...) before.

mashephe, 2023-10-25:
> On Oct 25, 2023, at 10:44 AM, Sean Dobbs ***@***.***> wrote:
> 
> Also I guess I don't understand the race condition here - is it something that could be solved by just putting a lock around where the DTOFGeometry object is deleted / created again, or maybe by changing the ownership of the object, so that it's not getting deleted while being used by whatever function?

I was watching this thread with one eye.... this brings up some issues I recall in the past.

Usually the geometry and other things that get updated change when the run number changes.  I believe at such points the JANA framework triggers a call to brun().

When running multithreaded then the various threads can be at different stages of crossing the run boundary and I think it is in principle possible for one thread to make the transition to the next run and update the geometry while other threads are still processing an event from the previous run.

Maybe this is not the issue in the current discussion, but I recall spending quite some time talking with Paul how to handle this.  You almost need the analogue of a "thread sync" at brun() that allows all threads to catch up before moving forward.  This may be inefficient for sparse skims.

Matt

sdobbs, 2023-10-25:
Right - it seems like an ownership issue here, that these objects are being deleted while they are still being used in some evnt() function.  So it's fine that it's fixed for the TOF, but if it's a bad pattern we should try to resolve it.

This also leads to a chunk of copied geometry-related code in DTOFPoint_factory.cc and DTOFPaddle_factory.cc - for what it's worth.

staylorjlab, 2023-10-25:
In principle the copied-code issue can be fixed.

staylorjlab, 2023-10-25:
.. but it might not be worth it...

mashephe, 2023-10-25:
> Right - it seems like an ownership issue here, that these objects are being deleted while they are still being used in some evnt() function. So it's fine that it's fixed for the TOF, but if it's a bad pattern we should try to resolve it.

To aim for a general solution, we might need some knowledge of how JANA manages the brun() calls across various threads.  I imagine each thread gets its own stream of events, so is brun() invoked when the run number changes within a thread?   Or is JANA looking at the primary input stream of events and invoking brun() across threads at the same time when a change of run number is noticed.  (Or maybe something else...)  

Below is some old code I have but I can imagine multiple ways in which this will fail... it looks quite thread "unsafe".. and shouldn't be used as an example!  (Although a quick look at the TOF code referenced in this discussion indicates that has the same problems.)

Inside brun():
```
 DApplication *dapp = dynamic_cast<DApplication*>(loop->GetJApplication());
 const DGeometry *geom = dapp->GetDGeometry(runnumber);
    
  if( geom ) {

    geom->GetTargetZ( m_zTarget );
    geom->GetFCALZ( m_FCALUpstream );
  }
  else{
      
    cerr << "No geometry accessbile." << endl;
    return RESOURCE_UNAVAILABLE;
  }
```

Problems:

- One thread will update member data m_zTarget and m_FCALUpstream and this gets shared amongst all threads.  The update may even happen while another thread is in the middle of processing an event.  
- The plugin also works with a pointer to DGeometry* but who knows how safe/stable this pointer is.  In the safest case, there is a persistent object, but I suspect that produces large memory consumption if a particular job requires many different instances of the DGeometry.  I don't see a backwards-compatible way around it.  One needs a reference counting object so that when the last pointer pointing the a particular DGeometry object goes out of scope then the object goes out of scope.  As your job then crosses a run boundary there is an uptick in memory use as threads begin using the DGeometry for the new run and then the memory is freed as the last threads stop using the geometry from the old run.  Or the run transition is synchronized across threads so all make the change at the same time, which can be inefficient.

A real problem "use case" is a multithreaded run over a sparse skim where the geometry (or something else) is changing frequently on a run by run basis.  This is not the use case we have in many examples because the jobs that depend on this low level stuff are often processing subsets of individual runs.   

I don't know if this is really a "bug."  Seems like a standard solution to it might be a feature request but not sure how feasible that is.  In the case that run numbers change asynchronously within a thread (likely the case) but brun() and erun() are called (also asynchronously) as this happens it is possible for plugin can keep track of a set of run numbers that all threads are using and preserve and release data as needed.  

So in that DTOFGeometry_factory.h, you need a counter, and a method to hang on to the objects... something like this:

map< long int, int > m_runCounter;
map< long int, DTOFGeometry* > m_tofGeomRunMap;  

In brun():

if( m_runCounter[run]++ == 0 ) {  m_tofGeomRunMap[run] = new ... }

Then use:   m_tofGeomRunMap[run]

in erun():

if( --m_runCounter[run] == 0 ) { delete m_tofGeomRunMap[run] }

... that factory class has other problems, but a solution like this would manage memory and the DTOFGeometry objects would live for as long as needed.

Getting rid of the bad examples that already exist in the code base might be hard.

igjaegle, 2023-12-20:
This issue has been solved.

closed 2023-12-20 by aaust