JeffersonLab/halld_recon #763: Use of TAGH energy calibration from REST files

jrstevenjlab, 2023-11-13:
When using REST files as the source for analysis launches over REST files from GlueX-I production we have been correcting the tagger energy calibration to be consistent with the newer scheme used for PrimEx and GlueX-II.   The TAGM energy seems to be recalculated and assigned to the DBeamPhoton objects

https://github.com/JeffersonLab/halld_recon/blob/master/src/libraries/HDDM/DEventSourceREST.cc#L659

However, the TAGH energy isn't recalculated and is assigned to the DBeamPhoton object with the energy stored in the REST file 

https://github.com/JeffersonLab/halld_recon/blob/master/src/libraries/HDDM/DEventSourceREST.cc#L718

Is that the intention?  If so, we need to make this consistent in the calculation of the flux being discussed here https://github.com/JeffersonLab/hd_utilities/issues/217.  If not, should this be corrected for future analysis launches?

sdobbs, 2023-11-13:
Indeed this is the wrong behavior.  I opened #764 that should fix the problem for testing - probably it shouldn't be merged before we make sure it works properly.

It is weird that both tagger detectors were being handled differently.  I will do some archaeology to make sure that some change wasn't squashed at some point in the past (see recent halld_sim PR for an example of this).

aaust, 2023-11-22:
@sdobbs Did you find out what happened? The fix looks reasonable

jrstevenjlab, 2024-01-29:
Following up on this thread with respect to the flux determination, we implemented a fix for this to determine the flux properly based on the analysis launch version https://github.com/JeffersonLab/hd_utilities/compare/plot_flux_ver1.0

We need to check that the database containing this information is updated for all the GlueX-I periods and then should send an announcement that people should use the new flux script and specify their launch version.

sdobbs, 2024-08-22:
just bumping this issue - was everything updated for all of the run periods and properly communicated?