JeffersonLab/halld_recon #776: Reconstruction of low-momentum protons in 2023-01 data set

aaust, 2024-01-10:
We observe a strange peak at low -t for rho production in the latest REST production of 2023-01 data (bottom right):
[image]

This peak corresponds to an enhancement in the reconstructed proton momenta around 250MeV:
[image]

This issue is present in both ver01 ([version_5.8.0.xml](https://halldweb.jlab.org/halld_versions/version_5.8.0.xml)) and ver02 ([version_5.13.0.xml](https://halldweb.jlab.org/halld_versions/version_5.13.0.xml)). As a next step, we will compare the results with [version_4.24.0.xml](https://halldweb.jlab.org/halld_versions/version_4.24.0.xml), which was used for the 2020 data when the issue was not visible.

rjones30, 2024-01-10:
Hello all,

Not sure if this is related, but it looks somewhat like what we saw in
comparing reconstructed track momenta in MC with what we found in the REST
file. These were for kaons, but could probably obtain similar results for
somewhat heavier protons. Does this look related? We investigated the
low-energy track failure, and understood it.

-Richard Jones

aaust, 2024-01-10:
Did we solve it though? It is still listed as open issue #378 here

rjones30, 2024-01-11:
This issue 378 that you point to is different from the one I raised, with
the plot I shared earlier today with low-energy kaons reconstructing as a
spike at 250 MeV/c. They are unrelated, I believe. -Richard

aaust, 2024-01-11:
I repeated the reconstruction with [https://halldweb.jlab.org/halld_versions/version_4.24.0.xml[version_4.24.0.xml], which was used for the reconstruction of the 2020 data. The peak at low -t disappears, but also do 20% of the events
[image]

aaust, 2024-06-27:
I repeated the analysis with version_5.18.0.xml and the current geometry. The problem seems less apparent now:
[image]
Something was seriously wrong with the histogram normalization that came out of the production, so don't trust the number of rho's in the first post in this thread.

By request of the tracking group, the same analysis was also performed with a threshold on the CDC hits:
`CDC:CDC_HIT_THRESHOLD 180`
Consistent with the observations by @zihlmann , we see a few percent increase in the yields by applying this cut. It does not have a strong effect on the issue at hand. 
[image]

closed 2024-12-12 by nsjarvis