JeffersonLab/halld_recon #791: Incorrect Trigger type CFCAL

aaust, 2024-03-25:
For the 2022-08 PrimEx data set, the current halld_recon master prints many error messages of this type:

> Incorrect Trigger type CFCAL

The trigger emulation does not seem to include this type in

https://github.com/JeffersonLab/halld_recon/blob/1d98aa5c06198d83129c32a2ab27f23e6feb4857/src/libraries/TRIGGER/DL1MCTrigger_factory.cc#L891

somovalex, 2024-03-25:
Yes, the trigger emulation has not yet been included for this trigger type. We can skip it at the moment.

https://github.com/JeffersonLab/halld_recon/blob/1d98aa5c06198d83129c32a2ab27f23e6feb4857/src/libraries/TRIGGER/DL1MCTrigger_factory.cc#L891<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_JeffersonLab_halld-5Frecon_blob_1d98aa5c06198d83129c32a2ab27f23e6feb4857_src_libraries_TRIGGER_DL1MCTrigger-5Ffactory.cc-23L891&d=DwMCaQ&c=CJqEzB1piLOyyvZjb8YUQw&r=UEJkvUcBvE4mBa6l1pQs5Q&m=6PVTX64ly26ADgUk55NNczAr007m6Ige_PEIEp55L_kVyOmn0YMZVA6vx4GWGOnP&s=q4sc-L_6-MVkqxKk7BQwpoetpnA-cpOgSOASyvUqRdo&e=>

aaust, 2024-03-26:
@somovalex  Should this be included before we start REST production?

somovalex, 2024-03-26:
We can skip it for this launch

sdobbs, 2024-03-28:
for the record, this code is called in the data reconstruction to estimate the trigger calorimeter energy sums, and the trigger bit info is ignored.  So maybe the short term fix is to limit the number of times this error message is printed out.

Of course, simulation of this trigger type is needed for acceptance corrections.

somovalex, 2024-03-28:
Yes, this would be the temporary solution (before I finish up with the final implementation)