JeffersonLab/halld_recon #792: BCAL timing distribution data/MC comparison

jrstevenjlab, 2024-03-27:
Starting a new issue to focus on the BCAL timing for charged hadrons which evolved from the original topic in https://github.com/JeffersonLab/halld_recon/issues/780. 

The goal here is to understand the impact of our existing +/-1 ns cut on pions and protons (+/-0.75 ns cut on kaons) in the ANALYSIS library and how well (or not) that's modeled in MC.  For the TOF we saw that using the "first hit" time in hdgeant4 was a much better representation of the data, removing a long tail of late arrivals from the timing distribution that existed for the "energy weighted" time previously.

The default BCAL hit time is currently computed with the energy weighted time with a switch that can be modified to use the first hit time.  The BCAL reconstruction is significantly more complicated than the TOF, including both charged and neutral showers, so any potential modification to the simulation would need to be thoroughly tested

https://github.com/JeffersonLab/HDGeant4/blob/master/src/GlueXSensitiveDetectorBCAL.cc#L24

Please post any further studies of BCAL timing distributions here, so we can factorize this issue from the other thread.

jrstevenjlab, 2024-03-27:
@s6pepaul, for the plots posted here https://github.com/JeffersonLab/halld_recon/issues/780#issuecomment-2023392376 for the slow proton and K-, how "pure" are the samples from data that you're comparing simulation to?  Can you select the Lambda(1520) region or make a tight KinFit chi^2 cut to make sure the sample is pure?  

I ask because if there were mis-identified pi- in the K- sample then they would tend to populate the negative values of DeltaT where you see the biggest data/MC disagreement, right?  If that's the case then I would suggest we focus on the tail for positive times and potentially normalize by the maximum rather than the integral which may include some background.

Finally, when you make the trees for the MC could you widen the BCAL timing cut to something very large like 20 ns?  We can't do this in the data without another analysis launch, but it would be good to see how high in time the tail persists in MC to followup on @rjones30 's observation from the particle gun.

s6pepaul, 2024-03-28:
Hi Justin,

you raise an important point about the purity. The truth is, I don't know how much background there will be, but I would guess there will be some. I placed a 10^-6 CL cut and analysed only m(pK)<1.7GeV. I didn't want to cut tighter for stats. But I am currently rerunning the data with a 10^-3 CL cut, tighter MM2 and only the L(1520) peak. I am not sure yet if we have enough stats to bin, but we should be able to look at least at the 0-2GeV momentum range.

At the same time I am also rerunning the MC with +/-20ns windows. I will apply the same stricter cuts for comparison and post the plots as soon as they are done.

s6pepaul, 2024-03-28:
Hi, here are the updates with tighter cuts on the final state as outlined above and normalised to the maximum. The shoulder on the left is reduced significantly as expected.

There is still some discrepancy on the high side, and the means are off in the avg BCAL case.

[image]
[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]

farah88, 2024-03-28:
I have increased the BCAL timing cut to 20 ns for MC and I used the ver13 data runs from Fall 18 with a cut at 3 ns. Here is the direct comparison for the slow pions from the Delta decay. The mean and width values are about the same as in my last post and therefore not shown. The tail drops considerably after 3 ns in the MC. 
  
[bcal_slowpionsfromDelta_logy_20ns.pdf](https://github.com/JeffersonLab/halld_recon/files/14794975/bcal_slowpionsfromDelta_logy_20ns.pdf)

jrstevenjlab, 2024-03-28:
Here are plots of the proton DeltaT distribution recoiling against the omega -> pi+pi-pi0 as Sean requested.  There are several bins of momentum (in GeV) given by the file names.  

[DeltaT_BCAL_Proton_p0.0_0.5.pdf](https://github.com/JeffersonLab/halld_recon/files/14795649/DeltaT_BCAL_Proton_p0.0_0.5.pdf)
[DeltaT_BCAL_Proton_p0.0_2.0.pdf](https://github.com/JeffersonLab/halld_recon/files/14795650/DeltaT_BCAL_Proton_p0.0_2.0.pdf)
[DeltaT_BCAL_Proton_p0.5_1.0.pdf](https://github.com/JeffersonLab/halld_recon/files/14795651/DeltaT_BCAL_Proton_p0.5_1.0.pdf)
[DeltaT_BCAL_Proton_p1.0_1.5.pdf](https://github.com/JeffersonLab/halld_recon/files/14795652/DeltaT_BCAL_Proton_p1.0_1.5.pdf)
[DeltaT_BCAL_ProtonFit_p0.0_0.5.pdf](https://github.com/JeffersonLab/halld_recon/files/14795653/DeltaT_BCAL_ProtonFit_p0.0_0.5.pdf)
[DeltaT_BCAL_ProtonFit_p0.0_2.0.pdf](https://github.com/JeffersonLab/halld_recon/files/14795654/DeltaT_BCAL_ProtonFit_p0.0_2.0.pdf)
[DeltaT_BCAL_ProtonFit_p0.5_1.0.pdf](https://github.com/JeffersonLab/halld_recon/files/14795655/DeltaT_BCAL_ProtonFit_p0.5_1.0.pdf)
[DeltaT_BCAL_ProtonFit_p1.0_1.5.pdf](https://github.com/JeffersonLab/halld_recon/files/14795656/DeltaT_BCAL_ProtonFit_p1.0_1.5.pdf)

Generally, the agreement is quite good in the width of the gaussian peak between data and MC.  The times are shifted a bit later in the MC, which is expected for the default energy-weighted algorithm and agrees with @s6pepaul  and @farah88's plots.  There is a tail at large DeltaT, but it doesn't seem to be as big of an effect as in the TOF case.

@rjones30 and @aaust, should we do an analysis launch over run 51768 of the data with the BCAL timing cut set at +/-10 or 20 ns, so we can check the full tail of the distribution to large times?

aaust, 2024-03-28:
@jrstevenjlab Please check out these trees:
`/volatile/halld/home/aaustreg/RF_Bunch_ver03/BCAL_20.0ns/`

jrstevenjlab, 2024-03-29:
Thanks @aaust, below are the plots with the 20 ns wide window for protons recoiling agains the omega.  The tail in data does seem to be weaker than that in MC, but doesn't appear to be as big of an effect as we saw with the TOF using the energy-weighted time.   

@farah88, could you remake your plots of the slow pions and proton for the Delta++ with this data that Alex produced?

[DeltaT_BCAL_Proton_p0.0_0.5.pdf](https://github.com/JeffersonLab/halld_recon/files/14797398/DeltaT_BCAL_Proton_p0.0_0.5.pdf)
[DeltaT_BCAL_Proton_p0.5_1.0.pdf](https://github.com/JeffersonLab/halld_recon/files/14797400/DeltaT_BCAL_Proton_p0.5_1.0.pdf)
[DeltaT_BCAL_Proton_p0.0_2.0.pdf](https://github.com/JeffersonLab/halld_recon/files/14797399/DeltaT_BCAL_Proton_p0.0_2.0.pdf)

sdobbs, 2024-03-29:
It's interesting to see how the tail changes in this case with and without accidental subtraction, which I think is a little surprising.  Maybe it's useful to make some MC without accidental photons (or just completely without the random trigger backgrounds?) in order to check this?

jrstevenjlab, 2024-04-03:
Here are the same plots of proton DeltaT recoiling against omega -> 3pi from above https://github.com/JeffersonLab/halld_recon/issues/792#issuecomment-2026415934, but with larger statistics from the additional data Alex produced.  

[DeltaT_BCAL_Proton_p0.0_0.5.pdf](https://github.com/JeffersonLab/halld_recon/files/14855933/DeltaT_BCAL_Proton_p0.0_0.5.pdf)
[DeltaT_BCAL_Proton_p0.5_1.0.pdf](https://github.com/JeffersonLab/halld_recon/files/14855935/DeltaT_BCAL_Proton_p0.5_1.0.pdf)
[DeltaT_BCAL_Proton_p0.0_2.0.pdf](https://github.com/JeffersonLab/halld_recon/files/14855934/DeltaT_BCAL_Proton_p0.0_2.0.pdf)

It seems that this long tail to very large times persists in both data and MC, but at a smaller level compared to the original effect seen in the TOF.  The width of the gaussian core of the proton time distributions is also in reasonable agreement, despite the 30-60 ps shift due to using the default energy-weighted time for the BCAL in hdgeant4

[DeltaT_BCAL_ProtonFit_p0.0_0.5.pdf](https://github.com/JeffersonLab/halld_recon/files/14856035/DeltaT_BCAL_ProtonFit_p0.0_0.5.pdf)
[DeltaT_BCAL_ProtonFit_p0.5_1.0.pdf](https://github.com/JeffersonLab/halld_recon/files/14856038/DeltaT_BCAL_ProtonFit_p0.5_1.0.pdf)
[DeltaT_BCAL_ProtonFit_p0.0_2.0.pdf](https://github.com/JeffersonLab/halld_recon/files/14856037/DeltaT_BCAL_ProtonFit_p0.0_2.0.pdf)

farah88, 2024-04-03:
Here are the updated plots for the slow pions and protons from the Delta decay with a MC sample that has tagger accidentals included and all of the data runs that Alex produced. They are shown with (w) and without (wo) accidental subtraction for both data and MC. 

For these plots, only the timing cut windows for BCAL were set to 20 ns. Having widened cuts in addition for the other detectors leads to issues in finding the right RF bunch and several of the side RF bunches show up (see my last post).
There is a long tail in MC and data that also remains after accidental subtraction (in my previous plot a cut was responsible for the cutoff of the long tail).   

[bcal_slowpionsfromDelta_20ns_withtag.pdf](https://github.com/JeffersonLab/halld_recon/files/14859147/bcal_slowpionsfromDelta_20ns_withtag.pdf)

[bcal_slowprotonsfromDelta_p_0_0.5_20ns_withtag.pdf](https://github.com/JeffersonLab/halld_recon/files/14859155/bcal_slowprotonsfromDelta_p_0_0.5_20ns_withtag.pdf)
[bcal_slowprotonsfromDelta_p_0.5_1_20ns_withtag.pdf](https://github.com/JeffersonLab/halld_recon/files/14859165/bcal_slowprotonsfromDelta_p_0.5_1_20ns_withtag.pdf)
[bcal_slowprotonsfromDelta_p_1_2_20ns_withtag.pdf](https://github.com/JeffersonLab/halld_recon/files/14859179/bcal_slowprotonsfromDelta_p_1_2_20ns_withtag.pdf)

aaust, 2024-04-05:
I determined the rho cross section with the samples analyzed with various BCAL timing cuts applied to all pions and protons. As expected, the flux normalized yield slightly increases when widening the default cut (1.0ns ) to 1.5 or even 2ns, and decreases when narrowing the cut to 0.75ns. The absolute cross section, however, is remarkably stable in this channel. The average over the full energy varies by less then 0.5%.
[image]
[image]

aaust, 2024-04-08:
For completeness, here is the data/mc comparison of the DeltaT distribution for protons:
[image]
I don't observe a tail for positive values of deltaT in MC, and the width of the distribution approximately matches the one from real data. The structure at negative values in the data sample is background from misidentified pions.

[image]
[image]

jrstevenjlab, 2024-04-09:
For the study of the BCAL time window cut on omega -> 3 pi channel I repeated a similar study to Alex, computing the omega -> 3pi yield for a subset of the Fall 2018 data (runs 51722-51768) with varied BCAL timing cuts on protons and pions from 0.75-20 ns.  The omega -> 3pi yield was also computed for a sample of MC with the same cuts to correct for the efficiency.  

The first figure below shows the pi+pi-pi0 yield for MC (left) and data (right) with these varied BCAL timing cuts, which shows that there is a small difference in the yield for both.  The yield decreases with a tighter than nominal cut (0.75 ns) and increases with a looser cut up to 20 ns, but the effect is not very large

[image]

To be quantitative about the variation in the cross section with the different BCAL timing cuts I took the ratio of the omega->3pi yield Data/MC from each of these cuts and plotted it vs the BCAL DeltaT cut width.  A red line is fit with a constant to see the size of the deviation.  

[image]

The largest deviation between the points is 0.3% between the nominal 1 ns cut and the widest 20 ns cut, which is consistent with Alex's observation for the rho. 

This is also consistent with the reasonable data/MC agreement seen in the previous DeltaT spectra for this channel (earlier in the issue thread https://github.com/JeffersonLab/halld_recon/issues/792#issuecomment-2035168903)

s6pepaul, 2024-04-10:
The previous spectra I have shown were not acceptance corrected. So I repeated my studies. The results are here:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

The MC tails don't extend too far I think. Looks decent overall. There is however this shift between data and MC. One can see that better when zooming in. The green dotted line shows the nominal cut limits applied during analysis launches.
[image]
[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]

Here the shift becomes really apparent. One can also see that the first hit time MC does not show this shift, but it is too narrow.
For the Lambda(1405) line shape studies I cut at 0.5ns for kaons. I performed some Barlow tests to see what the effect is. The plot below shows the result (ignore the TOF, it is irrelevant for this study but I had this plot at hand and didn't want to remake it).
[all_KmPID_2018_08.pdf](https://github.com/JeffersonLab/halld_recon/files/14931060/all_KmPID_2018_08.pdf)

Here one can see that there is no real difference between a cut at 0.5 and 0.6ns. If I cut tighter, there is a small shift in the pulls, albeit not yet significant for the Barlow test.

s6pepaul, 2024-04-10:
(my previous comment accidentally said "acceptance corrected" what I meant was "accidental subtracted", I edited the comment on GitHub, posting here for people who only see the email)

zihlmann, 2024-04-10:
I think it is fair to say that there are several points here:
1) The peak position difference between DATA and MC is "inflated" by averaging the hit-times
     taking the first hit for the time is clearly a better choice.
2) the peak position difference between Data and MC depends on momentum
3) the resolution difference between Data and MC depends on particle type
    e.g. the width of the peaks in data and MC for the proton are very similar while
   for the Pions the MC is much narrower.

jrstevenjlab, 2024-04-11:
Here's a followup on the data/MC comparison for the Delta++ channel, where we have recoil protons and slow pi+ in in the BCAL to consider.  For proton-pi+ mass < 1.35 GeV, here are the DeltaT distributions for a few different ranges on the x-axis.  Using the same color scheme as @s6pepaul, these include MC both with the nominal (energy-weighted) time and the first-hit time option in hdgeant4.  The proton distributions appear to match rather well, like in the other channel we've considered.

The pions however seems to have a long tail at large DeltaT in both BCAL simulation versions, but its much smaller in magnitude than the TOF issue we already addressed.

[image]
[image]
[image]

Below are plots showing fits to the zoomed in BCAL DeltaT distributions for both simulation versions

[image]
[image]

The nominal simulation version (red) seem to match the width of the distribution rather well for both pions and protons, but there is a ~100 ps shift in the centroid between between data and MC.  The first-hit simulation version (black) have a narrower width than the data, but have a smaller shift in the centroid.

Finally, here is the ratio of data/MC yields for the Delta++ in the mass range < 1.35 GeV that uses the same procedure as the omega yield ratios earlier in the thread.  The 2 plots below are for the nominal BCAL MC:

[image]
[image]

The ratio decreases as the DeltaT cut is widened, with a 1% difference between the nominal cut of 1 ns and the widest cut of 20 ns.  For narrower cuts than the nominal 1 ns the MC efficiency seems to be underestimated, which increases the ratio due to the difference in the DeltaT distributions with the significant shift in the centroid shown above.

Finally, the same ratio was determined below for the first-hit BCAL MC:

[image]

The ratio for this first-hit BCAL MC actually decreases for narrower DeltaT cuts, indicating that the resolution of the DeltaT distribution is narrower than the data, which causes the MC efficiency to be overestimated and lower the ratio of data/MC yields.  Again, the difference from the 1 to 20 ns cuts is ~1%.

So for this case it seems both BCAL simulations versions have a cross section that will depend on the DeltaT cut applied, but the systematic appear to be at the ~1% level.

sdobbs, 2024-04-11:
Thanks for these studies - they are very reassuring.  I think the variation seen here are good enough for the analyses that are currently underway, and there's some room for improvement in the future.

Do we want to close this issue for now, if we're happy with the current code for now, or leave it open if there are people interested in fine tuning this response some more?

jrstevenjlab, 2024-04-12:
I agree, I think for analyses that use the nominal cuts of 1.0 ns (0.75 ns) for pions and protons (kaons) in the BCAL we can set a systematic at ~1% per track and continue using the existing code.  The alternate BCAL code with first-hit timing seems to overshoot in the other direction, in addition to other potential complications with neutral reconstruction, so would not be beneficial at this point.

That said, if analyses are using BCAL DeltaT cuts narrower than the nominal analysis launch requirements, we will need to consider additional systematic studies.

Closing this issue for now and further tuning should probably be discussed in halld_sim since that's where the smearing parameters are applied.

closed 2024-04-12 by jrstevenjlab