JeffersonLab/halld_recon #800: Crash Using mcsmear When Attempting to Merge Real Noise with Simulated Events

Shuqihong, 2024-04-16:
I encountered an issue when trying to merge real noise into simulated events following the instructions from the [https://halldweb.jlab.org/wiki/index.php/HOWTO_merge_real_noise_into_simulated_events](url) page.

Command Used:
`mcsmear /volatile/halld/home/alfab/bhmuons/hddm/101142/geant4_smeared_101142_0*.hddm /volatile/halld/home/shong/random_trigger/converted_random/101142/converted_random_101142_0*.hddm:1`

Expected Behavior:
The command should successfully merge the real noise from `converted_random_101142_0*.hddm` into the simulated events file `geant4_smeared_101142_0*.hddm`.

Actual Behavior:
The process crashes with unreadable error messages (gibberish).

Debugging Attempts:
I attempted to diagnose the issue using the gdb debugger with the following commands:
`gdb --args mcsmear /volatile/halld/home/alfab/bhmuons/hddm/101142/geant4_smeared_101142_0*.hddm /volatile/halld/home/shong/random_trigger/converted_random/101142/converted_random_101142_0*.hddm:1`
`run`
`handle SIGTTOU nostop noprint pass`
`continue`
The outputs from gdb were also unreadable and the application went into an endless loop after the `continue` command.

Could someone please help investigate this issue? Any suggestions on how to resolve or further diagnose the problem would be greatly appreciated.

[image]

zihlmann, 2024-04-16:
I think this issue should be in "halld_sim"

mcsmear already crashes with a segmentation fault just one input file not even trying to use noise hit file:
I do not think this has anything to do with merging noise files but more with mcsmear and smearing an MC file. 

running gdb:
gdb mcsmear 

run /volatile/halld/home/alfab/bhmuons/hddm/101142/geant4_smeared_101142_000.hddm

and get 
JANA >> -------------------------------
  4.4k events processed  (4.4k events read)  306.0Hz  (avg.: 293.8Hz)         
WARNING: CDC hits already exist in input file! Overwriting!

  4.6k events processed  (4.6k events read)  304.0Hz  (avg.: 294.1Hz)     
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd65d6700 (LWP 123454)]
0x0000000000000065 in ?? ()
Missing separate debuginfos, use: debuginfo-install bzip2-libs-1.0.6-13.el7.x86_64 cyrus-sasl-lib-2.1.26-24.el7_9.x86_64 expat-2.1.0-15.el7_9.x86_64 freetype-2.8-14.el7_9.1.x86_64 glibc-2.17-326.el7_9.x86_64 gsl-1.15-13.el7.x86_64 keyutils-libs-1.5.8-3.el7.x86_64 krb5-libs-1.15.1-55.el7_9.x86_64 libcom_err-1.42.9-19.el7.x86_64 libcurl-7.29.0-59.el7_9.2.x86_64 libgcc-4.8.5-44.el7.x86_64 libgfortran-4.8.5-44.el7.x86_64 libicu-50.2-4.el7_7.x86_64 libidn-1.28-4.el7.x86_64 libpng-1.5.13-8.el7.x86_64 libquadmath-4.8.5-44.el7.x86_64 libselinux-2.5-15.el7.x86_64 libssh2-1.8.0-4.el7_9.1.x86_64 libstdc++-4.8.5-44.el7.x86_64 libtirpc-0.2.4-0.16.el7.x86_64 mariadb-libs-5.5.68-1.el7.x86_64 ncurses-libs-5.9-14.20130511.el7_4.x86_64 nspr-4.35.0-1.el7_9.x86_64 nss-3.90.0-2.el7_9.x86_64 nss-softokn-freebl-3.90.0-6.el7_9.x86_64 nss-util-3.90.0-1.el7_9.x86_64 openldap-2.4.44-25.el7_9.x86_64 openssl-libs-1.0.2k-26.el7_9.x86_64 pcre-8.32-17.el7.x86_64 sssd-client-1.16.5-10.el7_9.16.x86_64 xz-libs-5.2.2-2.el7_9.x86_64 zlib-1.2.7-21.el7_9.x86_64
(gdb) bt
#0  0x0000000000000065 in ?? ()
#1  0x00000000006b6b62 in hddm_s::HDDM_ElementList<hddm_s::FdcTruthPoint>::streamer(hddm_s::ostream&) ()
#2  0x00000000006b7693 in hddm_s::ostream::operator<<(hddm_s::streamable&) ()
#3  0x00000000006b57b2 in hddm_s::HDDM_ElementList<hddm_s::FdcChamber>::streamer(hddm_s::ostream&) ()
#4  0x00000000006b7693 in hddm_s::ostream::operator<<(hddm_s::streamable&) ()
#5  0x00000000006b7693 in hddm_s::ostream::operator<<(hddm_s::streamable&) ()
#6  0x00000000006ba5ec in hddm_s::HitView::streamer(hddm_s::ostream&) ()
#7  0x00000000006b7693 in hddm_s::ostream::operator<<(hddm_s::streamable&) ()
#8  0x00000000006ba89b in hddm_s::PhysicsEvent::streamer(hddm_s::ostream&) ()
#9  0x00000000006b4822 in hddm_s::HDDM_ElementList<hddm_s::PhysicsEvent>::streamer(hddm_s::ostream&) ()
#10 0x00000000006b7693 in hddm_s::ostream::operator<<(hddm_s::streamable&) ()
#11 0x00000000006b7693 in hddm_s::ostream::operator<<(hddm_s::streamable&) ()
#12 0x0000000000762ff5 in hddm_s::ostream::operator<<(hddm_s::HDDM&) ()
#13 0x000000000075f498 in MyProcessor::evnt(jana::JEventLoop*, unsigned long) ()
#14 0x000000000130a89a in jana::JEventLoop::OneEvent (this=this@entry=0x7fffc40008c0) at src/JANA/JEventLoop.cc:693
#15 0x000000000130b834 in jana::JEventLoop::Loop (this=this@entry=0x7fffc40008c0) at src/JANA/JEventLoop.cc:496
#16 0x00000000012e32ba in LaunchThread (arg=0x7fffffff9d30) at src/JANA/JApplication.cc:1382
#17 0x00007ffff01d9ea5 in start_thread () from /lib64/libpthread.so.0
#18 0x00007fffef4deb0d in clone () from /lib64/libc.so.6
(gdb) 
(gdb) up
#14 0x000000000130a89a in jana::JEventLoop::OneEvent (this=this@entry=0x7fffc40008c0) at src/JANA/JEventLoop.cc:693
693				proc->evnt(this, event_number);
(gdb) list
688			}
689			proc->UnlockState();
690	
691			// Call the event routine
692			try{
693				proc->evnt(this, event_number);
694			}catch(exception &e){
695				error_call_stack_t cs = {"JEventLoop", "OneEvent  (evnt)", __FILE__, __LINE__};
696				error_call_stack.push_back(cs);
697				PrintErrorCallStack();
(gdb) print event_number
$1 = 4632

rjones30, 2024-04-16:
Shuqihong, you are trying to smear an already smeared file a second time.
This is not supported. Why are you double-smearing it? Did you want to
simulate twice the uncertainty on one of the detectors?
-Richard Jones

Shuqihong, 2024-04-16:
> Shuqihong, you are trying to smear an already smeared file a second time. This is not supported. Why are you double-smearing it? Did you want to simulate twice the uncertainty on one of the detectors? -Richard Jones
> […](#)

Ah! You're right. I understood. Thanks!

closed 2024-04-18 by rjones30