JeffersonLab/halld_recon #802: TOF performance fail for counter(paddle location, UP/DW) 23,24, vertical plane and counter(paddle location, S) 23

zihlmann, 2024-04-17:
the demonstrate the problem two plots are shown of PMT# 23 the TOP PMT of paddle location 23 (a single ended paddle)
for run 111790 (all good) and for run 120286 (bad). The plots show the hit time of the TDC minus the hit time of the adc on the vertical axis and the adc signal peak value on the horizontal axis. This plot is used to determine the time walk caused by the discriminator.

**RUN 111790**
[image]

**RUN 120286**
[image]

The time difference in the second plot clearly does not show a nice correlation between the time difference and the signal amplitude. A smilar distribution like this one would see if the TDC time is measured using its internal clock rather than the external clock provided by the DAQ. However, this is most likely caused by a "bad" cable connection between the discriminator and the TDC. Since this is seen in the four additional neighboring channels, so the last 5 channels of the connector but not the first 3 of this special cable used for this particular case it is most likely that there is a problem with the cable rather than the TDC. As a consequence these counters will cause a "gap" in the overall acceptance of the TOF as seen in the monitoring plots like this one:
**Monitoring plot TOF2 run 120286**
[image]

zihlmann, 2024-04-17:
To support the assumption of a "bad" cable here the readout list of the TDC in slot8:
   <slot number="8" type="VX1290A">
      <channel number="0" detector="TOF" plane="0" bar="31" end="DW"/>
      <channel number="1" detector="TOF" plane="0" bar="32" end="DW"/>
      <channel number="2" detector="TOF" plane="0" bar="33" end="DW"/>
      <channel number="3" detector="TOF" plane="0" bar="34" end="DW"/>
      <channel number="4" detector="TOF" plane="0" bar="35" end="DW"/>
      <channel number="5" detector="TOF" plane="0" bar="36" end="DW"/>
      <channel number="6" detector="TOF" plane="0" bar="37" end="DW"/>
      <channel number="7" detector="TOF" plane="0" bar="38" end="DW"/>
      <channel number="8" detector="TOF" plane="0" bar="39" end="DW"/>
      <channel number="9" detector="TOF" plane="0" bar="40" end="DW"/>
      <channel number="10" detector="TOF" plane="0" bar="41" end="DW"/>
      <channel number="11" detector="TOF" plane="0" bar="42" end="DW"/>
      <channel number="12" detector="TOF" plane="0" bar="43" end="DW"/>
      <channel number="13" detector="TOF" plane="0" bar="44" end="DW"/>
      <channel number="14" detector="TOF" plane="0" bar="45" end="DW"/>
      <channel number="15" detector="TOF" plane="0" bar="46" end="DW"/>
      <channel number="16" detector="TOF" plane="1" bar="23" end="N"/>
      <channel number="17" detector="TOF" plane="1" bar="24" end="N"/>
      <channel number="18" detector="TOF" plane="1" bar="23" end="S"/>
      <channel number="19" detector="TOF" plane="1" bar="24" end="S"/>
      <channel number="20" detector="TOF" plane="0" bar="23" end="UP"/>
      <channel number="21" detector="TOF" plane="0" bar="24" end="UP"/>
      <channel number="22" detector="TOF" plane="0" bar="23" end="DW"/>
      <channel number="23" detector="TOF" plane="0" bar="24" end="DW"/>
      <channel number="26" detector="TAC"/>
      <channel number="30" detector="DIRC_REF"/>
      <channel number="31" detector="RF" system="TOF" type="VX1290A"/>
    </slot>
and the picture showing the input cabling to the TDC:
[image]

channel 31 is the RF and channel 30 is the DIRC-Reference Signal channel 26 to 29 are not used

sdobbs, 2024-08-26:
We decided to mask out these channels, and the PR to add some new code to help with this is #825.

Closing...

closed 2024-08-26 by sdobbs