JeffersonLab/halld_recon #819: Patch recon-compatible version sets for ECAL

aaust, 2024-06-25:
After the inclusion of the ECAL library, the recon-compatible tags are not consistent with current halld_sim and hdgeant4 versions and need to be patched.

sdobbs, 2024-08-08:
I've updated the following branches

recon-2017_01-ver03-sim
recon-2018_01-ver02-sim
recon-2018_08-ver02-sim
recon-2019_11-ver01-sim

sdobbs, 2024-08-22:
is there anything that I should do to help test this?

aaust, 2024-08-22:
I plan to make a set of new version sets this week. If all goes well, I can close this issue.

aaust, 2024-09-04:
After a few iterations, I tagged the halld_recon releases:
[recon-2017_01-ver03.11](https://github.com/JeffersonLab/halld_recon/releases/tag/recon-2017_01-ver03.11)
[recon-2018_01-ver02.10](https://github.com/JeffersonLab/halld_recon/releases/tag/recon-2018_01-ver02.10)
[recon-2018_08-ver02.10](https://github.com/JeffersonLab/halld_recon/releases/tag/recon-2018_08-ver02.10)
[recon-2019_11-ver01.4](https://github.com/JeffersonLab/halld_recon/releases/tag/recon-2019_11-ver01.4)

and built all version sets for the CentOS7 container:
[recon-2017_01-ver03_40.xml](https://halldweb.jlab.org/halld_versions/recon-2017_01-ver03_40.xml)
[recon-2018_01-ver02_32.xml](https://halldweb.jlab.org/halld_versions/recon-2018_01-ver02_32.xml)
[recon-2018_08-ver02_31.xml](https://halldweb.jlab.org/halld_versions/recon-2018_08-ver02_31.xml)
[recon-2019_11-ver01_13.xml](https://halldweb.jlab.org/halld_versions/recon-2019_01-ver01_13.xml)

closed 2024-09-04 by aaust