JeffersonLab/halld_recon #838: Inconsistent treatment of resonances

aaust, 2024-09-16:
Is there a reason why we treat omega(782) and phi(1020) different from other hadronic resonances in our analysis code? bggen for example creates the resonances rho(770) and Delta(1232) as particles, but we actively suppress these states in our topology.

By including them in this function
https://github.com/JeffersonLab/halld_recon/blob/3e07e1a4cfdc08225d91158b45d4dda9e8edaf24/src/libraries/include/particleType.h#L1613
we would be able to discern for example rho0 production from non-resonant pi+pi- or pi- Delta++ in the pi+pi-p final state:
[image]

Since resonances do not have the IsFixedMass() flag set, there is no danger that the mass is constrained in the kinematic fit. It would only change the behavior of the ReactionFilter, which currently aborts with this error message:
```
ERROR: CANNOT SET RESONANCE PID. ABORTING.
Aborted (core dumped)
```
Default decays should then also be implemented for the resonances in the ReactionFilter.

Thanks to @igjaegle for noticing this inconsistency.

dglazier, 2024-09-17:
Just to note, in my own (CLAS12) code I handle combinitorials differently for omega/phi and rho/Delta. Perhaps it is the same cause here. If a state is _sufficiently_ narrow then you may decide to handle it as a genuine free final state particle. If it is broad then its production may interfere with other processes of the same final state. For the case of a reaction producing a narrow state, such as : omega(pi+pi-pi0) + pi0 then you may just take both combinations of the pi0 , cut around the omega mass and subtract off any residual background assuming interference effects are neglible. This allows you to apply a vector-pseudoscalar analysis formalism to analyse this reaction. If you have, for example, rho(pi+pi-) + pi+ then it is not such a good approximation to do this and you should analyse this as only a single combinatoric event and symmetrise the pions at the amplitude level.

For topology counting this is probably not important, so it might be fine here.

sdobbs, 2024-09-18:
I agree with Derek that the interpretation of a narrow resonance is a lot more straightforward for both event selections and further analysis.

I think in halld_recon, there are two different issues.  In terms of propagating information from the event generator, I can sort of see the motivation here - though like with the rho0, it's maybe worth thinking / discussing in a little more detail why it would be important to separate out pi+pi-p events that come from the rho0 and ones that don't.

The implementation for the ReactionFilter would be fine.... but then if this is just applying a mass window, wouldn't most of the time the mass window cut that is being applied be so wide that you wouldn't be rejecting that many events anyway?

nsjarvis, 2024-09-26:
I'm just here to register my objection to the rude behaviour of the code if someone does try to include DeltaPlusPlus in a reaction filter.  A nice polite error message and exit would suffice, instead of SHOUTING in capitals and then a core dump.