JeffersonLab/halld_recon #843: Rarely-used plugins should be optional targets, outdated ones removed

aaust, 2024-09-27:
We currently compile 165 plugins by default. This takes time and the compiled libraries take up disk space. Only about 50% of the plugins are regularly used for in production, for monitoring, calibration or analysis. Many others are outdated or only serve a very limited purpose.

Could we identify which plugins are important to be included in the default build, which ones can be kept as optional targets and which ones can be retired? Here is a list of all plugins that are currently compiled by default. I marked the ones used in production:

- [ ] 1p1pi1pi0.so
- [ ] 1p2pi.so
- [ ] 2trackskim.so
- [x] b1pi_hists.so
- [ ] B3pi_eff_missgamma.so
- [ ] B3pi_eff_misspim.so
- [ ] B3pi_eff_misspip.so
- [ ] B3pi_eff_missprot.so
- [x] BCAL_attenlength_gainratio.so
- [x] BCAL_Eff.so
- [ ] BCAL_gainmatrix.so
- [x] BCAL_Hadronic_Eff.so
- [x] BCAL_inv_mass.so
- [x] BCAL_LEDonline.so
- [ ] BCAL_LED.so
- [ ] BCAL_LED_time.so
- [x] BCAL_online.so
- [ ] BCAL_point_calib.so
- [ ] BCAL_point_time.so
- [ ] BCAL_saturation.so
- [ ] BCAL_SiPM_saturation.so
- [x] BCAL_TDC_Timing.so
- [x] BEAM_online.so
- [ ] bigevents_skim.so
- [ ] cal_high_energy_skim.so
- [x] CCAL_online.so
- [x] CDC_amp.so
- [ ] cdcbcal_skim.so
- [x] CDC_dedx.so
- [x] CDC_drift.so
- [x] CDC_Efficiency.so
- [x] CDC_expert_2.so
- [ ] CDC_expert.so
- [ ] cdc_goodtrack_skim.so
- [x] CDC_online.so
- [ ] CDC_PerStrawReco.so
- [ ] CDC_roc_hits.so
- [x] CDC_TimeToDistance.so
- [x] coherent_peak_skim.so
- [ ] compton_neutral_skim.so
- [ ] compton.so
- [ ] cppFMWPC_ana.so
- [ ] cppFMWPC.so
- [x] cpp_hists.so
- [x] cpp_itrig.so
- [ ] cpp_skim.so
- [x] danarest.so
- [x] DAQ_online.so
- [ ] DAQTreeBCAL.so
- [ ] DAQTree.so
- [ ] dedx_tree.so
- [x] dirc_hists.so
- [x] DIRC_online.so
- [x] dirc_reactions.so
- [ ] dirc_tree.so
- [ ] EPICS_dump.so
- [ ] es_test.so
- [ ] eta2g_primexd_skim.so
- [ ] eta6g_primexd_skim.so
- [ ] etapi0_primexd_skim.so
- [ ] EventTagPi0.so
- [ ] evio-hddm.so
- [x] evio_writer.so
- [ ] exclusivepi0skim.so
- [x] fa125_itrig.so
- [ ] fcal_charged.so
- [x] FCAL_cpp.so
- [x] FCAL_Hadronic_Eff.so
- [x] FCAL_invmass.so
- [ ] FCAL_LED_shifts.so
- [ ] FCALLEDTree.so
- [x] FCAL_online.so
- [ ] FCAL_Pi0HFA.so
- [ ] FCAL_Pi0TOF.so
- [x] FCAL_TimingOffsets_Primex.so
- [x] FDC_Efficiency.so
- [ ] FDC_InternalAlignment.so
- [x] FDC_online.so
- [ ] FDCProjectionResiduals.so
- [x] FMWPC_online.so
- [x] FMWPC_Performance.so
- [ ] HELI_online.so
- [x] highlevel_online.so
- [x] HLDetectorTiming.so
- [x] imaging.so
- [x] L1_online.so
- [x] lowlevel_online.so
- [x] lumi_mon.so
- [ ] lut_dirc.so
- [ ] mcthrown_hists.so
- [x] mcthrown_tree.so
- [x] MilleFieldOff.so
- [x] MilleFieldOn.so
- [x] MilleKs.so
- [x] monitoring_hists.so
- [x] npp_hists.so
- [ ] npp_skim.so
- [x] occupancy_online.so
- [ ] omega_skim.so
- [ ] p2gamma_hists.so
- [ ] p2k_hists.so
- [ ] p2pi0_hists.so
- [x] p2pi_hists.so
- [x] p3pi_hists.so
- [ ] p4pi_hists.so
- [ ] pedestal_online.so
- [ ] phi_skim.so
- [ ] pi0bcalskim.so
- [ ] pi0fcalskim.so
- [x] pi0fcaltofskim.so
- [ ] Pi0Finder.so
- [ ] pid_dirc.so
- [x] ppi0gamma_hists.so
- [x] primex-online.so
- [x] PSC_online.so
- [ ] PSC_TW.so
- [ ] PS_E_calib.so
- [x] PS_flux.so
- [x] PS_online.so
- [x] PSPair_online.so
- [x] ps_skim.so
- [x] PS_timing.so
- [x] randomtrigger_skim.so
- [x] ReactionEfficiency.so
- [x] ReactionFilter.so
- [x] RF_online.so
- [ ] RSAI_KO.so
- [ ] run_summary.so
- [ ] scaler_primex.so
- [ ] SC_Eff.so
- [ ] single_neutral_skim.so
- [ ] ST_online_efficiency.so
- [x] ST_online_lowlevel.so
- [ ] ST_online_multi.so
- [ ] ST_online_tracking.so
- [ ] ST_online_Tresolution.so
- [ ] ST_Propagation_Time.so
- [x] ST_Tresolution.so
- [ ] st_tw_corr_auto.so
- [x] syncskim.so
- [x] TAGGER_online.so
- [x] TAGH_online.so
- [x] TAGH_timewalk.so
- [x] TAGM_clusters.so
- [x] TAGM_online.so
- [x] TAGM_TW.so
- [ ] timing_online.so
- [x] TOF_calib.so
- [ ] TOF_Eff.so
- [x] TOF_online.so
- [x] TOF_TDC_shift.so
- [x] TPOL_online.so
- [x] TPOL_tree.so
- [ ] trackeff_hists.so
- [x] trackeff_missing.so
- [x] TrackingPulls.so
- [x] TrackingPulls_straight.so
- [ ] track_skimmer.so
- [ ] TRD_hists.so
- [ ] TRD_online.so
- [x] trigger_skims.so
- [x] TRIG_online.so
- [ ] truth_dirc.so
- [ ] TS_scaler.so
- [ ] twogamma_fcal_skim.so

nsjarvis, 2024-09-27:
Hi,

CDC_expert can be retired.

I think these can be made optional.  Someone else (Sean, Mark?) might wish
to retain the skims in the default build set.

   -  cdcbcal_skim.so
   -  cdc_goodtrack_skim.so

   -  CDC_PerStrawReco.so
   -  CDC_roc_hits.so
   -  dedx_tree.so

Could we make all of the non-production plugins optional unless someone
speaks up for them?

Naomi.

sdobbs, 2024-10-03:
what does "retired" mean, here? removed from the repo?

aaust, 2024-10-03:
@sdobbs Correct. In scenarios where a plugin was superseded by another one or if a plugin would require significant changes to even compile, we could remove it.

sdobbs, 2024-10-07:
One thing we could consider is making some of the plugin subdirectories optional targets - like if you had to explicitly request the alignment or monitoring plugins to be built

sdobbs, 2024-10-07:
Here are my comments / opinions on these plugins:

- [ ] 1p1pi1pi0.so - optional
- [ ] 1p2pi.so - optional 
- [ ] 2trackskim.so - optional
- [x] b1pi_hists.so - default 
- [ ] B3pi_eff_missgamma.so - optional, but can we maybe merge these 4 B3pi_eff plugins?  we could add flags to toggle which to use
- [ ] B3pi_eff_misspim.so
- [ ] B3pi_eff_misspip.so
- [ ] B3pi_eff_missprot.so
- [x] BCAL_attenlength_gainratio.so - default 
- [x] BCAL_Eff.so - default
- [ ] BCAL_gainmatrix.so - optional
- [x] BCAL_Hadronic_Eff.so - default
- [x] BCAL_inv_mass.so - default
- [x] BCAL_LEDonline.so - is this actually used, and is it actually run online? would be good to merge this with BCAL_LED if possible, since there seems to be some redundancy 
- [ ] BCAL_LED.so
- [ ] BCAL_LED_time.so - freezer 
- [x] BCAL_online.so
- [ ] BCAL_point_calib.so - freezer
- [ ] BCAL_point_time.so - freezer
- [ ] BCAL_saturation.so - optional 
- [ ] BCAL_SiPM_saturation.so  - optional
- [x] BCAL_TDC_Timing.so - default
- [x] BEAM_online.so  - default
- [ ] bigevents_skim.so   - optional
- [ ] cal_high_energy_skim.so   - optional
- [x] CCAL_online.so   - default
- [x] CDC_amp.so   - default
- [ ] cdcbcal_skim.so   - optional
- [x] CDC_dedx.so   - default
- [x] CDC_drift.so   - default
- [x] CDC_Efficiency.so   - default
- [x] CDC_expert_2.so
- [ ] CDC_expert.so
- [ ] cdc_goodtrack_skim.so - couldn't find this for some reason
- [x] CDC_online.so
- [ ] CDC_PerStrawReco.so
- [ ] CDC_roc_hits.so
- [x] CDC_TimeToDistance.so
- [x] coherent_peak_skim.so - optional
- [ ] compton_neutral_skim.so - optional
- [ ] compton.so - optional
- [ ] cppFMWPC_ana.so - optional
- [ ] cppFMWPC.so - optional
- [x] cpp_hists.so - optional
- [x] cpp_itrig.so - optional
- [ ] cpp_skim.so - optional
- [x] danarest.so - default 
- [x] DAQ_online.so - default 
- [ ] DAQTreeBCAL.so  - optional
- [ ] DAQTree.so  - optional
- [ ] dedx_tree.so  - optional
- [x] dirc_hists.so - default
- [x] DIRC_online.so - default
- [x] dirc_reactions.so  - optional
- [ ] dirc_tree.so   - optional
- [ ] EPICS_dump.so    - optional
- [ ] es_test.so    - optional
- [ ] eta2g_primexd_skim.so    - optional
- [ ] eta6g_primexd_skim.so    - optional
- [ ] etapi0_primexd_skim.so    - optional
- [ ] EventTagPi0.so    - optional
- [ ] evio-hddm.so - optional
- [x] evio_writer.so - default
- [ ] exclusivepi0skim.so - optional
- [x] fa125_itrig.so - default
- [ ] fcal_charged.so - remove
- [x] FCAL_cpp.so - optional
- [x] FCAL_Hadronic_Eff.so  - default
- [x] FCAL_invmass.so  - default
- [ ] FCAL_LED_shifts.so - optional
- [ ] FCALLEDTree.so - optional
- [x] FCAL_online.so   - default
- [ ] FCAL_Pi0HFA.so - optional
- [ ] FCAL_Pi0TOF.so - optional
- [x] FCAL_TimingOffsets_Primex.so - optional
- [x] FDC_Efficiency.so - default
- [ ] FDC_InternalAlignment.so - optional
- [x] FDC_online.so    - default
- [ ] FDCProjectionResiduals.so - optional
- [x] FMWPC_online.so - optional
- [x] FMWPC_Performance.so - optional
- [ ] HELI_online.so - optional
- [x] highlevel_online.so - default
- [x] HLDetectorTiming.so - default
- [x] imaging.so - default
- [x] L1_online.so - default
- [x] lowlevel_online.so - default
- [x] lumi_mon.so - default
- [ ] lut_dirc.so - optional
- [ ] mcthrown_hists.so - optional
- [x] mcthrown_tree.so - default
- [x] MilleFieldOff.so - optional
- [x] MilleFieldOn.so - optional
- [x] MilleKs.so - optional
- [x] monitoring_hists.so - default
- [x] npp_hists.so - optional
- [ ] npp_skim.so - optional
- [x] occupancy_online.so - default
- [ ] omega_skim.so - optional
- [ ] p2gamma_hists.so - optional
- [ ] p2k_hists.so - optional
- [ ] p2pi0_hists.so - optional
- [x] p2pi_hists.so - default
- [x] p3pi_hists.so - default
- [ ] p4pi_hists.so - optional
- [ ] pedestal_online.so - optional
- [ ] phi_skim.so - optional
- [ ] pi0bcalskim.so - optional
- [ ] pi0fcalskim.so - optional
- [x] pi0fcaltofskim.so - default
- [ ] Pi0Finder.so - optional - not sure what this is
- [ ] pid_dirc.so - optional
- [x] ppi0gamma_hists.so
- [x] primex-online.so - optional
- [x] PSC_online.so - default
- [ ] PSC_TW.so - optional
- [ ] PS_E_calib.so - optional
- [x] PS_flux.so - default
- [x] PS_online.so - default
- [x] PSPair_online.so - default
- [x] ps_skim.so - default
- [x] PS_timing.so - default - could merge this with one of the other PS plugins
- [x] randomtrigger_skim.so -optional
- [x] ReactionEfficiency.so - default
- [x] ReactionFilter.so - default
- [x] RF_online.so - default
- [ ] RSAI_KO.so - optional
- [ ] run_summary.so - optional
- [ ] scaler_primex.so - optional
- [ ] SC_Eff.so - optional
- [ ] single_neutral_skim.so - optional
- [ ] ST_online_efficiency.so - optional
- [x] ST_online_lowlevel.so - default - note that the start counter plugins should be refreshed
- [ ] ST_online_multi.so - optional - probably get rid of this?
- [ ] ST_online_tracking.so - optional - probably get rid of this?
- [ ] ST_online_Tresolution.so - optional - probably get rid of this?
- [ ] ST_Propagation_Time.so
- [x] ST_Tresolution.so - default
- [ ] st_tw_corr_auto.so - optional
- [x] syncskim.so - default
- [x] TAGGER_online.so - default
- [x] TAGH_online.so - default
- [x] TAGH_timewalk.so - default
- [x] TAGM_clusters.so - optional
- [x] TAGM_online.so - default
- [x] TAGM_TW.so - default
- [ ] timing_online.so - optional
- [x] TOF_calib.so - default
- [ ] TOF_Eff.so
- [x] TOF_online.so - default
- [x] TOF_TDC_shift.so - default
- [x] TPOL_online.so - default
- [x] TPOL_tree.so  - optional
- [ ] trackeff_hists.so  - optional
- [x] trackeff_missing.so - default
- [x] TrackingPulls.so - default
- [x] TrackingPulls_straight.so  - optional
- [ ] track_skimmer.so  optional
- [ ] TRD_hists.so  optional
- [ ] TRD_online.so  optional
- [x] trigger_skims.so  - default
- [x] TRIG_online.so  - default
- [ ] truth_dirc.so  optional
- [ ] TS_scaler.so  optional
- [ ] twogamma_fcal_skim.so  optional

aaust, 2025-03-13:
From #888: BCAL_LEDonline is crashing in Init(), FCALLEDTree in Finish() and FCAL_LED_shifts, should be made optional