JeffersonLab/halld_recon #850: Inconsistently counted beam photon in reconstructed MC/thrown MC at beam energies where TAGH and TAGM overlaps

lihaoahil, 2024-10-29:
## What happened?
This is another subtle issue that differs a bit from the already resolved issue #810 (which create a dip in the reconstructed efficiency). Instead, it results in spike-like structure in the reconstructed efficiency at around beam energy of 8.9 GeV where the tagger hodoscope TAGH\#127 overlaps almost entirely with microscope column TAGM\#1. Specific beam energies may differ according to different electron beam energy from run to run and different tagger energy scale from run period to run period, but similar spike structure is seen in both ppbar and lamlambar channel reconstruction efficiencies from all run periods in GlueX Phase-I:
[image][image]

## How to reproduce the issue and what is the cause?
Further investigation is done using ppbar MC for a single run 30496 from sp17, and no random triggers. The simulation is done using version set 5.21.0. And the beam energy is set to be between 8.880-8.890 GeV. The REST files shows that for each event, the generated beamPhoton is registered as signal in both TAGM\#1 and TAGH\#127, while the thrown information remains to be only hits at TAGM\#1:
```
Event: 1
DBeamPhoton:
   E(GeV): System: Counter: t(ns):
----------------------------------
 8.884361    TAGM        1   -4.0 
 8.888765    TAGH      127   -3.6 

DBeamPhoton:TAGGEDMCGEN
   E(GeV): System: Counter: t(ns):
----------------------------------
 8.884361    TAGM        1   -4.0 
```
 and the reconstructed efficiency at this range is estimated to be twice higher than the neighboring bin:
[image] 
(blue is MC with beam energy range 8.88-8.89 GeV, and red is another MC with beam energy range 8.88-8.90 GeV including the neighboring bin whose efficiency is normal). 

The cause seems to be that for each DBeamPhoton:TAGGEDMCGEN that goes into the mcthrown tree, there are twice number of reconstructed photons that end up in the numerator of the efficiency calculation. As shown in the table above, it seems that only TAGM hit is recorded for DBeamPhoton:TAGGEDMCGEN, which is inconsistent from the double counting in DBeamPhoton factory. Hence the efficiency doubled at that energy range. For the entire run period, because of the slightly fluctuating electron beam energy, it is smeared into a spike-like structure as spotted in the plots above.

The sample rest file with 100 events, along with other HDDM files produced during the simulation pipeline, is stored at `/w/halld-scshelf2101/home/haoli/test/spike_test/8.880-8.890/hddm`.

## Which part of the HallD code is relevant?

The double-counted beam photon seems to originated from the [Pseudo tagger in HDGeant4](https://github.com/JeffersonLab/HDGeant4/blob/master/src/GlueXPseudoDetectorTAG.cc#L305) where it allows beam photon to be detected by both hodoscope and microscope respectively.  However it might create broader impact if we don't allow beam photon to be detected by both TAGH and TAGM.

Another option might be allow double-counting beam photon in the thrown tree so that the effect cancels out in both numerator and denominator of the reconstructed efficiency calculation.

## Where else is affected?
Beside the largely overlapped TAGH\#127 and TAGM\#1, there is also a small overlapping between TAGH\#179 and TAGM\#102. Double-counted reconstructed beam photon can also be seen:
```
Event: 49
DBeamPhoton:
   E(GeV): System: Counter: t(ns):
----------------------------------
 7.898822    TAGM      102   -0.2 
 7.856271    TAGH      179   -0.3 

DBeamPhoton:TAGGEDMCGEN
   E(GeV): System: Counter: t(ns):
----------------------------------
 7.898822    TAGM      102   -0.2 
```
what is weirder is that here the two tagger are quite apart from each other. The simulation here is generated with beam energy range 7.860-7.870 GeV. (HDDM file sample at `/w/halld-scshelf2101/home/haoli/test/spike_test/7.860-7.870/hddm`)

rjones30, 2024-10-29:
> Hao,
>
> The doubling of the tags in these overlap regions would indeed produce a
> doubling of the apparent reconstruction efficiency in these regions, as you
> describe. Is a similar coincidence seen in the real data at the two ends of
> the microscope? Here are some pictures from the tagger simulation of an
> electron with energy close to the center of the first TAGM column
>
>    1. zoomed out, showing the rays through the entire focal plane array
>    2. zoomed in on TAGM fiber column 1
>    3. zoomed in on TAGH counter 128
>    4. zoomed in on TAGH counter 126 (this counter is not installed for
>    some run periods)
>    5. zoomed in on TAGH counter 127
>
> Multiple scattering means that the alignment is not perfect in real life,
> as the pseudo-tags pretend that it is. Still there should be some enhanced
> reconstruction "efficiency" in this region for real data.
>
> -Richard

PS. Correction -- I think it is counter 128 that is not installed for some
run periods, 126 should be there.
-Richard Jones

lihaoahil, 2024-10-29:
@rjones30 Hi Richard, the pictures are not shown up in your replies. 
In real data (flux normalized), it is smooth in the transition area between TAGH and TAGM:
[image]
It appears that the double counting is consistent in both raw yield and the tagged flux, so no spikes shown up.

rjones30, 2024-10-29:
>
> It appears that the double counting is consistent in both raw yield and
> the tagged flux, so no spikes shown up.
>

Aha, ok. So what we want in the simulation is to do the same thing: divide
the MC raw yield by the MC tagged flux. Then these spikes should disappear,
right?

-Richard Jones

aaust, 2024-10-29:
This issue will become central for the CPP data set, where there is a large overlap between TagM and TagH. Are you suggesting to duplicate events in data and MC when the photon can hit both detectors?

rjones30, 2024-10-29:
>
> Are you suggesting to duplicate events in data and MC when the photon can
> hit both detectors?
>

In the tagm right now we have a clustering algorithm that reduces multiple
hits from a single tag electron across adjacent counters. There is a
similar algorithm applied to adjacent counters in the tagh, which is
somewhat worse because of the stagger in the focal plane geometry. To
reduce these mixed tagh + tagm clusters we could modify our clustering
algorithm to cluster between the tagm+tagh in the overlap region, or we
could just let the simulation duplicate the "split tags" and account for it
that way.

-Richard

lihaoahil, 2024-11-14:
@rjones30 Hi Richard, I missed the Beamline meeting this week, has there been any discussion about this github issue?