JeffersonLab/halld_recon #855: ReactionFilter outputs slightly different number of events when reading from HDDM or EVIO.  (Also JANA's events read & processed don't match)

nsjarvis, 2024-11-26:
Using version_5.21.0.xml, I found that when I use ReactionFilter to generate a tree file from EVIO it contains a different number of  events than when I use danarest to make an HDDM file and then run ReactionFilter on the HDDM file.  I found larger discrepancies when testing the AddFDCHits branch, which writes some extra lines into the HDDM file.  Idk why this would be relevant.  The numbers below are from using version_5.21.0.xml.

Also I found that JANA's output of events read and events processed don't match, and the discrepancies are different depending on whether I'm using ReactionFilter or danarest.  I could imagine that it might not process SYNC events, but I can't imagine that this should vary with the plugin.  

Sean suggested that I try the above with the kinematic fit switched off and indeed that removed the discrepancies for the file that I was looking at.  

**hd_rawdata_030894_000.evio**  (raw mode)

EVIO with danarest.  126531 events processed  (126531 events read) 
HDDM with ReactionFilter -PReaction1=1_14__11_12_14 -PReaction1:Flags=B4: 118060 events processed  (118062 events read)
hd_root.root: /kpkm__B4/NumEventsSurvivedAction->GetBinContent(1) : 118062.00
tree_kpkm__B4.root: /kpkm__B4_Tree->GetEntries():63

EVIO with ReactionFilter -PReaction1=1_14__11_12_14 -PReaction1:Flags=B4   126531 events processed  (126531 events read) 
NumEventsSurvivedAction->GetBinContent(1)
hd_root.root: /kpkm__B4/NumEventsSurvivedAction->GetBinContent(1) : 118062.00
tree_kpkm__B4.root: /kpkm__B4_Tree->GetEntries():62

nsjarvis, 2024-12-02:
Looking at hd_rawdata_030894_000.evio again with different numbers of threads

[reactionfilter.pdf](https://github.com/user-attachments/files/17983315/reactionfilter.pdf)

nsjarvis, 2024-12-03:
I think this shows that (1) when reading HDDM files, hd_root's stated number of 'events processed' is not always correct, and (2) the number of events that the ReactionFilter plugin writes into an analysis tree can be slightly different, depending on whether it was reading EVIO or HDDM.

nsjarvis, 2024-12-11:
The discrepancy could be because the momenta are not stored with double precision in the HDDM file, and so when ReactionFilter requires kinematic fit convergence, it is not doing exactly the same calculation in both cases.

aaust, 2024-12-17:
Could you please make a few tests with the latest jana2 build?
> version_5.22.0_jana2.xml

The commands for hd_root are almost the same, but you have to use `--loadconfigs` for a configuration file. And the number of events to process is steered with  `jana:nevents`.

nsjarvis, 2024-12-19:
Using that build, unfortunately the danarest plugin crashed.   @RaiqaRasool 

16:53:28.526 [fatal] Exception in worker thread 1 while executing PhysicsEventMap2 on event #149
16:53:28.526 [fatal] Exception in worker thread 4 while executing PhysicsEventMap2 on event #93
----------------------------------------------------------
JException
  Type:     std::runtime_error
  Message:  hddm_r::ostream::unlock_streambufs error - mutex unlock requested when lock not held.
  Function: JEventProcessor::Process
  Plugin:   danarest
  
I attached the log file, it's quite long because it runs hd_root several times, the command line is written into the log file each time.  This was the 4-thread version.

[log_rftest_030894_000_j2_n4.txt](https://github.com/user-attachments/files/18191940/log_rftest_030894_000_j2_n4.txt)

RaiqaRasool, 2024-12-19:
Hi Naomi,

Thank you for pointing out this issue. The issue stemmed from the recent change to `hddm_r` that we discussed in the last meeting. I’ve already applied a fix to the `rasool_jana2` branch of the `halld_recon` repository and have asked Alex to rebuild the version with this fix. I’ll provide an update once the new build is available.

Thanks again for your patience!

aaust, 2024-12-19:
I just built [version_5.22.1_jana2.xml](https://halldweb.jlab.org/halld_versions/version_5.22.1_jana2.xml) and ran a successful test with the danarest plugin and 4 threads. Can you please try again?

nsjarvis, 2025-01-13:
Results from retesting with JANA2 are attached.  This time the numbers of events read are consistent.  There is still 1 less event in the kpkm__B4 tree when generated directly from EVIO than when generated via HDDM.  

[JANA2tests.pdf](https://github.com/user-attachments/files/18400180/JANA2tests.pdf)

nsjarvis, 2025-01-13:
Stats when using ReactionFilter with EVIO:
```
Input: 	118062
Has Particle Combos: 	211
Hist_MissingMassSquared_PreKinFit: 	211
KinFit Convergence: 	211
Hist_KinFitResults: 	186
Hist_PID: 	186
Cut_PIDDeltaT_11_4_0.75: 	186
Cut_PIDDeltaT_11_8_0.3: 	176
Cut_PIDDeltaT_11_32_2.5: 	149
Cut_PIDDeltaT_11_256_2.5: 	147
Cut_PIDDeltaT_11_1024_10: 	125
Cut_PIDDeltaT_12_4_0.75: 	125
Cut_PIDDeltaT_12_8_0.3: 	119
Cut_PIDDeltaT_12_32_2.5: 	106
Cut_PIDDeltaT_12_256_2.5: 	101
Cut_PIDDeltaT_12_1024_10: 	95
Cut_PIDDeltaT_14_4_1: 	95
Cut_PIDDeltaT_14_8_0.6: 	78
Cut_PIDDeltaT_14_32_2: 	70
Cut_PIDDeltaT_14_256_2.5: 	70
Cut_PIDDeltaT_14_1024_10: 	62
Hist_MissingMassSquared_PostKinFit: 	62
Hist_ParticleComboKinematics: 	62
```

Stats when using ReactionFilter with HDDM:
```
Input: 	118062
Has Particle Combos: 	211
Hist_MissingMassSquared_PreKinFit: 	211
KinFit Convergence: 	211
Hist_KinFitResults: 	187
Hist_PID: 	187
Cut_PIDDeltaT_11_4_0.75: 	187
Cut_PIDDeltaT_11_8_0.3: 	178
Cut_PIDDeltaT_11_32_2.5: 	151
Cut_PIDDeltaT_11_256_2.5: 	149
Cut_PIDDeltaT_11_1024_10: 	127
Cut_PIDDeltaT_12_4_0.75: 	127
Cut_PIDDeltaT_12_8_0.3: 	121
Cut_PIDDeltaT_12_32_2.5: 	108
Cut_PIDDeltaT_12_256_2.5: 	103
Cut_PIDDeltaT_12_1024_10: 	97
Cut_PIDDeltaT_14_4_1: 	97
Cut_PIDDeltaT_14_8_0.6: 	80
Cut_PIDDeltaT_14_32_2: 	72
Cut_PIDDeltaT_14_256_2.5: 	72
Cut_PIDDeltaT_14_1024_10: 	63
Hist_MissingMassSquared_PostKinFit: 	63
Hist_ParticleComboKinematics: 	63
```

nsjarvis, 2025-01-13:
Stats when the F0 flag was used, presumably all of the missing PID cuts were on kinfit values.
```
Input: 	118062
Has Particle Combos: 	211
Hist_MissingMassSquared_PreKinFit: 	211
Hist_PID: 	211
Hist_ParticleComboKinematics: 	211 
```

nsjarvis, 2025-01-13:
So the difference in the number of events is not due to the kinfit convergence requirement, but by the minimum kinfit chi-sq required to keep the event in the tree.

closed 2025-01-13 by nsjarvis