JeffersonLab/halld_recon #859: hd_root completing with exit code 0 but truncated output files

nsjarvis, 2024-12-11:
As reported by Igal in C&P.  The production jobs are usually configured so that they do not save the log files if the hd_root exit code is 0, which indicates success. Igal found that approx 0.25 % of CPP jobs were affected by hd_root completing with exit code 0 but the output REST and skim files were truncated. When Ilya repeated these jobs he found no problems. Sasha reported some similar problems with approx 1% of PrimEx files, he sent Alex a list of evio files that caused these problems.

aaust, 2024-12-17:
I reprocessed 1378 files from Sasha's list. 310 files failed with error 
code !=0, all others seemingly completed successfully.

The output of the successful jobs can be found here:

/cache/halld/RunPeriod-2022-08/recon/ver01a/

This time, the log files for all jobs were saved permanently on the work 
disk:

/work/halld2/recon/RunPeriod-2022-08/ver01a/log/

Please check the files and let me know if the issue still appeared.

aaust, 2024-12-17:
There is 158 files with the error message: Error reading EVIO event (truncated?)

/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110743/stdout.110743_028.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110746/stdout.110746_003.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110771/stdout.110771_014.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110773/stdout.110773_017.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110788/stdout.110788_078.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110800/stdout.110800_122.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110802/stdout.110802_064.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110808/stdout.110808_005.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110810/stdout.110810_022.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110812/stdout.110812_075.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110820/stdout.110820_018.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110820/stdout.110820_045.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110822/stdout.110822_022.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110822/stdout.110822_080.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110846/stdout.110846_036.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110852/stdout.110852_017.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110858/stdout.110858_015.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110869/stdout.110869_029.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110931/stdout.110931_055.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110937/stdout.110937_002.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110942/stdout.110942_108.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110943/stdout.110943_111.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110944/stdout.110944_036.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110951/stdout.110951_014.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110951/stdout.110951_015.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110953/stdout.110953_057.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110954/stdout.110954_060.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110955/stdout.110955_092.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110961/stdout.110961_030.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110961/stdout.110961_113.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110965/stdout.110965_106.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110971/stdout.110971_034.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110972/stdout.110972_050.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/110972/stdout.110972_074.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111005/stdout.111005_006.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111006/stdout.111006_004.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111009/stdout.111009_055.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111009/stdout.111009_086.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111031/stdout.111031_014.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111048/stdout.111048_088.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111062/stdout.111062_061.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111078/stdout.111078_007.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111078/stdout.111078_098.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111082/stdout.111082_038.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111093/stdout.111093_018.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111101/stdout.111101_016.out:Error reading EVIO event (truncated?)
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111112/stdout.111112_012.out:Error reading EVIO event (truncated?)
[... cut ...]
Subject: Re: [EXTERNAL] [JeffersonLab/halld_recon] hd_root completing with exit code 0 but truncated output files (Issue #859)

I reprocessed 1378 files from Sasha's list. 310 files failed with error
code !=0, all others seemingly completed successfully.

The output of the successful jobs can be found here:

/cache/halld/RunPeriod-2022-08/recon/ver01a/

This time, the log files for all jobs were saved permanently on the work
disk:

/work/halld2/recon/RunPeriod-2022-08/ver01a/log/

Please check the files and let me know if the issue still appeared.

aaust, 2024-12-17:
Let me add most of them have a Return Code 0  but not all

grep truncated /work/halld2/recon/RunPeriod-2022-08/ver01a/log/*/*.out | grep "Return Code" /work/halld2/recon/RunPeriod-2022-08/ver01a/log/*/*.out

/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111977/stdout.111977_090.out:Return Code =  255
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111978/stdout.111978_020.out:Return Code =  3
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111978/stdout.111978_055.out:Return Code =  255
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111978/stdout.111978_121.out:Return Code =  3
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111984/stdout.111984_016.out:Return Code =  139
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111984/stdout.111984_084.out:Return Code =  0
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111985/stdout.111985_047.out:Return Code =  0
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111985/stdout.111985_065.out:Return Code =  0
/work/halld2/recon/RunPeriod-2022-08/ver01a/log/111985/stdout.111985_068.out:Return Code =  0
________________________________
From: Alexander Austregesilo ***@***.***>
Sent: Tuesday, December 17, 2024 2:50 PM
To: ***@***.*** ***@***.***>
Cc: Alexander Somov ***@***.***>; Igal Jaegle ***@***.***>
Subject: Re: [EXTERNAL] [JeffersonLab/halld_recon] hd_root completing with exit code 0 but truncated output files (Issue #859)

I reprocessed 1378 files from Sasha's list. 310 files failed with error
code !=0, all others seemingly completed successfully.

The output of the successful jobs can be found here:

/cache/halld/RunPeriod-2022-08/recon/ver01a/

This time, the log files for all jobs were saved permanently on the work
disk:

/work/halld2/recon/RunPeriod-2022-08/ver01a/log/

Please check the files and let me know if the issue still appeared.

sdobbs, 2025-02-19:
I looked at the above files, and generally it looks like these errors in reading the raw data files are due to errors caused by the DAQ (i.e. either not closing the file properly or making an error in writing some block).

So, I'm not sure what the actionable items are here.

sdobbs, 2025-03-19:
> So, I'm not sure what the actionable items are here.

nsjarvis, 2025-03-19:
Exit code 3 pointed to the file being truncated by too-early transfer from the gluons, hoss issue early in the run.  Alex is trying to track down the original evio files.   I don't remember the explanation for code 134.    Could we explain that and then close the issue?

sdobbs, 2025-03-19:
The segfaults are due to some crash in the analysis library, at least they have nothing to do with this issue.

sdobbs, 2025-04-23:
I pulled out the segfault crash issue as #908, going to close this issue.

closed 2025-04-23 by sdobbs