JeffersonLab/halld_recon #868: Strange bin shift in DigiHits_occupancy plot

aaust, 2025-01-14:
There are 2 error msgs in the hd_root output, but the plots look good.   Actually these errors are also generated by the master code branch, so not related to this PR. 
```
[njarvis@ifarm2401.jlab.org: ~/mywork/halld_recon/src ]> hd_root /cache/halld/RunPeriod-2023-01/rawdata/Run120286/hd_rawdata_120286_000.evio -PPLUGINS=occupancy_online -PEVENTS_TO_KEEP=9000
JANA >>OUTPUT_FILENAME: hd_root.root
JANA >>Initializing plugin "/work/halld/njarvis/halld_recon/Linux_Alma9-x86_64-gcc11.4.1/plugins/occupancy_online.so" ...
Opened ROOT file "hd_root.root" ...
Error in <TAxis::SetBinLabel>: Illegal bin number: 0
Error in <TAxis::SetBinLabel>: Illegal bin number: 0 
```

_Originally posted by @nsjarvis in https://github.com/JeffersonLab/halld_recon/issues/865#issuecomment-2588397418_

aaust, 2025-01-14:
The plot from the command above looks like this:
[image]

Bin Content and Bin Label match, but the first bin is missing. Instead, there is an empty column on the right side compared to the original plot produced by rootspy during the run:
[image]

nsjarvis, 2025-01-14:
My comment that the plots look good was in Justin's PR for the DIRC plots.  I did not inspect the others.

aaust, 2025-01-14:
Thank you for the clarification. From the monitoring launch history, it seems this change was introduced between halld_recon 4.41.0 and 4.42.0, even though I can't find the exact reason:
https://github.com/JeffersonLab/halld_recon/compare/4.41.0...4.42.0

aaust, 2025-01-15:
Fixed by #869

closed 2025-01-15 by aaust