JeffersonLab/halld_recon #870: ReactionFilter bug for Decay1 16__7_7 or KShort to pi^0 pi^0

igjaegle, 2025-01-15:
Hi, I am attaching the error message.

The reaction is:
Reaction7 1_45__16_m14_18 #is for gamma d -> K0 Lambda (missing proton)                                                                                                                                     
Reaction7:Decay1 16__7_7
Reaction7:Decay2 18__9_14
Reaction7:Decay3 7__1_1
Reaction7:Flags B4

When I change Decay1 to 16__7_8 it is working.   

[log.txt](https://github.com/user-attachments/files/18429472/log.txt)

sdobbs, 2025-01-15:
Thanks, the code shouldn't crash in this case, so will fix that.

that being said, you shouldn't require a vertex constraint in this analysis, there isn't sufficient information to constrain the primary vertex.

igjaegle, 2025-01-15:
Thank you, Sean. I changed the reaction to
Reaction7 1_45__16_m14_18 #is for gamma d -> K0 Lambda (missing proton)                                                                                                                                     
Reaction7:Decay1 16__7_7
Reaction7:Decay2 18__9_14
Reaction7:Decay3 7__1_1
Reaction7:Decay4 7__1_1
Reaction7:Flags B4,F1

and it is still crashing. If the vertex is not constrain will it still find the Lambda displaced vertex and constrain the proton and pion to this displaced vertex?

sdobbs, 2025-01-15:
You should specify the flags as

Reaction7:Flags B4_F1

and no, right now, vertex constraints are all or nothing

aaust, 2025-02-03:
@igjaegle , can you please give an example input file? And can you tell us which software you are using? 
I was not able to reproduce the crash with GlueX data and a recent software release.

sdobbs, 2025-03-19:
@igjaegle - do you have any more information on how to reproduce this problem?

If not, we should probably close this issue.

sdobbs, 2025-04-24:
I'm unable to reproduce this issue with the current software, so I'm going to close it - we can reopen the issue if it reappears.

closed 2025-04-24 by sdobbs